merbanan/rtl_433 #774: Interlogix sensor has 8 outputs for every event.  Is it possible to limit the amount of outputs?

abovino, 2018-07-24:
I have a NodeJS app that creates a child process and initiates rtl_433 for an Interlogix Window/Door sensor.

`
rtl_433 -f 319500000 -R 100 -F json
`

For each event I am getting up to 8 outputs.  IE:

`
*door opens*
rtl_433 output -> switch: OPEN
rtl_433 output -> switch: OPEN
rtl_433 output -> switch: OPEN
rtl_433 output -> switch: OPEN
rtl_433 output -> switch: OPEN
rtl_433 output -> switch: OPEN
rtl_433 output -> switch: OPEN
rtl_433 output -> switch: OPEN
*door closes*
rtl_433 output -> switch: CLOSED
rtl_433 output -> switch: CLOSED
rtl_433 output -> switch: CLOSED
rtl_433 output -> switch: CLOSED
rtl_433 output -> switch: CLOSED
rtl_433 output -> switch: CLOSED
rtl_433 output -> switch: CLOSED
rtl_433 output -> switch: CLOSED
`

Is there a way to limit the amount of outputs per event?  Or is this something that's most likely built into the device that I will have no control over?  Thanks

zuckschwerdt, 2018-07-24:
Sorry, the output follows whats on the air. You have to create a debounce yourself.

closed 2018-07-24 by zuckschwerdt

baileybrent, 2018-07-24:
Here is a quick debounce solution I did in python for these devices.  You could use this logic and send it to node via some message queuing bus like mqtt without much effort.
[https://gist.github.com/baileybrent/3b382e7ae814c0b5e071931e92be4ce7](url)