merbanan/rtl_433 #775: Compile Error

ichilton, 2018-07-24:
Compiling on a Raspberry Pi (1st gen), I get this:

```
  CCLD     rtl_433
baseband.o: In function `magnitude_true_cu8':
/home/pi/rtl_433/src/baseband.c:82: undefined reference to `sqrt'
baseband.o: In function `magnitude_true_cs16':
/home/pi/rtl_433/src/baseband.c:107: undefined reference to `sqrt'
collect2: error: ld returned 1 exit status
Makefile:731: recipe for target 'rtl_433' failed
make[2]: *** [rtl_433] Error 1
make[2]: Leaving directory '/home/pi/rtl_433/src'
Makefile:494: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/pi/rtl_433'
Makefile:380: recipe for target 'all' failed
make: *** [all] Error 2
```

Thanks,

Ian

ichilton, 2018-07-24:
Works ok using cmake! - didn't using autoconf.

zuckschwerdt, 2018-07-24:
Thanks, libm is missing from the makefile. I'll fix that.

closed 2018-07-24 by zuckschwerdt

zuckschwerdt, 2018-07-25:
It's fixed with d6017e4. Is there a reason (a gut feeling is good enough) why you prefered Autotools over Cmake? I'm trying to gauge the interest in the (older and less maintained) Autotools build.