merbanan/rtl_433 #776: Oil standard TEK377 E support

vuokko, 2018-07-25:
Hello!
I don't know if E version has been tested before. However I captured some data before putting it to tank.
I got same  kind of results as the test cases. Currently I'm waiting for live data.

The diff which decoded my data:
```
diff --git a/src/devices/oil_standard.c b/src/devices/oil_standard.c
index c28c41a..52aef12 100644
--- a/src/devices/oil_standard.c
+++ b/src/devices/oil_standard.c
@@ -47,7 +47,7 @@ static int oil_standard_decode(bitbuffer_t *bitbuffer, unsigned row, unsigned bi
 
        bitpos = bitbuffer_manchester_decode(bitbuffer, row, bitpos, &databits, 33);
 
-       if (databits.bits_per_row[0] != 32)
+       if (databits.bits_per_row[0] != 32 && databits.bits_per_row[0] != 33)
                return 0;
 
        b = databits.bb[0];

```

zuckschwerdt, 2018-07-25:
Thanks. Upload some sample data here or as PR to rtl_433_tests. I'll figure out why the data is seen as 33 bits.

vuokko, 2018-07-26:
Two packets which work with 33 

[sample_packets.zip](https://github.com/merbanan/rtl_433/files/2230400/sample_packets.zip)

zuckschwerdt, 2018-07-26:
Your sensor actually transmits a whole byte more information. The layout of the bits might be different thus.
Change the lines to:
```
	bitpos = bitbuffer_manchester_decode(bitbuffer, row, bitpos, &databits, 41);
	bitbuffer_print(&databits);
```
You should see lines of 40 bits (the Apollo has 32 bits).
Report back with those lines and what readings you see in the monitor display.

vuokko, 2018-07-26:
I run it through my test files.
What I did, was magnetic contact, few times up and down, turn the leak detector upside down.

`for i in ./*.cu8 ; do ../rtl_433/build/src/rtl_433 -q  -r $i; done >quicktest.txt 2>&1
[quicktest.txt](https://github.com/merbanan/rtl_433/files/2233544/quicktest.txt)
`

zuckschwerdt, 2018-07-27:
What manufacturer/distributor is your `TEK-377 E` from?

The additional byte is always zero, some kind of post-amble. This should work:
```
	bitpos = bitbuffer_manchester_decode(bitbuffer, row, bitpos, &databits, 41);
	if (databits.bits_per_row[0] < 32 || databits.bits_per_row[0] > 40
			|| (databits.bb[0][4] & 0xfe) != 0)
		return 0;
```

vuokko, 2018-07-28:
My sensor is Motoplast mSonic. The user manual (in Finnish) http://www.motoplast.fi/ruubikcms/download.php?f=mSonic_kaytto_ja_asennusohje.pdf has great hints. Its subject is TEK377E and manufacturer is Tekelek, Ireland. And on the PCB there is also print tekelek. I had to reassemble the sender, as it was assembled loosely and rattled.

I just got first live signal captured with your newest patch and old capture files also tell me that my tank is empty 128 cm when real distance is 126.5 cm. So OK by me.

zuckschwerdt, 2018-07-28:
Thanks, fixed.

closed 2018-07-28 by zuckschwerdt