merbanan/rtl_433 #777: Windows build description

Mindavi, 2018-07-27:
I've written a small guide on how to get building in Windows working, but I'm not sure where to put it. I've understood building for Windows isn't officially supported, but I thought it still might be useful to have it somewhere.

This is what I wrote:

# Building rtl_433 for windows

you'll need librtlsdr and libusb, they are dependencies for rtl_433
libusb has prebuilt binaries for windows, 
librtlsdr needs to be built (at least for the latest version, some binaries seem to be floating around)

## librtlsdr

taken and adapted from here: https://www.onetransistor.eu/2017/03/compile-librtlsdr-windows-mingw.html

* install mingw-w64 (https://mingw-w64.org/) and cmake (https://cmake.org/)
    * it's easiest if you select the option to include cmake in your path, otherwise you'll need to do this manually
* download the libusb binaries from https://sourceforge.net/projects/libusb/files/libusb-1.0/ or from https://libusb.info/
    * take the latest release and then download the .7z file, the other file contains the sources (or 'windows binaries' on the .info website)
* extract the archive and open the extracted folder
* copy the contents of the include folder to `<mingw_installation_folder>/include`
* copy the `mingw64/dll/libusb-1.0.dll.a` file to `<mingw_installation_folder>/lib
* copy the `mingw64/dll/libusb-1.0.dll` file to `<mingw_installation_folder>/bin`
* download the source code of librtlsdr https://github.com/steve-m/librtlsdr
* go into the librtlsdr folder
* open cmakelists.txt with an editor that knows unix line endings
* go to `# Find build dependencies` (around line 65) and comment/remove the line with `find_package(Threads)`
* add the following lines instead:

```
SET(CMAKE_THREAD_LIBS_INIT "-lpthread")
SET(CMAKE_HAVE_THREADS_LIBRARY 1)
SET(THREADS_FOUND TRUE)
```

* go into the cmake/modules folder and open FindLibUSB.cmake with a text editor
* find the lines with the following text in them

```
/usr/include/libusb-1.0
/usr/include
/usr/local/include
```

* add some extra lines to point to the mingw include folder where you extracted libusb-1.0, making it look like this
    * take note of the "" around the folder names, these are needed when there are spaces in the folder name
    * you'll need to find out the exact paths for your system

```
/usr/include/libusb-1.0
/usr/include
/usr/local/include
"C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/include"
"C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/include/libusb-1.0"
```

* open a mingw terminal in the librtlsdr folder
* create build folder and go into it: `mkdir build && cd build`
* generate makefiles for mingw: `cmake -G "MinGW Makefiles" ..`
* build the librtlsdr library: `mingw32-make`

## rtl_433

* clone the rtl_433 repository and cd into it
* create a build folder and go into it: `mkdir build && cd build`
* run `cmake -G "MinGW Makefiles" .. ` in the build directory
* run cmake-gui (this is easiest)
* set the source (the rtl_433 source code directory) and the build directory (one might create a build directory in the source directory)
* click configure
* select the grouped and advanced tickboxes
* go into the librtlsdr config group
* point the `LIBRTLSDR_INCLUDE_DIRS` to the include folder of the librtlsdr source
* point the `LIBRTLSDR_LIBRARIES` to the `librtlsdr.dll.a` file in the <librtlsdr_source>/build/src folder
    * that's the one you've built earlier
* edit `CMAKE_C_STANDARD_LIBRARIES` to include `-lws2_32` (this is for winsocks2)
* start a mingw terminal and run `mingw32-make` to build
    * when something in the tests folder doesn't build, you can disable it by commenting out `add_subdirectory(tests)` in the cmakelists.txt file in the source folder of rtl_433
* rtl_433.exe should be built now
* you need to place it in the same folder as librtlsdr.dll and libusb-1.0.dll (you should have seen both of them by now)
* good luck!

zuckschwerdt, 2018-07-27:
None of the active developers use windows, so we can't help out on that. But builds on Win32 with Mingw or Visual C should be possible without too many problems.
We should perhaps look into the problems you discovered with the Threads lib and others.

Not sure were to put this helpful guide. Something like a community wiki with recipies is needed. Maybe a project or gh-pages branch somewhere?

rct, 2018-07-27:
There are windows users, and there is even a guy who occasionally posts windows binaries on this blog.  And I've seen stuff from people who download and use his binaries.

So, if this was in a PR with a file name like `Build-Windows.md` I'd recommend merging it.  It would be great if someone else with Windows experience could check it.  But even if it is has errors and gets merged, people can always file issues and PRs and we'd have some progress.

winterrace, 2018-08-08:
Feel free to test and comment on my win64 port I just uploaded at https://github.com/winterrace/rtl_433_win

Besides four minor compatibility fixes it contains a Visual Studio project in the vs15 subdirectory
 (should compile on VS2015 or newer, see readme.md for details).

From my experience and further tests with another libusb-based project, such native compilations are better performing than binaries geneated using Cygwin/MinGW. Didn't perform similar tests for rtl_433, though...

zuckschwerdt, 2018-08-08:
Can you put your commits in a branch for easier comparison and PR?

Is it an option to use the  `ClCompile Include` with a wildcard? I fear that three build systems will be complicated to keep in sync.
VS2015 isn't C99 compliant? We could introduce some macros to handle `restrict` perhaps. S.a. https://en.wikipedia.org/wiki/Restrict
Is `static const` a problem or `const` itself? I guess we don't need static there anyway.
Would `uint16_t const` work? It's prefered anyway.
Good catch on the need to pack the struct there!

zuckschwerdt, 2018-08-08:
Try `git checkout -b feat-vs2015 ; git push` (it may print a push command you need to run).
If you want to clean the master branch after that use `git checkout master ; git reset --hard b49d299 ; git push -f ; git checkout feat-vs2015`. (`reset` might be dangerous if you don't known how to recover, so always make a backup before.)

Something like
```
#ifdef VS2015_OR_SUCH
#define restrict __restrict
#endif
```
might work if there is no restrict keyword in VS2015

The line you mention is an error, one can't generally initialize a const from another const. I'll fix that.
(The *access* here is constant, the *value* itself might strictly not be.)

zuckschwerdt, 2018-08-09:
A `const` can't be written to, but it's technically still a variable and thus can't be used as initializer. Your fix is correct. The compiler will optimize all that away anyway, no matter if it's macros, const, or plain vars.

I'd like to keep the `restrict` type qualifier though. If some compiler is lacking that just `ifdef` a replacement like you did. It may be "dirty" but so is lack of support for it ;)

Open a PR from your feat-vs2015 against our master so we can merge this.

Mindavi, 2018-08-09:
Be wary that the RelWithDebInfo linked in the @winterrace repo is a version from 2014, not the newest version of the rtl_sdr library.
These are (about) the changes that have occured between the build and now.
It seems like the build is a 'nightly' from the 23rd of January, but I'm not totally sure about that.
https://github.com/osmocom/rtl-sdr/compare/v0.5.2...master

It seems like it aren't that many fixed, but might be relevant for some use cases.

winterrace, 2018-08-09:
@Mindavi: Thanks for pointing that out, I'll try the newest version in the next few days. If all works well, I'll update the readme and the binaries - there should be no implications to further files...

@zuckschwerdt: thanks. PR will follow soon.

winterrace, 2018-09-03:
@Mindavi: Today I upgraded the rtl_433 compilation instructions and the pre-compiled release build to the recent librtlsdr 0.6.0 release (https://github.com/winterrace/rtl_433_win)
My new Win64 project for librtlsdr 0.6.0 can be found here: https://github.com/winterrace/librtlsdr

closed 2018-09-17 by Mindavi