merbanan/rtl_433 #780: Ambient Weather TX-8365: need help identifying coding

roger-, 2018-07-29:
Hi,

I'm trying to decode this sensor: https://www.ambientweather.com/amrc8500.html
FCC: https://fccid.io/OQH-000000-13-01

Output is temperature and battery status only, and it supports one of three different 433 MHz channels.

rtl_433 doesn't seem to recognize it, so I captured this recording (with `-a -t` and temperature around 20 C): https://files.fm/u/xef593fx

Output of `-A`:
```
Test mode active. Reading samples from file: g001_433.92M_250k.cu8
Detected OOK package	@ @0.000000s
Analyzing pulses...
Total count:   75,  width: 90339		(361.4 ms)
Pulse width distribution:
 [ 0] count:   74,  width:   492 [490;495]	(1968 us)
 [ 1] count:    1,  width:   979 [979;979]	(3916 us)
Gap width distribution:
 [ 0] count:   39,  width:   483 [482;486]	(1932 us)
 [ 1] count:   35,  width:   972 [969;974]	(3888 us)
Pulse period distribution:
 [ 0] count:   39,  width:   976 [974;980]	(3904 us)
 [ 1] count:   34,  width:  1464 [1462;1467]	(5856 us)
 [ 2] count:    1,  width:  1951 [1951;1951]	(7804 us)
Level estimates [high, low]:  15994,     42
Frequency offsets [F1, F2]:    2499,      0	(+9.5 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_limit: 492, long_limit: 0, reset_limit: 975, sync_width: 0
pulse_demod_manchester_zerobit(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] {76} 10 00 00 00 00 00 00 00 00 00 
```

Output is OOK, but the data doesn't look correct. I looked at the samples in Audacity and I'm not sure what the coding is. Any ideas?

zuckschwerdt, 2018-07-29:
Try http://triq.net/iqs to visualize the signal. Could be PPM or (rather unlikely) PCM. I.e.
```
rtl_433 -q -F json -R 0 -X 'tx8365:OOK_PPM_RAW:3000:5000:6000' -r g001_433.92M_250k.cu8

{74}63a45c983c5ba367e00
```
or
```
rtl_433 -q -F json -R 0 -X 'tx8365:OOK_PCM:1968:3888:6000' -r g001_433.92M_250k.cu8

{106}f56ef6aeebf55ed6adeb5d557f8
```
Note that "channel" is very likely not a physical property of the signal but some bits. Try changing the channel and observer which bits change.

roger-, 2018-08-03:
Thanks, will have to do more testing because PPM and PCM decoding only gave a couple unique codes over several hours and I would have expected more variation.

merbanan, 2018-10-13:
12bits give 0c9 -> dec -> 194 -> /10 -> 19.4

merbanan, 2018-10-13:
Try that with observing the temperature. The nibble after the 4 is changing between 3,7,b and might contain the channel bits. Anyway the output is very inconsistent and the analyzer guesses Manchester coding so I think the output is incorrect. Better logs and sample signals are needed.

roger-, 2018-10-13:
I believe it's PPM as the pulse widths are constant, but as you mentioned the data is inconsistent when I change the channel.

Thanks, I'll try using the receiver and log the temperature.

merbanan, 2018-10-13:
Send some more samples also, preferably a PR to the tests repository.

roger-, 2018-10-28:
The channel numbers in my previous message may have been incorrect as I didn't reset the transmitter after changing it.

Add some data here: https://github.com/merbanan/rtl_433_tests/pull/243

zuckschwerdt, 2018-10-28:
[Here is a BitBench](http://triq.net/bitbench?c=%5BCH1_77.7F%5D%20%7B74%7D63a40c953c5bf36ae80&c=%5BCH1_78.2F%5D%20%7B74%7D63a40c95fc5bf36a1fc&c=%5BCH1_80.2F%5D%20%7B74%7D23a40c9a3c5bf365d54&c=%5BCH1_82.4F%5D%20%7B74%7D63a40ca03c5bf35fee4&c=%5BCH2_84.5F%5D%20%7B74%7D23a93ca47c56c35ba00&c=%5BCH2_85.2F%5D%20%7B74%7D23893ca5bc76c35a6f4&c=%5BCH2_85.6F%5D%20%7B74%7D23893ca63c76c359e44&c=%5BCH2_86.1F%5D%20%7B74%7D23893cc07c76c33fa00&c=%5BCH2_82.2F%5D%20%7B74%7D23a93c9e7c56c3618b0&c=%5BCH3_83.3F%5D%20%7B74%7D23ad84a17c527b5e800&f=16h%20) with your example data loaded to play around.

roger-, 2018-10-28:
That's very useful, thanks.

Looks like there's at least a 16 bit word that's inverted and repeated, and I can clearly see the bits for the channel number.

zuckschwerdt, 2018-10-28:
I guess I already found [Channel, ID and Temp](http://triq.net/bitbench?c=%5BCH1_77.7F%5D%20%7B74%7D63a40c953c5bf36ae80&c=%5BCH1_78.2F%5D%20%7B74%7D63a40c95fc5bf36a1fc&c=%5BCH1_80.2F%5D%20%7B74%7D23a40c9a3c5bf365d54&c=%5BCH1_82.4F%5D%20%7B74%7D63a40ca03c5bf35fee4&c=%5BCH2_84.5F%5D%20%7B74%7D23a93ca47c56c35ba00&c=%5BCH2_85.2F%5D%20%7B74%7D23893ca5bc76c35a6f4&c=%5BCH2_85.6F%5D%20%7B74%7D23893ca63c76c359e44&c=%5BCH2_86.1F%5D%20%7B74%7D23893cc07c76c33fa00&c=%5BCH2_82.2F%5D%20%7B74%7D23a93c9e7c56c3618b0&c=%5BCH3_83.3F%5D%20%7B74%7D23ad84a17c527b5e800&f=4b%204b%20%7C%204b%20CH2d%20ID%3A8b%20%7C%20TEMP_C%3A4h4h.4h%20%7C%204b%20%7C%204b%204b%20%7C%204b%204b%20%7C%204b%204b%20%7C%204b%204b%20%7C%204b)

zuckschwerdt, 2018-10-28:
Yes you are right the values are also [present inverted](http://triq.net/bitbench?c=%5BCH1_77.7F%5D%20%7B74%7D63a40c953c5bf36ae80&c=%5BCH1_78.2F%5D%20%7B74%7D63a40c95fc5bf36a1fc&c=%5BCH1_80.2F%5D%20%7B74%7D23a40c9a3c5bf365d54&c=%5BCH1_82.4F%5D%20%7B74%7D63a40ca03c5bf35fee4&c=%5BCH2_84.5F%5D%20%7B74%7D23a93ca47c56c35ba00&c=%5BCH2_85.2F%5D%20%7B74%7D23893ca5bc76c35a6f4&c=%5BCH2_85.6F%5D%20%7B74%7D23893ca63c76c359e44&c=%5BCH2_86.1F%5D%20%7B74%7D23893cc07c76c33fa00&c=%5BCH2_82.2F%5D%20%7B74%7D23a93c9e7c56c3618b0&c=%5BCH3_83.3F%5D%20%7B74%7D23ad84a17c527b5e800&f=4b%204b%20%7C%204b%20CH2d%20ID%3A8b%20%7C%20TEMP_C%3A4h4h.4h%20%7C%204b%20%7C%204b%204b%20%7C%20ID%3A~4b~4b%20%7C%20TEMP_C%3A~4h~4h.~4h%7C%204b%204bb%20%7C%204b).

roger-, 2018-10-28:
[With the inverted channels parsed](http://triq.net/bitbench?c=%5BCH1_77.7F%5D%20%7B74%7D63a40c953c5bf36ae80&c=%5BCH1_78.2F%5D%20%7B74%7D63a40c95fc5bf36a1fc&c=%5BCH1_80.2F%5D%20%7B74%7D23a40c9a3c5bf365d54&c=%5BCH1_82.4F%5D%20%7B74%7D63a40ca03c5bf35fee4&c=%5BCH2_84.5F%5D%20%7B74%7D23a93ca47c56c35ba00&c=%5BCH2_85.2F%5D%20%7B74%7D23893ca5bc76c35a6f4&c=%5BCH2_85.6F%5D%20%7B74%7D23893ca63c76c359e44&c=%5BCH2_86.1F%5D%20%7B74%7D23893cc07c76c33fa00&c=%5BCH2_82.2F%5D%20%7B74%7D23a93c9e7c56c3618b0&c=%5BCH3_83.3F%5D%20%7B74%7D23ad84a17c527b5e800&f=4b%204b%20%7C%204b%20CH%3A2d%20ID%3A8b%20%7C%20TEMP_C%3A4h4h.4h%20%7C%204b%20%7C%204b%202b%20CH%3A~2d%20%7C%20ID%3A~4b~4b%20%7C%20TEMP_C%3A~4h~4h.~4h%7C%204b%204bb%20%7C%204b)

What is the ID about? Shouldn't it be the same for all the messages?

roger-, 2018-10-28:
Makes sense.

[Rolling count added](http://triq.net/bitbench?c=%5BCH1_77.7F%5D%20%7B74%7D63a40c953c5bf36ae80&c=%5BCH1_78.2F%5D%20%7B74%7D63a40c95fc5bf36a1fc&c=%5BCH1_80.2F%5D%20%7B74%7D23a40c9a3c5bf365d54&c=%5BCH1_82.4F%5D%20%7B74%7D63a40ca03c5bf35fee4&c=%5BCH2_84.5F%5D%20%7B74%7D23a93ca47c56c35ba00&c=%5BCH2_85.2F%5D%20%7B74%7D23893ca5bc76c35a6f4&c=%5BCH2_85.6F%5D%20%7B74%7D23893ca63c76c359e44&c=%5BCH2_86.1F%5D%20%7B74%7D23893cc07c76c33fa00&c=%5BCH2_82.2F%5D%20%7B74%7D23a93c9e7c56c3618b0&c=%5BCH3_83.3F%5D%20%7B74%7D23ad84a17c527b5e800&f=1b%20CNT%3A1d%202b%20%7C%204b%20%7C%204b%20CH%3A2d%20ID%3A8b%20%7C%20TEMP_C%3A4h4h.4h%20%7C%204b%20%7C%204b%202b%20CH%3A~2d%20%7C%20ID%3A~4b%20~4b%20%7C%20TEMP_C%3A~4h~4h.~4h%7C%20CRC%3F%3A8b%20)

roger-, 2018-10-28:
[Added 2 unknown bits that also get inverted and repeated](http://triq.net/bitbench?c=%5BCH1_77.7F%5D%20%7B74%7D63a40c953c5bf36ae80&c=%5BCH1_78.2F%5D%20%7B74%7D63a40c95fc5bf36a1fc&c=%5BCH1_80.2F%5D%20%7B74%7D23a40c9a3c5bf365d54&c=%5BCH1_82.4F%5D%20%7B74%7D63a40ca03c5bf35fee4&c=%5BCH2_84.5F%5D%20%7B74%7D23a93ca47c56c35ba00&c=%5BCH2_85.2F%5D%20%7B74%7D23893ca5bc76c35a6f4&c=%5BCH2_85.6F%5D%20%7B74%7D23893ca63c76c359e44&c=%5BCH2_86.1F%5D%20%7B74%7D23893cc07c76c33fa00&c=%5BCH2_82.2F%5D%20%7B74%7D23a93c9e7c56c3618b0&c=%5BCH3_83.3F%5D%20%7B74%7D23ad84a17c527b5e800&f=1b%20CNT%3A1d%202b%20%7C%204b%20%7C%202b%20UNKWN%3A2b%20CH%3A2d%20ID%3A8b%20%7C%20TEMP_C%3A4h4h.4h%20%7C%204b%20%7C%204b%20UNKWN%3A~2b%20CH%3A~2d%20%7C%20ID%3A~4b%20~4b%20%7C%20TEMP_C%3A~4h~4h.~4h%7C%20CRC%3F%3A8b%20).

roger-, 2018-10-28:
I think there's a chance the unknown field could be humidity. The decimal values are 56-58 and the FCC report mentions it being a humidity and temperature sensor (although the actual product literature says only temperature). Also the [FCC internal pictures](https://fccid.io/OQH-000000-13-01/Internal-Photos/Internal-Photos-1-1953407) seem to show a restive humidity sensor. Maybe it wasn't deemed accurate enough to be used by the receiver manufacturer to be used in any product, as they don't list any that support humidity. If it is humidity, then I'm not sure how may bits it uses (I assigned 9 but could be 7).

Did a reset and the ID changed, so that seems correct.

Also stuck it in the freezer for a bit [added some data](http://triq.net/bitbench?c=%5BCH1_77.7F%5D%20%7B74%7D63a40c953c5bf36ae80&c=%5BCH1_78.2F%5D%20%7B74%7D63a40c95fc5bf36a1fc&c=%5BCH1_80.2F%5D%20%7B74%7D23a40c9a3c5bf365d54&c=%5BCH1_82.4F%5D%20%7B74%7D63a40ca03c5bf35fee4&c=%5BCH2_84.5F%5D%20%7B74%7D23a93ca47c56c35ba00&c=%5BCH2_85.2F%5D%20%7B74%7D23893ca5bc76c35a6f4&c=%5BCH2_85.6F%5D%20%7B74%7D23893ca63c76c359e44&c=%5BCH2_86.1F%5D%20%7B74%7D23893cc07c76c33fa00&c=%5BCH2_82.2F%5D%20%7B74%7D23a93c9e7c56c3618b0&c=%5BCH3_83.3F%5D%20%7B74%7D23ad84a17c527b5e800&c=%5BCH1_81.1F%5D%20%7B74%7D63a5209cfc5adf63000&c=%5BCH1%5D%20%7B74%7D63a5205c3c5adfa3d00&c=%5Bnegative%5D%7B74%7D23a72015bc58dfea674&c=%5Bnegative%5D%7B74%7D63a720143c58dfebfc4&c=%7B74%7D63a72005fc58dffa300&f=1b%20CNT%3A1d%201b%20%20%7C%20HUM%3F%3A9d%20CH%3A2d%20ID%3A8b%20%7C%20TEMP_C%3A4h4h.4h%20%7C%201b%20%7C%20HUM%3F%3A~9d%20CH%3A~2d%20%7C%20ID%3A~4b%20~4b%20%7C%20TEMP_C%3A~4h~4h.~4h%7C%20CRC%3F%3A8b%20). Not sure where the sign bit is though.