merbanan/rtl_433 #781: Ford TPMS: Values missing

UhtredTheBold, 2018-07-30:
Hi,

I'm trying to read the tire pressure from my Ford and I have picked up the TPMS signal with the built in decoder [89] but I'm not seeing any values being returned:

-r -F json

`{"time" : "@0.524288s", "model" : "Ford", "type" : "TPMS", "id" : "452f111f", "code" : "8d4c05", "mic" : "CHECKSUM"}`

Is this something I'm doing wrong or is the decoder not able to read this particular model of the sensor?

example -a -t capture - [https://1drv.ms/u/s!AilDCtYZODMhisQpypCco1IH_LKpFQ](url)

merbanan, 2018-07-30:
https://github.com/merbanan/rtl_433/blob/master/src/devices/tpms_ford.c

2 bytes in the message payload are unknown. That part has not been decoded because we don't know how to do it.

zuckschwerdt, 2018-07-30:
Perhaps you could take samples under known conditions and annotate the captures? Although that is quite an untertaking to get right…

zuckschwerdt, 2018-07-31:
If you get a chance, please also compare the ID. (could be that we got it backwards or inverted…) Maybe the car unit can display the connected IDs.

mioke77, 2018-11-24:
Any results?
I have these values read:
2018-11-24 16:29:39 :   Ford    :       TPMS    :       45b955ab        :       904606  :       CHECKSUM
2018-11-24 16:29:39 :   Ford    :       TPMS    :       45b955ab        :       904606  :       CHECKSUM
2018-11-24 16:29:40 :   Ford    :       TPMS    :       45b955ab        :       904606  :       CHECKSUM
2018-11-24 16:29:40 :   Ford    :       TPMS    :       45b955ab        :       904606  :       CHECKSUM
2018-11-24 16:31:43 :   Ford    :       TPMS    :       45b9ad88        :       974506  :       CHECKSUM
2018-11-24 16:31:43 :   Ford    :       TPMS    :       45b9ad88        :       974506  :       CHECKSUM
2018-11-24 16:31:43 :   Ford    :       TPMS    :       45b9ad88        :       974506  :       CHECKSUM
2018-11-24 16:31:43 :   Ford    :       TPMS    :       45b9ad88        :       974506  :       CHECKSUM
2018-11-24 16:41:27 :   Ford    :       TPMS    :       45b906f2        :       944506  :       CHECKSUM
2018-11-24 16:41:27 :   Ford    :       TPMS    :       45b906f2        :       944506  :       CHECKSUM
2018-11-24 16:41:27 :   Ford    :       TPMS    :       45b906f2        :       944506  :       CHECKSUM
2018-11-24 16:41:27 :   Ford    :       TPMS    :       45b906f2        :       944506  :       CHECKSUM

pressure in tires are about 2.4bar

zuckschwerdt, 2018-11-24:
The second number there (e.g. 904606, six hex digits) should somehow map to the pressure and temperature. Not sure how though. Would be great if someone could record a deflate/inflate cycle. Could be that the sensor needs to detect wheel spinning to activate sending.

merbanan, 2018-12-07:
As we don't have the protocol information or a good set of samples with corresponding values there is little we can do about it right now. Closing issue.

closed 2018-12-07 by merbanan

Krezalis, 2021-02-18:
Hi. Buy china sensor and want to research protocol. Is different sensor id marked and readed by id. Next step ill try verify code on FORScan and record deflate circle.

# | Label | id | wo tire | tire | Forscan
-- | -- | -- | -- | -- | --
1 | 353b6bd4 | 35bbed6b | 00510a |   |  
2 | 353b6b96 | 35bb599e | 00530a |   |  
3 | 353b6bb8 | 35bb4c9c | 00500a |   |  
4 | 353b6c33 | 35bbee40 | 00510a |   |

zuckschwerdt, 2021-02-18:
The IDs look somewhat the same, but also different. Strange!
[BitBench with IDs](https://triq.net/bitbench?c=353b%206bd4%5B1%5D&c=35bb%20ed6b%5B1%5D&c=&c=353b%206b96%5B2%5D&c=35bb%20599e%5B2%5D&c=&c=353b%206bb8%5B3%5D&c=35bb%204c9c%5B3%5D&c=&c=353b%206c33%5B4%5D&c=35bb%20ee40%5B4%5D&f=8b%20).

Krezalis, 2021-02-19:
Hi, today i learn BdyCM and read id of sensors in module memory (TPM_S_ID_*). FORScan display value in decimal format. Convert 901508459 to HEX: 35 bb ed 6b. rtl_433 work property, TMPS ID is correct. 
Verdict: label ID on 4 sensors ($7,5 per sensor) https://aliexpress.com/item/4000132562673.html not related with ID in IC. 

Pressure received by BdyCM from sensor wo tire: 2.1 kPa. Data from original sensor with normal pressure: 205.5, 214.1, 215.8, 217.5 (FORScan dispaly). I see we have conversation from HEX -> psi  -> kPa. And formula: pressure (psi) = 0.3 + x * 0.25. MAX value 63,8 PSI (~439.9kPa=4.4 bar) It nice to display pressure in console in PSI and Bar (kPa).
[image]

Krezalis, 2021-02-20:
Tested pressure formula in vehicle (activated sensor by EL-50449 and read value by FORScan from BdyCM module) formula is OK.
Assumption about temperature is valid, but i don`t now assured of subtrahend. I think it must be 60, not 64 - no method of verify.
[image]

zuckschwerdt, 2021-02-23:
#1654 manually merged. The output is very likely wrong. We really need a big table of inflation/deflation codes and a range of temperatures (note: tire temperature is *not* ambient temperature).
For our stored samples the temperature values seem way off, even if treated as F.

Krezalis, 2021-05-19:
[image]

At FORSCan 2.3.40 i see difference start coefficient, not 0.3 => 0.45. Need to check on the car where is display on the IPC pressure values.
(psi) = 0.45 + x * 0.25

Also captured data packages and verify with TPM_L_TEMP.
Formula is (°C) = x - 56