merbanan/rtl_433 #785: Kerui D026 not reliable

roger-, 2018-08-03:
I noticed that the Kerui PIR driver is supposed to work with the D026 sensor, but it only seems to detect some events intermittently (and nothing else). 

Testing with two different sensors:
```
rtl_433 -q -F json  -R 68 

{"time" : "2018-08-03 19:00:01", "model" : "Kerui PIR Sensor", "id" : 968314, "data" : "0xe (open)"}
...
{"time" : "2018-08-03 19:00:40", "model" : "Kerui PIR Sensor", "id" : 345738, "data" : "0xe (open)"}
```
It missed all the closed and tamper events and wasn't reliable. However this works for all events:

```
rtl_433 -q -F json  -X "Kerui D026:OOK_PWM:320:960:10000:3000,repeats>=20,bits>=15,rows>=20"

{"time" : "2018-08-03 19:06:31", "model" : "Kerui D026", "count" : 37, "num_rows" : 39, "rows" : [{"len" : 1, "data" : "8"}, {"len" : 3, "data" : "2"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 25, "data" : "1398518"}, {"len" : 16, "data" : "1398"}], "codes" : ["{1}8", "{3}2", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{25}1398518", "{16}1398"]}
```

Looks like the timing is off. For reference:

```
Detected OOK package    @ 2018-08-03 19:07:48
2018-08-03 19:07:48 :   Smoke detector GS 558   :       12081   :       23      :       0
        Raw Code:        d5e637
Analyzing pulses...
Total count:  926,  width: 369847               (1479.4 ms)
Pulse width distribution:
 [ 0] count:    1,  width:   123 [123;123]      ( 492 us)
 [ 1] count:  555,  width:   236 [232;241]      ( 944 us)
 [ 2] count:  370,  width:    79 [76;83]        ( 316 us)
Gap width distribution:
 [ 0] count:   37,  width:  2360 [2357;2363]    (9440 us)
 [ 1] count:  555,  width:    77 [74;82]        ( 308 us)
 [ 2] count:  333,  width:   235 [231;240]      ( 940 us)
Pulse period distribution:
 [ 0] count:   37,  width:  2441 [2437;2483]    (9764 us)
 [ 1] count:  888,  width:   314 [310;320]      (1256 us)
Level estimates [high, low]:  15930,     44
Frequency offsets [F1, F2]:  -23476,      0     (-89.6 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_limit: 79, long_limit: 236, reset_limit: 2364, sync_width: 123
pulse_demod_pwm_precise(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {925} 13 98 54 89 cc 2a 44 e6 15 22 73 0a 91 39 85 48 9c c2 a4 4e 61 52 27 30 a9 13 98 54 89 cc 2a 44 e6 15 22 73 0a 91 39 85 48 9c c2 a4 4e 61 52 27 30 a9 13 98 54 89 cc 2a 44 e6 15 22 73 0a 91 39 85 48 9c c2 a4 4e 61 52 27 30 a9 13 98 54 89 cc 2a 44 e6 15 22 73 0a 91 39 85 48 9c c2 a4 4e 61 52 27 30 a9 13 98 54 89 cc 2a 44 e6 15 22 73 0a 91 39 85 48
``` 

FYI @zuckschwerdt

zuckschwerdt, 2018-08-03:
Interesting. Yes, the timing is rather narrow. It's right on the border here, short is 236 to 396 (ok) and long is 940 to 1100 (your shortest long is 232*4 = 928 µs).
Do you have a PIR sensor, there shouldn't be open, close, or tamper, just `0xa PIR`?

zuckschwerdt, 2018-08-03:
Oh, sorry, you did mention D026 -- that's the door/window sensor :) I'll relax the timings a bit tomorrow.

roger-, 2018-08-03:
Thanks!

BTW it would be nice to have a convention for the states of various sensors (e.g. `open` instead of `0xe (open)`)

zuckschwerdt, 2018-08-04:
Should be fixed with 9a76ea3
The naming of events will be changed too, someday.

closed 2018-08-06 by zuckschwerdt

rushidesai, 2020-09-14:
I'm finding the decoding of my Kerui D026 to also be quite inconsistent. I'm only able to decode 1 in every 4 signals for tamper, open and close. I'm quite unfamiliar with rtl_433 and how to read the analyzer output. Can provide any data you want if you tell me what command line options to use.

eg>

```
Detected OOK package    2020-09-13 23:28:06
Analyzing pulses...
Total count:  923,  width: 1519.75 ms           (379937 S)
Pulse width distribution:
 [ 0] count:  442,  width:  328 us [312;484]    (  82 S)
 [ 1] count:  478,  width:  964 us [952;988]    ( 241 S)
 [ 2] count:    1,  width: 8516 us [8516;8516]  (2129 S)
 [ 3] count:    1,  width: 2152 us [2152;2152]  ( 538 S)
 [ 4] count:    1,  width: 11804 us [11804;11804]       (2951 S)
Gap width distribution:
 [ 0] count:   37,  width: 9544 us [9532;9556]  (2386 S)
 [ 1] count:  478,  width:  308 us [284;324]    (  77 S)
 [ 2] count:  404,  width:  940 us [880;956]    ( 235 S)
 [ 3] count:    1,  width:  472 us [472;472]    ( 118 S)
 [ 4] count:    1,  width:  124 us [124;124]    (  31 S)
 [ 5] count:    1,  width: 7200 us [7200;7200]  (1800 S)
Pulse period distribution:
 [ 0] count:   39,  width: 9828 us [8640;10036] (2457 S)
 [ 1] count:  881,  width: 1272 us [1264;1288]  ( 318 S)
 [ 2] count:    1,  width:  796 us [796;796]    ( 199 S)
 [ 3] count:    1,  width: 12736 us [12736;12736]       (3184 S)
Level estimates [high, low]:   3325,    316
RSSI: -6.9 dB SNR: 10.2 dB Noise: -17.1 dB
Frequency offsets [F1, F2]:    3069,      0     (+11.7 kHz, +0.0 kHz)
Guessing modulation: No clue...
```

```
Detected OOK package    2020-09-13 23:28:18
Analyzing pulses...
Total count:  183,  width: 122.88 ms            (30721 S)
Pulse width distribution:
 [ 0] count:   12,  width:  624 us [624;636]    ( 156 S)
 [ 1] count:   69,  width:  416 us [412;420]    ( 104 S)
 [ 2] count:  102,  width:  228 us [224;236]    (  57 S)
Gap width distribution:
 [ 0] count:   12,  width:  588 us [584;592]    ( 147 S)
 [ 1] count:   69,  width:  196 us [196;200]    (  49 S)
 [ 2] count:   99,  width:  384 us [380;408]    (  96 S)
 [ 3] count:    2,  width: 2188 us [2188;2188]  ( 547 S)
Pulse period distribution:
 [ 0] count:   12,  width: 1212 us [1208;1224]  ( 303 S)
 [ 1] count:  168,  width:  612 us [612;640]    ( 153 S)
 [ 2] count:    2,  width: 2416 us [2416;2416]  ( 604 S)
Level estimates [high, low]:  15907,    241
RSSI: -0.1 dB SNR: 18.2 dB Noise: -18.3 dB
Frequency offsets [F1, F2]:   17914,      0     (+68.3 kHz, +0.0 k
Hz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 228, long_width: 416, rese
t_limit: 2192, sync_width: 624
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=228,l=416,r=2192,g=
0,t=0,y=624'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 3
[00] {57} 33 86 7b c6 f6 dd d2 80
[01] {57} 33 86 7b c6 f6 dd d2 80
[02] {57} 33 86 7b c6 f6 dd d2 80
```

zuckschwerdt, 2020-09-14:
You can see live raw data with `rtl_433 -X 'n=Kerui-Raw,m=OOK_PWM,s=420,l=960,g=1100,r=9900'` take note which packets are received but don't decode. Then just copy the `codes` lines here, thanks!

rushidesai, 2020-09-14:
Just curious why you suggested `s=420` in the cmdline when the code in 9a76ea3 uses `.short_limit   = 320`. Anyways, this is what I get following your instructions. Almost seems like the not-detected cases are where some data is just missing?

### tamper
Detected:
* codes     : {1}8, {3}2, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {16}fb24

Not detected:
* codes     : {26}4d926a4, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {2}c
* codes     : {14}26a4, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {25}9b24d48, {22}9b24d4

### close
* codes     : {1}8, {19}c934e, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {9}9ba

Not detected:
* codes     : {14}26c4, {25}9b24d88, {25}9b24d88, {13}9b20
* codes     : {1}8, {8}a3, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {23}9b2622, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b24d88, {25}9b7cd88

### open

Detected:
* codes     : {1}8, {12}9b1, {25}9b24d18, {23}9b24d0, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {19}9b24c, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {17}9b24d

Not detected:
* codes     : {16}21a3, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18, {25}9b24d18

zuckschwerdt, 2020-09-15:
We currently use 420 µs: https://github.com/merbanan/rtl_433/blob/master/src/devices/kerui.c#L105 -- timing observed is 304 - 560 µs, 420 will cover that well.
We require at least 9 identical codes with length 25. At a glance the "not detected" transmissions should be ok. Strange.
Can you capture (`-S unknown`) such a transmission and upload here as zip?

y2kdread, 2021-03-11:
Was this issue ever resolved? Ordered some of these recently and still see the problem.

y2kdread, 2021-06-29:
@zuckschwerdt - is there any way this can be reopened to try and get the issue resolved?

roger-, 2021-06-29:
FYI I gave up on these sensors. The battery life is terrible (months vs 5 years for others, like WS4975W), they're fairly large and I don't remember if I got past all the decoding issues.

zuckschwerdt, 2021-06-29:
The timing currently is 420+/-160(260 to 580) and 960+/-160 (800 to 1120) -- this should everything seen well.
We also require 9 of the expected 25 packets. You need to debug with -A to see if e.g. timings are still off.

A flex decoder to play around with would be:
`rtl_433 -X "n=Kerui-D026,m=OOK_PWM,s=320,l=960,t=160,g=1100,r=9900,repeats>=9,bits>=25"`

y2kdread, 2021-11-22:
@zuckschwerdt - sorry for the delay in getting back to you.

Using that decoder, things do work a bit better but it still does miss some events. I don't know much about the specifics of this command so I am not sure how to tweak it. What do I need to provide you with to try and tweak things?

y2kdread, 2021-11-22:
@roger- what device protocol does the WS4975W sensor use?

roger-, 2021-11-23:
@y2kdread uses the dsc decoder, which works out of the box. Highly recommended, better than Zigbee too.

zuckschwerdt, 2021-11-23:
Grab a few samples and upload here as zip. See https://triq.org/rtl_433/ANALYZE.html#grab-a-sample

y2kdread, 2021-11-23:
> Grab a few samples and upload here as zip. See https://triq.org/rtl_433/ANALYZE.html#grab-a-sample

Please find the attached logs. Let me know if these are what you need.

[logs.zip](https://github.com/merbanan/rtl_433/files/7589378/logs.zip)

Thanks!