merbanan/rtl_433 #794: Pointing towards a pointless flex_device pointer usage (bug)

winterrace, 2018-08-22:
Here's a first report from my side about an suspected bug (I spotted some suspicius constellations in other places, so there might follow some more...)

**Affected functionality:**
Param "-X" creates a flex device which is nicely registered afterwards. At the same time, a pointer to it is written to the global variable "flex_device".
This global variable is used by the "-F" treatment to pass it to "add_csv_output" and "determine_csv_fields" functions, respectively.
There it is used to alloc mem for storing all field names later used by the CSV output module.

**Problem 1:**
If the command line contains -F before -X, the passed global variable will still contain a NULL pointer and the flex device's field names will be missed.

**Problem 2:**
"-X" might be used multiple times to register multiple flex devices. Each flex device pointer stored in the global variable will overwrite the previous one.
On "-F" treatment only the last flex device pointer will be passed. Consequently, not all field names of flex devices will be processed.

**Proposed solutions:**
Extending the global flex_device pointer to an array of some maximum capacity seems a bad idea to me. It is not very flexible and the complexity of determine_csv_fields will be increased further.

I'd prefer to remove the global variable flex_device and extend the register_protocol function (which is called in both cases) to also copy r_device's field name pointers into a new respective member of the dm_state struct.
This way determine_csv_fields could gather all field names from a straight iteration of the registered protocol list, no matter if they are regular or flex devices.

However, the calling of add_csv_output would still have to be postponed so that all (flex) protocols are guaranteed to be registered.

**Proceeding...**
While I could offer to implement a fix and place a PR, I expect there are better candidates who are more familiar with the affected parts of the code...

zuckschwerdt, 2018-08-22:
Good catch and thorough analysis. Multiple `-X` shouldn't be a problem as the field names are fixed. But that might change.
I will take a look at this, but with low priority :/

winterrace, 2018-08-22:
I agree that multiple -X usage should be no problem.
What shall be fixed is that all created objects should be available to the determine_csv_fields function so it can extract the field names.
Currently determine_csv_fields function will never see more than 1 flex device object...

winterrace, 2018-11-21:
Am I correct that this is fixed now since the latest updates?

zuckschwerdt, 2018-11-22:
I hope so. The idea was to collect every decoder in the same list (r_devs) and let the CSV determine the fields from there.

closed 2018-12-15 by winterrace