merbanan/rtl_433 #799: Acurite 899 rain gauge

faspina, 2018-08-30:
The Acurite 899 Rain Gauge is the newest model. Does anyone know if it is a working device on rtl_433

merbanan, 2018-12-07:
The 899 model is not referenced in the source code but that does not mean that it is not supported. Someone needs to test it.

closed 2018-12-07 by merbanan

K2AAE, 2019-02-01:
Just tried testing, fails with sensor 0X0426 Ch C, Status 0, Unknown message type 0x30 and repeats three times
I commented the device 10, Acurite 0896 no change. I tried rtl_433 -R 10 and just get the help display I also tried rtl_433 -G and still just get the help display. I can run rtl_433 with -c rtl_433.conf and it runs.

zuckschwerdt, 2019-08-09:
There is now code for the Acurite 899 rain gauge (4c4182e). Please test and report if something doesn't fit (currently assumes a tip is 1mm -- is that correct?).