merbanan/rtl_433 #801: Acurite 3n1 Wind+Temp+Humidity Sensor

vvanders, 2018-09-01:
First off, fantastic work with the project, had this up and running in no time picking off signals from the airwaves.

Looking to contribute Acurite 3n1 to the devices supported list as I just picked one up. It appears to use PWM but I'm not able to determine what set of flags I need to get actual bit data to dump. Looking at the PCM data in Audacity I see a couple things:

1. Data is repeated 2x.
2. There's a preamble that's got a different pulse width from the data. Data appears to be 150us period, 100us for 1, 50us for 0(pulse parser confirms this). However the preamble is 600us High, 150us Low, 150us High x3 then 150us followed by the data. If each 150us block is a character this looks like:
```
XXXXXX===XXX===XXX===XXX===<data>
```
3. I'm able to pick out data bits via Audacity(was able to see wind 3->4 flipping the right binary sets) so I know it's there.

Whenever I run rtl_433 -a -r <file> I get:

```
*** signal_start = 22676, signal_end = 65537
signal_len = 42861,  pulses = 136
Iteration 1. t: 346    min: 73 (134)    max: 619 (2)    delta 890
Iteration 2. t: 346    min: 73 (134)    max: 619 (2)    delta 0
Pulse coding: Short pulse length 73 - Long pulse length 619

Short distance: 46, long distance: 107, packet distance: 155

p_limit: 346
bitbuffer:: Number of rows: 10
[00] { 1} 80                      : 1
[01] { 1} 00                      : 0
[02] { 1} 00                      : 0
[03] { 1} 00                      : 0
[04] {64} 00 00 00 00 00 00 00 00
[05] { 1} 80                      : 1
[06] { 1} 00                      : 0
[07] { 1} 00                      : 0
[08] { 1} 00                      : 0
[09] {64} 00 00 00 00 00 00 00 00
Test mode file issued 1 packets
```

It does seem to detect Acurite when run normally:
```
@0.000000s Acurite 5n1 sensor 0x0D21 Ch A, Status D4, Unknown message type 0x20
@0.000000s Acurite 5n1 sensor 0x0D21 Ch A, Status D4, Unknown message type 0x20
```

However I can't get the raw bits to dump out. I tried a couple custom decoders with '-X' option but still just get zeros for the bit rows.

Any hints on what parameters I'm screwing up here or is it going to be easier just to start cracking open acurite.c?

vvanders, 2018-09-01:
Also have the raw samples up at https://nextcloud.vvanders.com/s/TZBeJXPYyAipSnq if that helps. Happy to spend the effort on my, I think it's mostly just not being familiar with the various decoding options.

zuckschwerdt, 2018-09-01:
I already have a sketch for the device at #720 
But the layout of the fields doesn't seem right. Humidity looks ok, but temperature and wind don't.
Perhaps you can correlate those values to the bits in the packet?

vvanders, 2018-09-01:
Okay, awesome. Happy to grab more captures here later tonight.

For #4 & #5 the values were:

#4:
Temp: 58F
Hum: 84%
Wind: 3mph

#5:
Temp: 58F
Hum: 84%
Wind: 4mph

I can see 3->4 in the tail end of the bit pattern so tha should show where the LSB of wind ends. I'll update as I have more data.

merbanan, 2018-09-01:
Well 99 is max on a 2 digit display so the limit might just be in the presentation layer. And the accuracy might be tuned to the visible range.

vvanders, 2018-09-02:
Okay, another couple samples.

https://nextcloud.vvanders.com/s/pdiQeXaHH3BgBRJ

Temp: 52F
Hum: 83%
Wind: 1mph

https://nextcloud.vvanders.com/s/6xfjpojCrfwmYPN

Temp: 53F
Hum: 81%
Wind: 1mph

Since my sensor is installed outside it's hard to get a good adjustment in temp but getting a couple more data points.

Pulled your branch and been running it here for a bit, wind + humidity seem on the mark. Need to narrow down temp somehow. Since I've got a few datapoints going to poke around and see what I can tell.

vnosovsky, 2018-09-29:
@zuckschwerdt,

I've been running 3n1 from #720 successfully for quite some time now (couple of months I think)
Temperature, humidity and wind are pretty close to what I see on weather station synched with 3n1 sensor. They also close to Acurite 592TXR Temp/Humidity readings.
The only change I made was line 739 in acurite.c should be:

`tempf = acurite_getTemp(bb[4], bb[5]) + 96.7;`

zuckschwerdt, 2018-09-29:
thanks for the feedback. `96.7` sounds somewhat arbitrary, is there any margin so that it be some reasonable rounded value like 96, 98 or maybe 100?

vnosovsky, 2018-09-29:
I played with differ my values until I started getting same readings as on weather station. I guess 97 is the closest round value will do as well. Let me try running it for some time

zuckschwerdt, 2018-09-30:
I don't think `97` is a value a device/protocol designer would choose. Something else is going on.
Did you get a good range of values? Could it be some scaling?

vnosovsky, 2018-09-30:
Good point. I was running it indoor all the time with temp range 20-24 Celsius. Just put it outdoor next to other sensors. Will let you know in couple of days.

RDK-45, 2018-10-03:
Gentlemen.....I've been communicating with Robert Terzi and Christian since earlier this year about decoding the RTL_433 transmissions for an Acurite 3-in-1 sensor, but never via this GitHub system.  Perhaps my comments and analyses have been lost?  Regardless, after a recent note from Robert I have taken the time to organize my findings and data into the attached files:  
- 3-in-1_DataAnalysis.docx  is a WORD document with details my findings
- 3n1Data.xlsx is an Excel file with holds the data used in my analyses.

I have algorithms for temperature, humidity, wind and sensor id.  For several months now, I have been using these algorithms in a modified version of acurite.c in my RTL_433 installation on two of my Raspberry Pi's and I'm happy with the data I'm getting for all of my sensors (5-in-1, 3-in-1, towers and water temperature probe).
Let me know if you have comments or questions....RDK
 
[3n1Data.xlsx](https://github.com/merbanan/rtl_433/files/2440725/3n1Data.xlsx)
[3-in-1_DataAnalysis.docx](https://github.com/merbanan/rtl_433/files/2440726/3-in-1_DataAnalysis.docx)

zuckschwerdt, 2018-10-03:
Very well done research. I'll update the 3n1 with your findings.

(with "Sensor ID “and-ing” byte0 and byte1 … 0x11&0x1e" you really mean concatenation, the `and` operation is masking. So 0x11<<8|0x1e like in your next code example).
The 5n1 id is commented as

    5-n-1 sensor ID is the last 12 bits of byte 0 & 1
    byte 0     | byte 1
    CC RR IIII | IIII IIII

Do you know what the `RR` would be? It's the data you included in the id for the 3n1.
Following your analysis on the channels I'd say channel setting is the same as in 5-n-1, the top 2 bits with

     00 = C
     10 = B
     11 = A

I guess you somehow triggered a power-on reset between channel A and B, maybe?
And the next-to top 2 bits (`RR`) may not be in the ID after all. It's the only difference between B and C setting.

    2099  = 0x0833 = 0000 1000 0011 0011
    578   = 0x0242 = 0000 0010 0100 0010
    12866 = 0x3242 = 0011 0010 0100 0010
    
    0xC8 = 11 00 1000
    0x82 = 10 00 0010
    0x32 = 00 11 0010

RDK-45, 2018-10-03:
Christian….Thanks for your reply and kind words.

 

As I’m not real comfortable working with and manipulating HEX and binary numbers, this was not easy and I got more than one headache during the process.

 

Thanks also for correcting my terminology, and-ing , or-ing, parity bits, etc were pretty foreign concepts  before I stated this project.

 

        5-n-1 sensor ID is the last 12 bits of byte 0 & 1
        byte 0     | byte 1
        CC RR IIII | IIII IIII

Do you know what the RR would be? It's the data you included in the id for the 3n1.

Sorry, I saw that too and did not make a connection.

 

Your hypothesis for the Channel Number makes sense, except that for my reference 3-in-1 (11 1E 60 E1 0F 6F 00 EE) that would say it is on Channel C.  I would have thought it would be on “A”.  It is not easy to get to right now.  Perhaps over the weekend I will climb up to it and change the batteries and check on the channel code.

 

Finally, “I guess you somehow triggered a power-on reset between channel A and B, maybe?”, I’m not sure what you mean???  If you mean that I had to take out the batteries to change channels, then yes.  

 

Thank again….RDK

 

Rob Koller Sr

RDK-45, 2018-10-03:
Christian….That is not my experience with my other Acurite sensors.  I have changed batteries before without getting a new ID.  We’ll see what happens when I get up to my 3-in-1 and change the batteries….Rob

 

Rob Koller Sr

 

Yes, usually removing power from a device randomly chooses a new ID. That's probably why the 12 ID bits changed.

vnosovsky, 2018-10-03:
I have bunch of different Acurite sensors, and only 3n1 changes channel on power reset.

rct, 2018-10-03:
@vnosovsky - That's interesting to know. 

The 5n1, 592TXR town, 6045M lightning, and now 3n1 seem to use fairly similar messages. Those devices have fixed, static IDs and physical channel switches.  The [instructions that I found](https://www.acurite.com/media/manuals/06008-instructions.pdf) show the 3n1 has a physical A-B-C- switch so I would expect the encoding to be the same and wouldn't expect it change at power cycles.  But I could be wrong.

The 5n1 is only 12 bits of ID because there are two bits after the channel id that are incremented for each message repeat.   In other words you can tell whether you are looking at the first, second or third transmission that is part of the burst.   

It's been a while since I looked at the log file @RDK-45 provided but it didn't look like those bits incremented like they did with the 5n1.  So I'd expect the 3n1 to use 14 bits of ID like the 592TXR.    

@zuckschwerdt - FYI - ISTR when I was looking at @RDK-45's log file I seem to recall there was some data that didn't fit, but then when I checked the checksum or parity was wrong.  Also I believe @RDK-45 has an Acurite aculink or access, so he can see the decimal integer ID as Acurite is interpreting it.  I don't know if his device ID is high enough to tell whether the ID is 12 or 14 bits.

rct, 2018-10-03:
@zuckschwerdt  -- actually looking back through some old email I found one of the aculink messages @RDK-45 had sent:

```
mt=3N1x20&sensor=00004382&humidity=44&tempf=89.6&windspeedmph=1&baromin=29.43&battery=normal&rssi=4
```

So at the time his 3n1 was being interpreted by the aculink receiver as having an ID of 4382 (0x111e) so at least Acurite's aculink thinks the ID is > 12 bits.

RDK-45, 2018-10-05:
In my Word document I comment on three 3-in-1 Sensor id's:

For the following sensor byte strings I get:
1.	11 1E 60 60 0F F6 00 F4 – 4382
2.	ED 2B 60 22 90 93 00 BD – 11563
3.	C8 33 60 BD 11 AC 00 D5 – 2099

Those ID's agreed with the web interfaces for the smartHUB units.…..RDK

rct, 2018-10-05:
Ok good - it's straight forward then -- the ID is 14 bits like the 592TXR/Tower and the 6045M lightning.   Channel is the two high order bits.

Just call acurite_txr_getSensorId(bb[0], bb[1])).  

```
// Tower sensor ID is the last 14 bits of byte 0 & 1
// byte 0    | byte 1
// CCII IIII | IIII IIII
//
static uint16_t acurite_txr_getSensorId(uint8_t hibyte, uint8_t lobyte){
    return ((hibyte & 0x3f) << 8) | lobyte;
}
```

RDK-45, 2018-10-06:
Folks.....Christian and Vnosovsky were correct, when removing the batteries to check on the Channel setting, the Sensor ID for this 3-in-1 changed from 4382 to 3650.

ALSO, while I thought this 3-in-1 was set to Channel A, it was actually on Channel C, so the existing 5-in-1 channel decoding works for the 3-in-1....RDK

RDK-45, 2018-10-09:
Regarding the Sensor Id changing, I have changed batteries on both the tower and 5-in-1 sensor units without observing a change in the id.  I use those codes in my Python systems for monitoring sensor data from both the RTL_433 functions and the traffic from the smartHUB.

Thus, it is unique to the 3-in-1 in my experience....RDK

zuckschwerdt, 2018-10-11:
Support merged with #720. Reopen if anything needs tuning.

closed 2018-10-11 by zuckschwerdt

RDK-45, 2018-10-11:
Christian…..I have looked at your modifications and I don’t understand your calculation for tempf.   Should the -108 really be -148?  And I have to go back to my notes to verify if the acurite_getTemp(bb[4], bb[5]) value also needs to be divided by 10?....RDK

Rob Koller Sr

France 33-6-74909740

 <http://www.rdkscorner.com/rdk/> WWW.RDKsCorner.Com

zuckschwerdt, 2018-10-11:
There is already some conversion in [acurite_getTemp()](https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L137). Maybe it should really be [acurite_6045_getTemp()](https://github.com/merbanan/rtl_433/blob/master/src/devices/acurite.c#L350) as mentioned, but that would widen the bits to 0x1F 0x7F instead of the 0x0F 0x7F now. I'm not sure about that?

RDK-45, 2018-10-12:
I have revisited my analysis and the Word document I published here 10 days ago.  The temperature calculation in the above proposed modification (_**tempf = acurite_getTemp(bb[4], bb[5]) - 108; // regression yields (rawtemp-1480)*0.1**_) does NOT correspond to the results of my analysis. At the very least the constant "108" should be "149".

> •	Temperature in degrees F involves using parts of byte4 and byte5:
o	int highbits = (bb[4] & 0x7F) << 7 ;
//   remove parity bit
int lowbits = bb[5] & 0x7F;
//   add results to get rawtemp
int rawtemp = highbits | lowbits;
  here using the code terminology in the acurite.c module
o	Then rawtemp is converted to temperature F using the relationship shown in the below figure.  This figure shows the results of the above data transformation algorithm Decimal(byte4)*128 + decimal(byte5)  yielding rawtemp which was plotted against the data captured from the 3-in-1 via the smartHUB system.  There are 952 points on this plot.  The linear regression fit yield this equation:  TempF = (rawtemp-1480.3)/10.004.
o	I suspect the real relationship used by the developer engineers was 
TempF=(rawtemp-1500)/10, each bit change being 0.1 degrees F.

….RDK

RDK-45, 2018-10-12:
OK, yes that is reasonable, although to split up the formula like that is a bit awkward, IMHO.  

Regarding the choice of bits and bytes, you folks are more expert than I am.  If your decoding works with my data then I have no problems with it....RDK

vnosovsky, 2018-10-14:
@zuckschwerdt, not sure what's going on but your algorithm works fine with 97 for 13 days in temperature range 14-27 degrees Celsius. 
¯\_(ツ)_/¯

RDK-45, 2019-01-24:
Folks….I’ve been using the new Acurite 3-in-1 algorithm now for 5- months.  It works correctly 99+ % of the time. However, every now and again, it throws out a seriously wrong temperature. The humidity and the rest the values appear ok.  

 

I think it has something to do with byte 2 (Battery and Message Type). Normally this byte is “60”.  However, in my original regression data package there was an instance of three readings:

08/21/2018 13:52:35      11 1E E2 95 99 2B 01 6B 11           1E           E2

08/21/2018 13:52:54      11 1E E2 95 99 2B 01 6B 11           1E           E2

08/21/2018 13:53:13      11 1E E2 95 99 2B 01 6B 11           1E           E2

which had byte 2 as “E2” and the associated temperature calculation was out of range. I discarded these reading when I did my regression analysis.

 

I’m wondering if we should modify the 3-in-1 code to check for byte 2 not equal to “60” and discard the reading?  Right now I’m doing something like that when I post process the data.

 

Thanks again for your help getting the Acurite 3-in-1 sensor supported by RTL_433……Rob

 

Rob Koller Sr

France 33-6-74909740

 <http://www.rdkscorner.com/rdk/> WWW.RDKsCorner.Com

zuckschwerdt, 2019-01-24:
Don't we already check this? There is a `message_type == ACURITE_MSGTYPE_WINDSPEED_TEMP_HUMIDITY_3N1` condition with `message_type = bb[2] & 0x3f` and `ACURITE_MSGTYPE_WINDSPEED_TEMP_HUMIDITY_3N1 = 0x20`.
The message_type for b[2]=0xe2 would be 0x22.

RDK-45, 2019-01-24:
Christian….You are probably right.  But, somehow there are occasionally bad temperatures being reported.  I see the bad numbers in my data, but unfortunately I do not have/save the raw data when this happens.  Perhaps I should add a trap for this issue to the code which is using the RTL_443 functions?

 

That won’t happen soon but I will put it on my list….Rob

 

Rob Koller Sr

zuckschwerdt, 2019-01-24:
If you just have a bad value and the good values next to it we could perhaps figure out if it is just a single (flag?) bit.

RDK-45, 2019-03-05:
Christian….My 3-in-1 has been giving bad temperature data off and on for the past few days.  Last evening I was able to capture some raw data records (rtl_433 -R 40 -F json -D &> debug030419.txt) when it was reporting -140F temperature values.  This morning it was working correctly so I captured more records when it was reporting correct temperatures (mid 50’s F).  In both cases the Acurite Dashboard was reporting correct temperatures during those periods.

 

Note the attached files also have outdoor data for a 5-in-1 unit.  The units are about 50 m apart and depending on time of day (morning or evening) one will be in the sunlight but not the other.  The tower unit being also reported in not outside.

 

Perhaps the attached files will help identify the issue??

 

Thanks….RDK

Rob Koller Sr

 

 

 

 

 

Registering protocol [1] "Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning"
Registered 1 out of 100 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
acurite_txr Parity: 10000000
Acurite 5n1 raw msg: CE 42 60 9C 90 30 03 CF
{"time" : "2019-03-04 18:37:38", "model" : "Acurite 3n1 sensor", "sensor_id" : 3650, "channel" : "A", "sequence_num" : 0, "battery" : "OK", "message_type" : 32, "wind_speed_mph" : 3.000, "temperature_F" : -143.200, "humidity" : 28}
acurite_txr Parity: 10000000
Acurite 5n1 raw msg: CE 42 60 9C 90 30 03 CF
{"time" : "2019-03-04 18:37:38", "model" : "Acurite 3n1 sensor", "sensor_id" : 3650, "channel" : "A", "sequence_num" : 0, "battery" : "OK", "message_type" : 32, "wind_speed_mph" : 3.000, "temperature_F" : -143.200, "humidity" : 28}
acurite_txr Parity: 0000001
{"time" : "2019-03-04 18:37:40", "model" : "Acurite tower sensor", "id" : 11824, "sensor_id" : 11824, "channel" : "A", "temperature_C" : 20.300, "humidity" : 52, "battery_low" : 0}
acurite_txr Parity: 0000001
{"time" : "2019-03-04 18:37:40", "model" : "Acurite tower sensor", "id" : 11824, "sensor_id" : 11824, "channel" : "A", "temperature_C" : 20.300, "humidity" : 52, "battery_low" : 0}
acurite_txr Parity: 0000001
{"time" : "2019-03-04 18:37:40", "model" : "Acurite tower sensor", "id" : 11824, "sensor_id" : 11824, "channel" : "A", "temperature_C" : 20.300, "humidity" : 52, "battery_low" : 0}
pulse_demod_pwm_precise(): Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning 
bitbuffer:: Number of rows: 3 
[00] {57} ee 30 44 b4 09 33 52 00 
[01] {57} ee 30 44 b4 09 33 52 00 
[02] {57} ee 30 44 b4 09 33 52 00 
acurite_txr Parity: 0100001
{"time" : "2019-03-04 18:37:41", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 15.800, "humidity" : 31, "battery_low" : 0}
acurite_txr Parity: 0100001
{"time" : "2019-03-04 18:37:41", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 15.800, "humidity" : 31, "battery_low" : 0}
acurite_txr Parity: 0100001
{"time" : "2019-03-04 18:37:41", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 15.800, "humidity" : 31, "battery_low" : 0}
pulse_demod_pwm_precise(): Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning 
bitbuffer:: Number of rows: 3 
[00] {57} fc f2 44 9f 09 06 e0 00 
[01] {57} fc f2 44 9f 09 06 e0 00 
[02] {57} fc f2 44 9f 09 06 e0 00 
acurite_txr Parity: 0000000
[... 561 lines cut ...]
[02] {57} ee 30 44 33 09 b2 50 00 
acurite_txr Parity: 0100000
{"time" : "2019-03-05 07:14:29", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 12.500, "humidity" : 31, "battery_low" : 0}
acurite_txr Parity: 0100000
{"time" : "2019-03-05 07:14:29", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 12.500, "humidity" : 31, "battery_low" : 0}
acurite_txr Parity: 0100000
{"time" : "2019-03-05 07:14:29", "model" : "Acurite tower sensor", "id" : 15602, "sensor_id" : 15602, "channel" : "A", "temperature_C" : 12.500, "humidity" : 31, "battery_low" : 0}
pulse_demod_pwm_precise(): Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning 
bitbuffer:: Number of rows: 3 
[00] {57} fc f2 44 9f 88 65 be 00 
[01] {57} fc f2 44 9f 88 65 be 00 
[02] {57} fc f2 44 9f 88 65 be 00 
acurite_txr Parity: 0000001
{"time" : "2019-03-05 07:14:31", "model" : "Acurite tower sensor", "id" : 2606, "sensor_id" : 2606, "channel" : "C", "temperature_C" : 10.600, "humidity" : 59, "battery_low" : 0}
acurite_txr Parity: 0000001
{"time" : "2019-03-05 07:14:31", "model" : "Acurite tower sensor", "id" : 2606, "sensor_id" : 2606, "channel" : "C", "temperature_C" : 10.600, "humidity" : 59, "battery_low" : 0}
acurite_txr Parity: 0000001
{"time" : "2019-03-05 07:14:31", "model" : "Acurite tower sensor", "id" : 2606, "sensor_id" : 2606, "channel" : "C", "temperature_C" : 10.600, "humidity" : 59, "battery_low" : 0}
pulse_demod_pwm_precise(): Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning 
bitbuffer:: Number of rows: 3 
[00] {57} 0a 2e 44 bb 88 d2 91 00 
[01] {57} 0a 2e 44 bb 88 d2 91 00 
[02] {57} 0a 2e 44 bb 88 d2 91 00 
Signal caught, exiting!
Reattached kernel driver