merbanan/rtl_433 #813: The sensor protocol [20] "Ambient Weather Temperature Sensor" no longer detects the sensors.

Dani22m, 2018-09-27:
My 3 Sensors from Type Ambient Weather F007T and F007TP would be recognized by rtl_433 but not "translated" with the protocoll [20] "Ambient Weather Temperature Sensor". The base station continues to recieve the correct values.

RTL433 recieves the signal bu don't decode it. I recorded the signal with rtl_433 -a -t
The Sensor is on Chanel4 and display the Themperature 23,4°C

[Ambient Wether F007T.tar.gz](https://github.com/merbanan/rtl_433/files/2423482/Ambient.Wether.F007T.tar.gz)

Dani22m, 2018-10-10:
Does anyone have any idea why the sensors are no longer detected?

zuckschwerdt, 2018-10-10:
Thanks for the sample files. I'll take a look.

zuckschwerdt, 2018-10-10:
The 0th byte, which was assumed to be fixed 0x45 is 0x46 in your samples. It might be part of the ID.
I'll push a fix shortly. Maybe we should widen the ID from just byte 1 to be byte 0 and byte 1.

zuckschwerdt, 2018-10-11:
Was this device reporting correctly before or is this a new device? Did it suddenly change from the 0x45 to the 0x46 prefix (i.e. stopped working suddenly), if so do you remember what might have changed? This would help deduce if those bits are status or ID.

zuckschwerdt, 2018-10-11:
"Slowly decreased" hints that the first byte might be a counter or status bits. We'll just ignore it for now.

closed 2018-10-11 by zuckschwerdt