merbanan/rtl_433 #821: Assistance for (another) Toyota TPMS

gcormier, 2018-10-05:
Wasn't sure if the issue should be opened here... 0 open in the other repo so I figured it might be best here.

PR with the information I've captured : https://github.com/merbanan/rtl_433_tests/pull/236

zuckschwerdt, 2018-10-07:
Thanks for the samples. Looks like this sensor has a half-bit width of 100 µs (all other so far have 50 µs).
I'd start by modifying the existing tpms_toyota.c device and see if that works. Then report back and we'll see where the new code fits.

E.g. at https://github.com/merbanan/rtl_433/blob/master/src/devices/tpms_toyota.c#L110

    .short_limit    = 100, // 25 samples @250k
    .long_limit     = 100, // FSK
    .reset_limit    = 250, // Maximum gap size before End Of Message [us].

Gives you (add `bitbuffer_print(bitbuffer);`)

    a.cu8: {144} fc b3 2b 32 ab 4b 2d 55 4d 52 aa d4 cc b4 aa ac b5 2c
    b.cu8: {143} fc cc aa ad 2a b4 d4 b3 4a ad 55 54 cc ca cc cd 2d 2c
    c.cu8: {144} fc cd 55 55 4a b4 d4 b2 b2 ad 55 2b 33 4b 55 4d 54 ac
    d.cu8: {144} fc b3 2b 32 ab 4b 2d 55 52 ad 55 2b 33 4b 55 52 d5 2c

in tpms_toyota_callback().
Which looks correct but has no simple preamble. E.g. the buffers start:

    1111110010110011
    1111110011001100
    1111110011001101
    1111110010110011

So maybe pick a preamble of `11111001` (admitting for a missed first bit-period and neatly fitting in a byte): 0xF9.

I.e.

    static const unsigned char preamble_pattern[2] = {0xf9}; // 8 bits
    …
    // Find a preamble with enough bits after it that it could be a complete packet
    while ((bitpos = bitbuffer_search(bitbuffer, 0, bitpos, (const uint8_t *)&preamble_pattern, 8)) + 134 <=
            bitbuffer->bits_per_row[0]) {
        events += tpms_toyota_decode(bitbuffer, 0, bitpos + 8);
        bitpos += 2;
    }

Now you can look at the DMC-decoded data:

    {67} f3 c2 b4 0c 41 3e a0 69 60
    {67} e0 50 b2 e8 40 3f 9f d5 60
    {67} c0 08 b2 cc 41 3e a0 c2 60
    {67} f3 c2 b4 04 41 3e a0 51 60

And see if it works or it's MC and not DMC, etc.
(best case: the IDs on the sensors match these codes already -- the pressure/temp bytes look about right.)

gcormier, 2018-10-08:
Done a bit of tinkering but not much success so far, although admittadely most of my time is trying to figure out the code and I'm in the dark a bit.

However, I blindly stumbled on this out of sheer luck, which documents most/all of the format of this sensor!

https://github.com/xnk/pacific-tpms

zuckschwerdt, 2018-10-08:
Very good. This confirms our guesses. I'll add the sensor in a few days.

zuckschwerdt, 2018-10-08:
To get a baseline and some checks: is this what you expect from the four tires?

    ID 79e15a0, 223 kPa, 24 C
    ID f028597, 218 kPa, 23 C
    ID e004596, 223 kPa, 25 C
    ID 79e15a0, 223 kPa, 24 C

gcormier, 2018-10-08:
Definitely very close to what I would expect. However there are only 3 unique sensor ID's. I'll try out my OBD reader today to see if it can pull the sensor ID's, but I have a feeling that won't be successful given not only the cheapness of my OBD adapter, but I also don't think it's made available via OBD for some of these Toyota's.

If you can't find a 4th unique, I can go for another drive today and try a different antenna location.

merbanan, 2018-10-08:
30 bit ids? All other have 24 iirc.

zuckschwerdt, 2018-10-08:
28 bits (7 nibbles). There is a corrosponding 7 hex-chars marking on the devices. I'd say it's reasonable.

zuckschwerdt, 2018-10-08:
@gcormier you can use https://github.com/zuckschwerdt/rtl_433/tree/feat-pmv107j
It's a quick hack for testing only.

gcormier, 2018-10-08:
Yup, works great! I tried out the OBD adapter with a few apps and indeed there is no TPMS data. The 4th sensor did show up, and they are all +/- 1 degree, and I didn't see any other sensors during the 15 minute testing window - sat in a concrete underground parking lot, so relatively isolated.

zuckschwerdt, 2018-10-08:
The 4th tire might be `0674e595`, it's recorded in g003_314.98M_250k.cu8

gcormier, 2018-10-08:
Yup, this was what came back:

```
bitbuffer:: Number of rows: 1 
[00] {68} 79 e1 5a 06 20 9f 4f f6 b0 
Realigned: 01 e7 85 68 18 82 7d 3f da
2018-10-08 11:02:16 :	Toyota 	:	TPMS 	:	079e15a0 	:	24 	:	OK 	:	3 	:	OK 	:	223.200 	:	23.000 	:	CRC
bitbuffer:: Number of rows: 1 
[00] {68} e0 04 59 66 20 5f 8f df b0 
Realigned: 03 80 11 65 98 81 7e 3f 7e
2018-10-08 11:02:37 :	Toyota 	:	TPMS 	:	0e004596 	:	24 	:	OK 	:	3 	:	OK 	:	220.720 	:	23.000 	:	CRC
bitbuffer:: Number of rows: 1 
[00] {68} f0 28 59 72 20 1f cf bc 70 
Realigned: 03 c0 a1 65 c8 80 7f 3e f1
2018-10-08 11:02:55 :	Toyota 	:	TPMS 	:	0f028597 	:	8 	:	OK 	:	1 	:	OK 	:	218.240 	:	22.000 	:	CRC
bitbuffer:: Number of rows: 1 
[00] {68} 67 4e 59 56 20 1f d0 25 b0 
Realigned: 01 9d 39 65 58 80 7f 40 96
2018-10-08 11:03:17 :	Toyota 	:	TPMS 	:	0674e595 	:	24 	:	OK 	:	3 	:	OK 	:	218.240 	:	24.000 	:	CRC

```

zuckschwerdt, 2018-10-11:
Added with #825, report back if anyhing needs tuning.

closed 2018-10-11 by zuckschwerdt