merbanan/rtl_433 #823: vcd and sr output seem to be broken under windows after lasts commits

Fra298, 2018-10-10:
First of all, I'm totally noob with this, so maybe I'm doing something wrong. 
Well, I did compile RTL_33 under cygwin and it seems to work ok. But : 
    - when trying to convert a cu8 file to sr, ASK or FSK are always empty under pulseview. It seems the U8:Logic file is filled with spaces only. I tried same process under ubuntu and it works perfectly.

    - when trying to convert cu8 file to vcd, I always got this :

$date 2018-10-10 11:36:46 $end
$version rtl_433 0.1.0 $end
$comment Acquisition at 250.000kHz Hz $end
$timescale 1 us $end
$scope module rtl_433 $end
$var wire 1 / FRAME $end
$var wire 1 ' AM $end
$var wire 1 " FM $end
$upscope $end
$enddefinitions $end
#0 0/ 0' 0

Did not try under ubuntu...

zuckschwerdt, 2018-10-10:
Another user reported garbled VCD files (NULL instead of quotes). There seems to be something subtly wrong.

zuckschwerdt, 2018-10-10:
Can you try the latest commit? I changed all `intmax_t` types, which weren't a good idea anyway.

Fra298, 2018-10-10:
Seems to work ok now! good catch!

zuckschwerdt, 2018-10-10:
Thanks for the feedback. I don't understand what the real problem was, the code seemed to compile fine before. Maybe I'll investigate the semantics of intmax_t on win32 some day ;)

Fra298, 2018-10-11:
Well, I have to reopen the issue...
I just figured out that vcd and sigrok converter is working only with files saved with 250kHz samplerate. When samplerate higher, at least with vcd file, 250k is still reported in metadata and file is empty. When using higher sample rate, if signal is in the 250kHz range of the center frequency, it seems to work.

So far, I'm using RTL_433 the "wrong" way... lurking neighbourgh signals, bnut just to try to figure out everything works as there's no real documentation. But I like to experiment and learn even if my skills in this aera are poor (no coding skills).

zuckschwerdt, 2018-10-11:
There is a third argument to the sigrok-conv.sh does that work?

    Usage: sigrok-conv.sh input.cu8 output.sr [sample rate in kHz]

Fra298, 2018-10-11:
No, not working with sample rate set during opening or converting. I just tried with a signal at 1000kHz sampling rate, signal  is around 400kHz from listening frequency and it's not working.

Here's [the link to the file](http://fra298.servegame.com/dl/g002_433.5M_1000k.cu8). Again, maybe I'm doing something wrong.

zuckschwerdt, 2018-10-11:
The sample rate auto-detection on file read is not wired up completly yet. I've added passing on the sample rate to the converter for now.
But the sample you attached is not usable -- it looks like FSK in the [spectrogram](http://triq.net/iqs) but I guess the sample rate is too low (also the amplitude is not stable -- maybe it's TPMS from a passing car?).

Fra298, 2018-10-11:
Well, it should be possible. I made some more testing and figured out everything was working with "strongs" signals but not with weak ones.

Once again, this is quite a new playground for me and I try to learn. I'm at ease with radio in general (former 11m dx'er), but less with these digital signals. I'll soon try to get a cheap sensor to make better testing and I'll try to figure out how this new world for me is working :)

zuckschwerdt, 2018-10-11:
You can inspect the samples in Audacity (Import Unsigned 8-bit, 2 Channels). Try some samples from the rtl_433_tests repo to get a feel. The ASK/FSK sine should be somewhat visible. You can also try the AM an FM decoder plugins from http://triq.net/articles/sdr-nyquist-plugin-am-decode-iq-signals
The Sigrok Pulseview ouput should match, but it's rather new and might have bugs.

closed 2018-11-08 by zuckschwerdt