merbanan/rtl_433 #829: Doxygen file cannot be used with autotools

maikeueule, 2018-10-12:
Hi,

When using autotools to build the documentation, the explicit use of CMake variables for OUTPUT_DIRECTORY in Doxyfile.in leads to the use of a /doc directory which breaks the build.

While cmake is likely the way to do, some tools like debmake autodetecting the build tool find autotools and the automated packaging process fails for instance.

```
$  debmake -t -i debuild 
[...]
mkdir -p doc
/usr/bin/doxygen Doxyfile
error: tag OUTPUT_DIRECTORY: Output directory `/doc' does not exist and cannot be created
Exiting...
[...]
```

zuckschwerdt, 2018-10-15:
Actually the prefixes should be set in the Makefile, but somehow it's broken.
See https://github.com/merbanan/rtl_433/blob/master/configure.ac#L26

zuckschwerdt, 2018-10-15:
And

    OUTPUT_DIRECTORY = @CMAKE_CURRENT_BINARY_DIR@/doc