merbanan/rtl_433 #838: OOK package disambigous: False positive for prologue sensor

ionum, 2018-10-22:
Detected OOK package    @ 2018-10-22 15:28:28
2018-10-22 15:28:28 :   Prologue sensor         :       9       :       96
        Channel:         3
        Battery:         OK
        Button:  1
        Temperature:     77.80 C
        Humidity:        172 %
2018-10-22 15:28:28 :   AlectoV1 Temperature Sensor
        House Code:      105
        Channel:         1
        Battery:         OK
        Temperature:     19.90 C
        Humidity:        55 %
        :       CHECKSUM
Analyzing pulses...
Total count:  287,  width: 248473               (993.9 ms)
Pulse width distribution:
 [ 0] count:  287,  width:   115 [113;120]      ( 460 us)
Gap width distribution:
 [ 0] count:    8,  width:  2234 [2233;2235]    (8936 us)
 [ 1] count:  115,  width:  1013 [1012;1016]    (4052 us)
 [ 2] count:  163,  width:   496 [493;527]      (1984 us)
Pulse period distribution:
 [ 0] count:    8,  width:  2350 [2349;2355]    (9400 us)
 [ 1] count:  115,  width:  1129 [1127;1132]    (4516 us)
 [ 2] count:  163,  width:   611 [607;643]      (2444 us)
Level estimates [high, low]:  15918,    214
Frequency offsets [F1, F2]:    -182,      0     (-0.7 kHz, +0.0 kHz)
Guessing modulation: Pulse Position Modulation with fixed pulse width
Attempting demodulation... short_limit: 754, long_limit: 1017, reset_limit: 2236, sync_width: 0
pulse_demod_ppm(): Analyzer Device
bitbuffer:: Number of rows: 9
[00] { 0}                :
[01] {36} 96 0e 30 aa c0 : 10010110 00001110 00110000 10101010 1100
[02] {36} 96 0e 30 aa c0 : 10010110 00001110 00110000 10101010 1100
[03] {36} 96 0e 30 aa c0 : 10010110 00001110 00110000 10101010 1100
[04] {36} 96 0e 30 aa c0 : 10010110 00001110 00110000 10101010 1100
[05] {36} 96 0e 30 aa c0 : 10010110 00001110 00110000 10101010 1100
[06] {36} 96 0e 30 aa c0 : 10010110 00001110 00110000 10101010 1100
[07] {36} 96 0e 30 aa c0 : 10010110 00001110 00110000 10101010 1100
[08] {26} 96 0e 30 80    : 10010110 00001110 00110000 10

merbanan, 2018-10-22:
Submit a sample to the test repository.

zuckschwerdt, 2018-10-22:
The Alecto device ID happens to match the fixed Prologue Model-Number here.
It should work again if you cycle the batteries to get a new ID. Not sure how to block the false positive here as Prologue has few check (and no MIC).

ionum, 2018-10-23:
Thanks for the hint. Maybe we could implement some plausibility checks: humidity cannot be over 100% for example ...

merbanan, 2018-10-23:
That would have to be a separate filter step. The decoder should just output whatever data it receives with as little modification of the data as possible.

merbanan, 2018-10-23:
In this case it would be possible to add the alecto checksum check to the prologue decoder. If the alecto checksum is valid, we assume it is not a prologue message. This is the selectivity we have to implement to reduce collisions.

merbanan, 2018-10-23:
Patch very welcome. Or colliding sample that can reproduce this issue.

ionum, 2018-10-23:
Sample available in https://github.com/merbanan/rtl_433_tests/tree/master/tests/prologue/4

merbanan, 2018-12-06:
PR #899 should get rid of collisions from the Alecto sensor now. @zuckschwerdt have a look.

merbanan, 2018-12-06:
@ionum thanks for the sample.