merbanan/rtl_433 #841: List possible JSON objects

vuokko, 2018-10-25:
I'm working on filter for JSON output which I would then give data to Mosquitto and on the other hand log unkown data and drop the unusable and nonintresting data.

Is there some easy way to dump possible JSON objects to check my match filter doesn't have totally bogus fields?
CSV outputs all headers as top row but with JSON it could be separated to each driver.
{ "decoder" : "Fine RTL_433 deoder", "time" : "string", "model" : "string", "id" : "integer", "temperature" : "decimal" } would be perfect.

zuckschwerdt, 2018-10-26:
Look at http://triq.net/articles/rtl-433-networking
Especially https://github.com/zuckschwerdt/rtl_433_relay/ — there is an example [MQTT filter](https://github.com/zuckschwerdt/rtl_433_relay/blob/master/rtl_433_mqtt_relay.py).

closed 2018-10-30 by zuckschwerdt

vuokko, 2018-10-31:
I got my filter working and I hope it has some usage to someone else too.
The repo is in https://github.com/vuokko/rtl_433_filter