merbanan/rtl_433 #844: Separate the WH24 and WH65B protocols

StephenR0, 2018-10-30:
I have received a communication from Fine Offset about the differences between the output of the WH24 and the WH65B sensor arrays.  Here is what they said:

---------------
Please note the factor difference for wh65b is as follow:

1. wind speed factor: 0.51m/s/hz, with wh24 sensor its wind speed factor is 1.19m/s/hz
2. rain cup each count is 0.01inch ( 0.254mm) while wh24 is 0.3mm.

Beside this, all other is same.

---------------

I assume they mean that the wind speed factor is for both the wind and gust data.  That is the same as my calculations of the wind speed factor.  There is a conflict about the WH24 wind speed factor, though.  Here they said that the WH24 wind speed factor is 1.19m/s/hz and the previous information was 1.12.  I can't verify that, so I suggest that we not disturb that until we get better information.  In any case, I would like to be able to post about how to use rtl_433 with the WH65B without hacking the code as I've done for my purposes.  Is there a way to distinguish between the two types of packets?  If not, is there another way to specify only looking for one type or the other that could be arranged?  Thanks.

zuckschwerdt, 2018-10-30:
As far as I can see the WH65B differes only in a slightly longer preable. The family code (0x24) is the same as the WH24. If we can always get the full preamble we can take a guess, but that's far from reliable, sadly.

StephenR0, 2018-10-30:
I suppose a new protocol code that's not enabled by default would work.  That way it could be selected on purpose.  It is kind of inconvenient that there's no clear difference.

StephenR0, 2018-10-31:
I just realized that I actually have two devices that come in under protocol 78.  In addition to the WH65B, I have a WH25 for indoor temp, humidity, and pressure.  If the WH65B was a different protocol, I would still see the WH24 packets as well.  So, I guess I need a way to only get the WH65B packets and still get my Wh25 packets.  Sorry, it sounds kind of messy.

zuckschwerdt, 2018-10-31:
I guess I'll add a threshold to the preamble length to decide wether it's a WH65B. You can then test that for a week or so and report if (and how many) wrong WH24 reports came in. Maybe the preamble is stable enough to use after all.

merbanan, 2018-10-31:
Can someone post analyzer (-A) output and sigrok pictures? Then more people might be able to help out.

merbanan, 2018-10-31:
Maybe they used different amount of repeats ? (Etc...)

zuckschwerdt, 2018-10-31:
It's NRZ, so rather hard to follow the bit timing by eye. This is the best representation I got:
[image]
So there is a slightly longer preamble and maybe a postamble. But I have to check if that's a demodulator artifact.

merbanan, 2018-10-31:
Well the payload differs also 08 vs 00 at the end. Can that be used ?

zuckschwerdt, 2018-10-31:
Payload is `FF II DD VT TT HH WW GG RR RR UU UU LL LL LL CC BB` so that `08` is Wind speed. But the visualization there might be wrong. The raw code are:

    [WH24]{196}d55555555516ea125f8571032704010025000080000047839
    [WH65B]{209}5555555555516ea122833f143a0800000008001000000460a1008

merbanan, 2018-10-31:
Well one signal seems to be longer 196 vs 209. Can you point me to the samples you use?

zuckschwerdt, 2018-10-31:
https://github.com/merbanan/rtl_433_tests/tree/master/tests/fineoffset_wh24
https://github.com/merbanan/rtl_433_tests/tree/master/tests/fineoffset_wh65b/01

merbanan, 2018-10-31:
It could be that the WH24 is on 433MHz and the WH65b is 868/915MHz.

StephenR0, 2018-10-31:
Mine is 915Mhz.

merbanan, 2018-10-31:
Ok, analysis by looking at the signal in sigrok (awesome job there @zuckschwerdt ) gives that one cycle in the preamble (one high frequency followed by one low frequency) is 116us in both the the wh24 and wh65 thus any difference in the transport layer does not depend on temperature or batter level. 

The signal duration of the wh65 signals are ca 12.03ms and the wh24 signals are ca 11.3ms. The preamble duration of the wh24  is 2.389ms and the wh65 2.625ms.  The differece of the preambles give this 2625-2389=236. 236 =~ 116*2 -> the wh65 has 2 more cycles of the preamble (4 bits) compared to the wh24.

From the posted example it is clear that there are more data present before the sync word visualized below as an X.

[WH24  ]{196}Xd55555555516ea
[WH65B]{209}5555555555516ea

So by locating the sync word and using the bit count before the sync word as heuristic should be quick and reliable way to decide between wh24 and wh65b signals.

zuckschwerdt, 2018-10-31:
Looking a the test samples we got:
```
[wh24/g001] {196} d5 55 55 55 55 16 ea 12 5f 85 71 03 27 04 01 00 25 00 00 80 00 00 47 83 9
[wh24/g002] {196} d5 55 55 55 55 16 ea 12 5f 81 71 03 27 03 80 80 25 00 00 80 00 00 00 37 1
[wh24/g003] {196} d5 55 55 55 55 16 ea 12 5f ff b1 03 27 81 80 80 25 00 00 80 00 00 7f f3 8
[wh24/g004] {196} d5 55 55 55 55 16 ea 12 5f fe 31 03 27 83 00 80 25 00 00 80 00 00 44 38 0
[wh24/g005] {196} d5 55 55 55 55 16 ea 12 5f 81 f1 03 27 82 00 80 25 00 00 80 00 00 44 7a 8
[wh24/g006] {196} d5 55 55 55 55 16 ea 12 5f 87 f1 03 27 82 00 80 25 00 00 80 00 00 34 f1 0
[wh24/g007] {196} d5 55 55 55 55 16 ea 12 5f 8c 71 03 27 81 80 80 25 00 00 80 00 00 2b ec 0
[wh24/g008] {196} d5 55 55 55 55 16 ea 12 5f 86 f1 03 27 81 00 80 25 00 00 80 00 00 6e 28 8
[wh24/g009] {196} d5 55 55 55 55 16 ea 12 5f 80 f1 03 27 82 80 80 25 00 00 80 00 00 32 e8 8
[wh24/g010] {196} d5 55 55 55 55 16 ea 12 5f 87 f1 03 27 83 80 80 25 00 00 80 00 00 43 81 8
[wh24/g011] {196} d5 55 55 55 55 16 ea 12 5f 8a f1 03 a7 83 00 80 25 00 00 80 00 00 3f 80 9

[wh65b/g001] {209} 55 55 55 55 55 51 6e a1 22 83 3f 14 3a 08 00 00 00 08 00 10 00 00 04 60 a1 00 8
[wh65b/g002] {209} 55 55 55 55 55 51 6e a1 22 80 03 14 3a 08 00 00 00 08 00 10 00 00 06 0b 11 02 8
[wh65b/g003] {208} 55 55 55 55 55 51 6e a1 22 83 3f 14 3a 08 00 00 00 08 00 10 00 00 04 60 a1 00
[wh65b/g004] {209} 55 55 55 55 55 51 6e a1 22 83 3f 14 3a 08 00 00 00 08 00 10 00 00 04 60 a1 02 8
[wh65b/g006] {209} 55 55 55 55 55 51 6e a1 22 83 3f 14 3a 08 00 00 00 08 00 10 00 00 04 60 a1 00 0
[wh65b/g007] {209} 55 55 55 55 55 51 6e a1 22 83 3f 14 3a 08 00 00 00 08 00 10 00 00 04 60 a1 00 8
```
and removing then extra 4 bits on front of the WH65B samples we can [inspect in BitBench](http://triq.net/bitbench?c=%5Bwh24%2Fg001%5D%20%20%7B196%7D%20d55555555516ea125f8571032704010025000080000047839&c=%5Bwh24%2Fg002%5D%20%20%7B196%7D%20d55555555516ea125f8171032703808025000080000000371&c=%5Bwh24%2Fg003%5D%20%20%7B196%7D%20d55555555516ea125fffb103278180802500008000007ff38&c=%5Bwh24%2Fg004%5D%20%20%7B196%7D%20d55555555516ea125ffe31032783008025000080000044380&c=%5Bwh24%2Fg005%5D%20%20%7B196%7D%20d55555555516ea125f81f10327820080250000800000447a8&c=%5Bwh24%2Fg006%5D%20%20%7B196%7D%20d55555555516ea125f87f1032782008025000080000034f10&c=%5Bwh24%2Fg007%5D%20%20%7B196%7D%20d55555555516ea125f8c7103278180802500008000002bec0&c=%5Bwh24%2Fg008%5D%20%20%7B196%7D%20d55555555516ea125f86f103278100802500008000006e288&c=%5Bwh24%2Fg009%5D%20%20%7B196%7D%20d55555555516ea125f80f1032782808025000080000032e88&c=%5Bwh24%2Fg010%5D%20%20%7B196%7D%20d55555555516ea125f87f1032783808025000080000043818&c=%5Bwh24%2Fg011%5D%20%20%7B196%7D%20d55555555516ea125f8af103a78300802500008000003f809&c=&c=%5Bwh65b%2Fg001%5D%20%7B205%7D%20555555555516ea122833f143a0800000008001000000460a1008&c=%5Bwh65b%2Fg002%5D%20%7B205%7D%20555555555516ea1228003143a080000000800100000060B11028&c=%5Bwh65b%2Fg003%5D%20%7B204%7D%20555555555516ea122833f143a0800000008001000000460a100&c=%5Bwh65b%2Fg004%5D%20%7B205%7D%20555555555516ea122833f143a0800000008001000000460a1028&c=%5Bwh65b%2Fg006%5D%20%7B205%7D%20555555555516ea122833f143a0800000008001000000460a1000&c=%5Bwh65b%2Fg007%5D%20%7B205%7D%20555555555516ea122833f143a0800000008001000000460a1008&f=41h%2016h%20136h%2012b).
That shows a longer postamble (at least the demodulator thinks that) which might be easier to detect reliably. I'm not sure if the postamble is really there and NRZ is to tedious to follow by eye :/

zuckschwerdt, 2018-10-31:
I'd say if (after aligning the sync word) there is a postamble of 8 or more (nominal 12) bits we'll call it a WH65B. The WH24 would have only 3 bits so it should be somewhat resilient to slight errors.
Now someone needs to confirm there is actually a postamble and it's not a demodulator artifact…

zuckschwerdt, 2018-10-31:
Something liek this (the actual calculation differences are missing): https://github.com/zuckschwerdt/rtl_433/commit/f09fa12

merbanan, 2018-10-31:
If you look at the signal in sigrok you can see that there is a real signal there but the decoded bits are bogus. But in our case it actually doesn't matter. We just need to measure the signal length somehow. Either towards the beginning or the end.

The is more reliable as the preamble has primed the receiver. As long as the demodulator spits out bits per time unit everything will be fine.

StephenR0, 2018-10-31:
It looks good at first glance.  I'll put it in service and get more experience with it.

-> rtl_433 -q -U -F json -p 39.741 -R 78 -f 914980000
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Bit detection level set to 0 (Auto).
{"time" : "2018-10-31 19:44:17", "model" : "Fine Offset WH65B", "id" : 89, "temperature_C" : 13.300, "humidity" : 44, "wind_dir_deg" : 309, "wind_speed_ms" : 0.191, "gust_speed_ms" : 0.510, "rainfall_mm" : 286.512, "uv" : 258, "uvi" : 0, "light_lux" : 17468.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-10-31 19:44:19", "model" : "Fine Offset Electronics, WH25", "id" : 21, "temperature_C" : 19.500, "humidity" : 48, "pressure_hPa" : 985.500, "battery" : "OK", "mic" : "CHECKSUM"}
{"time" : "2018-10-31 19:44:33", "model" : "Fine Offset WH65B", "id" : 89, "temperature_C" : 13.300, "humidity" : 44, "wind_dir_deg" : 309, "wind_speed_ms" : 0.446, "gust_speed_ms" : 0.510, "rainfall_mm" : 286.512, "uv" : 258, "uvi" : 0, "light_lux" : 17313.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-10-31 19:44:49", "model" : "Fine Offset WH65B", "id" : 89, "temperature_C" : 13.300, "humidity" : 45, "wind_dir_deg" : 309, "wind_speed_ms" : 0.127, "gust_speed_ms" : 0.510, "rainfall_mm" : 286.512, "uv" : 262, "uvi" : 0, "light_lux" : 17530.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-10-31 19:45:05", "model" : "Fine Offset WH65B", "id" : 89, "temperature_C" : 13.300, "humidity" : 46, "wind_dir_deg" : 309, "wind_speed_ms" : 0.000, "gust_speed_ms" : 0.000, "rainfall_mm" : 286.512, "uv" : 258, "uvi" : 0, "light_lux" : 17189.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-10-31 19:45:21", "model" : "Fine Offset WH65B", "id" : 89, "temperature_C" : 13.300, "humidity" : 45, "wind_dir_deg" : 309, "wind_speed_ms" : 0.000, "gust_speed_ms" : 0.000, "rainfall_mm" : 286.512, "uv" : 253, "uvi" : 0, "light_lux" : 17143.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-10-31 19:45:23", "model" : "Fine Offset Electronics, WH25", "id" : 21, "temperature_C" : 19.500, "humidity" : 48, "pressure_hPa" : 985.800, "battery" : "OK", "mic" : "CHECKSUM"}
{"time" : "2018-10-31 19:45:37", "model" : "Fine Offset WH65B", "id" : 89, "temperature_C" : 13.300, "humidity" : 46, "wind_dir_deg" : 309, "wind_speed_ms" : 0.510, "gust_speed_ms" : 1.020, "rainfall_mm" : 286.512, "uv" : 245, "uvi" : 0, "light_lux" : 16972.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-10-31 19:45:53", "model" : "Fine Offset WH65B", "id" : 89, "temperature_C" : 13.200, "humidity" : 46, "wind_dir_deg" : 309, "wind_speed_ms" : 0.191, "gust_speed_ms" : 0.510, "rainfall_mm" : 286.512, "uv" : 245, "uvi" : 0, "light_lux" : 17034.000, "battery" : "OK", "mic" : "CRC"}

zuckschwerdt, 2018-11-03:
@StephenR0 give a thumbs up on #845 if everything works as expected.

StephenR0, 2018-11-03:
It looks good from here.  I was thinking that it would be good to get some testing from a WH24 user, but I haven't gotten a response from that.  I think we're good, though.

zuckschwerdt, 2018-11-03:
Reopen if anything comes up.

closed 2018-11-03 by zuckschwerdt

Zodiac-69, 2018-11-04:
Hi Stephen

Here is a copy of my WH24 unit.
root@Domoticz:~# rtl_433 -q -U -F json -p 39.741 -R 78 -f 433920000
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Bit detection level set to 0 (Auto).
{"time" : "2018-11-04 09:40:26", "model" : "Fine Offset WH24", "id" : 234, "temperature_C" : 18.400, "humidity" : 78, "wind_dir_deg" : 135, "wind_speed_ms" : 6.160, "gust_speed_ms" : 6.720, "rainfall_m                              m" : 153.900, "uv" : 4658, "uvi" : 12, "light_lux" : 78429.602, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-11-04 09:40:42", "model" : "Fine Offset WH24", "id" : 234, "temperature_C" : 18.400, "humidity" : 78, "wind_dir_deg" : 118, "wind_speed_ms" : 6.160, "gust_speed_ms" : 7.840, "rainfall_mm" : 153.900, "uv" : 4666, "uvi" : 12, "light_lux" : 78454.898, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-11-04 09:40:58", "model" : "Fine Offset WH24", "id" : 234, "temperature_C" : 18.300, "humidity" : 78, "wind_dir_deg" : 120, "wind_speed_ms" : 6.300, "gust_speed_ms" : 6.720, "rainfall_mm" : 153.900, "uv" : 4655, "uvi" : 12, "light_lux" : 78367.500, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-11-04 09:41:14", "model" : "Fine Offset WH24", "id" : 234, "temperature_C" : 18.300, "humidity" : 78, "wind_dir_deg" : 130, "wind_speed_ms" : 5.880, "gust_speed_ms" : 7.840, "rainfall_mm" : 153.900, "uv" : 4665, "uvi" : 12, "light_lux" : 78560.703, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-11-04 09:41:30", "model" : "Fine Offset WH24", "id" : 234, "temperature_C" : 18.300, "humidity" : 78, "wind_dir_deg" : 165, "wind_speed_ms" : 7.000, "gust_speed_ms" : 7.840, "rainfall_mm" : 153.900, "uv" : 4666, "uvi" : 12, "light_lux" : 78519.297, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-11-04 09:41:46", "model" : "Fine Offset WH24", "id" : 234, "temperature_C" : 18.200, "humidity" : 78, "wind_dir_deg" : 120, "wind_speed_ms" : 5.740, "gust_speed_ms" : 6.720, "rainfall_mm" : 153.900, "uv" : 4659, "uvi" : 12, "light_lux" : 78434.203, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-11-04 09:42:02", "model" : "Fine Offset WH24", "id" : 234, "temperature_C" : 18.200, "humidity" : 78, "wind_dir_deg" : 120, "wind_speed_ms" : 6.580, "gust_speed_ms" : 7.840, "rainfall_mm" : 153.900, "uv" : 4642, "uvi" : 11, "light_lux" : 78073.102, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-11-04 09:42:18", "model" : "Fine Offset WH24", "id" : 234, "temperature_C" : 18.200, "humidity" : 78, "wind_dir_deg" : 153, "wind_speed_ms" : 5.600, "gust_speed_ms" : 7.840, "rainfall_mm" : 153.900, "uv" : 4628, "uvi" : 11, "light_lux" : 77744.203, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-11-04 09:42:34", "model" : "Fine Offset WH24", "id" : 234, "temperature_C" : 18.200, "humidity" : 78, "wind_dir_deg" : 113, "wind_speed_ms" : 6.580, "gust_speed_ms" : 7.840, "rainfall_mm" : 153.900, "uv" : 4644, "uvi" : 11, "light_lux" : 78057.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-11-04 09:42:50", "model" : "Fine Offset WH24", "id" : 234, "temperature_C" : 18.200, "humidity" : 78, "wind_dir_deg" : 122, "wind_speed_ms" : 7.140, "gust_speed_ms" : 8.960, "rainfall_mm" : 153.900, "uv" : 4658, "uvi" : 12, "light_lux" : 78275.500, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-11-04 09:43:06", "model" : "Fine Offset WH24", "id" : 234, "temperature_C" : 18.000, "humidity" : 78, "wind_dir_deg" : 129, "wind_speed_ms" : 8.120, "gust_speed_ms" : 10.080, "rainfall_mm" : 153.900, "uv" : 4671, "uvi" : 12, "light_lux" : 78305.398, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-11-04 09:43:22", "model" : "Fine Offset WH24", "id" : 234, "temperature_C" : 17.900, "humidity" : 78, "wind_dir_deg" : 151, "wind_speed_ms" : 5.460, "gust_speed_ms" : 6.720, "rainfall_mm" : 153.900, "uv" : 4682, "uvi" : 12, "light_lux" : 78314.602, "battery" : "OK", "mic" : "CRC"}
{"time" : "2018-11-04 09:43:38", "model" : "Fine Offset WH24", "id" : 234, "temperature_C" : 18.000, "humidity" : 79, "wind_dir_deg" : 113, "wind_speed_ms" : 5.880, "gust_speed_ms" : 6.720, "rainfall_mm" : 153.900, "uv" : 4681, "uvi" : 12, "light_lux" : 78376.703, "battery" : "OK", "mic" : "CRC"}

StephenR0, 2018-11-04:
That looks good.  I was hoping we didn't break anything.  👍

zuckschwerdt, 2024-05-25:
Yes, the differentiation is only a few bits in the preamble. Not sure if there is a way to improve that.
https://github.com/merbanan/rtl_433/blob/86abae0550eb134d2fad1badead623fd3a800356/src/devices/fineoffset.c#L211-L218