merbanan/rtl_433 #851: One of two Quhwa PIR motion sensors is not detected

pdo-smith, 2018-11-02:
I have two identical sets of Quhwa PIR motion sensors, doorbell chime receivers and activation/deactivation push button switches.

They work perfectly together. I can pair each motion sensor with either of the two doorbell chime receivers. I can pair each push button switch with either of the two doorbell chime receivers.

So far, so good. This is how it should be and the product works perfectly, with good interoperability.

Using rtl_433 I can detect both push button switches but only one of the two motion sensors using the -A switch.

The second motion sensor produces no results in rtl_433 with either switch -a or -A. Consequently I cannot record any output for analysis. But this sensor operates with either of the two doorbell chime receivers. Both motion sensors have fresh batteries. Both motion sensors have identical boards, QH-007T, REV12 by Quhwa. There are no user settings on the boards.

I am at a complete loss how to go forward. Can anyone suggest how I should go about detecting and recognising this one, working motion sensor that seems to be invisible to rtl_433?

zuckschwerdt, 2018-11-02:
The signal might be out of the default frequency window. Use `-s 1024k` to get a wider window (with default `-f 433.92` it would cover 433.4–434.4. Use perhaps `-f 433.8` for a better frequency window.
Also confirm visually with CubicSDR or Gqrx that you receive something from the device in question.

pdo-smith, 2018-11-02:
Thanks, -s 1024k did the trick. Terrific response!

pdo-smith, 2018-11-02:
Lovely. Now it working perfectly.

zuckschwerdt, 2018-11-02:
You might want to collect the "freq estimate" output from all sensors (offset to the `-f` freq) and pick a spot "in the middle" so you get all even with (default) `-s 250k`.
I guess `-f 433.85M` will work ;)

closed 2018-11-02 by zuckschwerdt