merbanan/rtl_433 #855: Six device decoders don't call the data_acquired_handler

winterrace, 2018-11-05:
The 6 device decoders listed below don't pass the parsed data elements to the data_acquired_handler. So none of the registered output handlers will be reached. Judging on the quick look I made, it seems they just print their findings to the console.

- ec3k
- elv
- lightwave_rf
- steffen
- valeo
- x10_rf

Is this already known and planned to make a fix?

zuckschwerdt, 2018-11-05:
https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L184
PR to convert the last remaining decoders to data_make very welcome!

merbanan, 2018-11-05:
We are missing samples for a few of the decoders. They are disabled by default.

zuckschwerdt, 2018-11-05:
But maybe not worth the effort, as @merbanan said samples are missing and if the decoders are simple I'll eventually convert them to a flex config anyway (see rtl_433.example.conf in https://github.com/merbanan/rtl_433/pull/790/files).

winterrace, 2018-11-05:
Ah, OK - I forgot that I already read about that exception ;-)
Thanks for your comments! I'll keep an eye on that as well.

zuckschwerdt, 2018-11-05:
E.g. valeo would be

    -X "Valeo Car Key:OOK_MC_ZEROBIT:106:106:400,bits=461,preamble=e8e8"

which is much more maintainable and hackable by users.

zuckschwerdt, 2018-12-10:
I think I'm going to add a `conf` folder and add flex configs for some x1527/PT2262 decoders there. Like valeo.conf, steffen.conf, and others (from current devices and the example conf). Then remove the old coded devices (deprecating those protocol numbers).

merbanan, 2018-12-10:
I was going to port them to structured output but your way is ok also. I'll go ahead with the duplicate filtering stuff instead.

merbanan, 2018-12-10:
True that.

zuckschwerdt, 2018-12-10:
I've added confs for ec3k, steffen, and valeo. I'll remove the old code soon.

zuckschwerdt, 2018-12-13:
The ec3k, steffen, valeo have been converted to flex confs and retired, the elv, lightwave_rf, x10_rf have been converted to data_make.

closed 2018-12-13 by zuckschwerdt