merbanan/rtl_433 #856: TPMS sensor recomedation

sowinski, 2018-11-05:
Hi,

I would like to buy this kind of TPMS sensor for my car. (Which is mounted on the outside of the tire)
[https://www.amazon.com/ZEEPIN-Universal-Monitoring-Real-time-Temperature/dp/B075848KTW/ref=sr_1_4?s=automotive&ie=UTF8&qid=1541451506&sr=1-4&keywords=tpms&dpID=41pIGCblRdL&preST=_SY300_QL70_&dpSrc=srch](url)

Has anyone of you experience with such a device? I would like to buy some of them and would be nice if they are already supported by rtl_433.

Thank you

merbanan, 2018-11-05:
Most likely unsupported. But sand sample signals and it could be.

zuckschwerdt, 2018-11-06:
That'll be the Zeepin C220 Solar from this [review](https://jwtechreviews.com/2018/01/15/zeepin-c220-solar-powered-tpms/)?

sowinski, 2018-11-06:
Cool thank you :) Sending samples is no problem.

I guess yes. I bought that one today: https://www.amazon.de/gp/product/B07FS7K6N1/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

zuckschwerdt, 2018-11-19:
Did you have time to record samples (`-a -t` option)?

zuckschwerdt, 2018-12-09:
Reopen with samples (`-S all`) if you need help or can confirm this works. Thanks!

enavarro222, 2018-12-25:
Hi, I recently get such TPMS.

But I got very few free time to reverse it...
For now I just got few record. I will try to record more data in next days.
https://github.com/merbanan/rtl_433_tests/pull/251

enavarro222, 2018-12-26:
Ok I will try to records some.
btw, I could'n find the online tool you have to analyze signals. If you see what I mean, can you give me the url ?

zuckschwerdt, 2018-12-26:
You can drop the sample files into http://triq.net/iqs and take a peek at the signal there.

enavarro222, 2018-12-29:
OK I records more data (see https://github.com/merbanan/rtl_433_tests/pull/255)

However it looks strange... no clear common section in decoded data (as one could expect for an ID section), see on [bitbench](http://triq.net/bitbench?c=%5Btime%20%20%20%20%3A%202018-12-28%2023%3A00%3A47%20%20data%20%20%20%20%3A%2011%C2%B0C%200.1bar%5D&c=%7B85%7Df80005656491b84422dd60&c=%7B80%7Dffffacac923708845bac&c=%5Btime%20%20%20%20%20%20%3A%202018-12-28%2023%3A40%3A37%20data%20%20%20%20%20%20%3A%2019%C2%B0C%200.1bar%5D&c=%7B84%7Df0000c4c4814ff09c43c4&c=%7B80%7Dffffc4c4814ff09c43c4&c=%5Btime%20%20%20%20%20%20%3A%202018-12-28%2023%3A50%3A34%20data%20%20%20%20%20%20%3A%2020%C2%B0C%200.1bar%5D&c=%7B81%7D80005e5e4523fc4a25de0&c=%7B80%7Dffffbcbc8a47f8944bbc&c=%5Btime%20%20%20%20%20%20%3A%202018-12-28%2023%3A55%3A33%20data%20%20%20%20%20%20%3A%2020%C2%B0C%200.1bar%5D&c=%7B89%7Dff80005e5e4523fc4a25de0&c=%7B80%7Dffffbcbc8a47f8944bbc&c=%5Btime%20%20%20%20%20%20%3A%202018-12-29%2000%3A00%3A31%20data%20%20%20%20%20%20%3A%2020%C2%B0C%200.1bar%5D&c=%7B94%7Dfffc0002f2f2291fe2512ef0&c=%7B80%7Dffffbcbc8a47f8944bbc&c=%5Btime%20%20%20%20%20%20%3A%202018-12-29%2000%3A10%3A29%20data%20%20%20%20%20%20%3A%20%209%C2%B0C%200.1bar%5D&c=%7B82%7Dc000282826cac51e19a80&c=%7B80%7Dffffa0a09b2b147866a0&c=%5Btime%20%20%20%20%20%20%3A%202018-12-29%2000%3A20%3A28%20data%20%20%20%20%20%20%3A%20%203%C2%B0C%200.1bar%5D&c=%7B86%7Dfc0003838155af52e09b80&c=%7B80%7Dffffe0e0556bd4b826e0&f=16h%204b-4b%204b-4b%204b-4b%204b-4b%204b-4b%204b-4b%204b-4b%204b-4b%20%204b-4b%20)

According to user manual, the receptor should learn an "inflate code". This is done at factory, but could be done after.  This code is send by the sensors when mount on a tire (i.e. I guess when pressure goes from 0 to >0).

I suspect that this code is a random number used to decode nexts msg.
If someone has some info on such mechanism...
I will try to record such "inflate code" sending.

enavarro222, 2018-12-29:
You are right g113 is an error it is an other sensor. I'll remove it. Also g009 get both sensors at same time I guess.

I didn't get the FSK you see... Looking at the signal in audacity it seems that the initial carrier took a time to get to the correct frequency (but I'm not very confident).

To summarize:
- One long "wake up" with just the carrier and then just some "empty" data (it is in a first record)
- then (most of time in a second record): two times the same data with a different preamble. OOK. Seems to be MC or DMC. But data looks encrypted or we miss something in the encoding

I also notice:
- same measurements (temperature and pressure) results in same data (it may not be always the case... But at least it is possible), so no rolling code !?
- no clear ID section for now
- a msg is send when sensor is mount on a tire (so when pressure increase quickly) but msg does look the same. At least there is nothing special in the first "wake up" part. I will try to get new record for that
- when battery are remove and then re-installed on sensor no special action is needed to make the display decode it again. It just works... So at least there shouldn't have a random code generated on sensor start up.

So kind of tricky tpms !

zuckschwerdt, 2018-12-29:
Sorry about "FSK" i was confused there ;) The good reception and decoding and also the observation that there is probably no rolling code are already a good step forward.
Don't bother to record more samples. Just the "codes" output is enough to go from here.

Perhaps you can put the sensors in a bottle and pressurize it, then release the pressure slowly and note the messages (doing that on a mounted tire seems too much work).
I can recommend a Garden Hand Sprayer like https://www.amazon.com/Planted-Perfect-Hand-Garden-Sprayer/dp/B01LYWAB2S if the sensor fits through the opening. It work in very fine steps up to 200 kPa.

If you get 100 to 1000 codes a pattern should become visible or at least brute-force decodable.

enavarro222, 2018-12-31:
Ok I got very few time on it, and it will be worth next days...

Nice idea for the bottle !

I just push some few records, pictures of the card + notes on  readme (on test repo).

I guess https://fccid.io/2ANZATP620 is the FCC application for these sensors. I have to dig on it, one may found some interesting clues.

Also I open the receiver display the radio chip is a "basic" transceiver with no special encryption, so if there is, it should be managed by the micro.

sowinski, 2019-03-06:
I got my TMPS now. I also bought a Pi and Installed RTL_433.

What can I do to help?

zuckschwerdt, 2019-03-06:
Find the ID code on the sensors and try to find that in the data (see above for ways to get data). Also you can pressurize a sensor and slowly release pressure (or the other way around) and record data.
We need to get an idea which bits related to the ID and which bits change with pressure (and how that is encoded). There might be some encryption or data whitening (basically and XOR over all bits). A nice series of readings with continously changing values (dropping pressure) would reveal that.

zuckschwerdt, 2019-03-14:
@enavarro222 your Zeepin C240 is Gaussian OOK if I remember correctly. This one here is Gaussian FSK. The FSK would be easier to fix for Gaussian filtering (softer edges -> narrow bandwidth). But we are still working on improvements to the demodulators (detectors really). And I'll see if I can apply the FSK ideas from here to the OOK part somehow.

closed 2020-11-13 by zuckschwerdt