merbanan/rtl_433 #859: Sampling the signal strength as diagnostic option - feature request

pdo-smith, 2018-11-06:
I suggest that we have an option that will report the signal strength at the same time that samples are acquired. This could be very useful for diagnosing issues to do with antennas, battery strength and distance between sender and receiver.

I have used rtlsdr_scanner to great effect to report signal strength when tuning antennas for the FM frequency band but it is ineffective in the very narrow 433 Mhz band with its brief signal bursts.

I have also used rtl_fm_streamer for listening to the FM band and used the Json-RPC feature to sample the signal strength. 

I find this incredibly useful and would very much value such a feature in rtl_433. I am tuning my rtl-sdr antenna for the 433 Mhz band and checking the transmission distances of my devices. Signal strength measurements would help so much.

The device supports this feature and the code has already been done by projects such as rtl_fm_streamer and rtlsdr_scanner.

merbanan, 2018-11-06:
Is pr 733 good enough for you?

pdo-smith, 2018-11-07:
Here is an example of signal strength output from rtlsdr_scanner

[image]

It shows the signal strengths of the FM transmitters in my locality and was incredibly useful for tuning my antenna.
Below is the output from rtl_fm_streamer, after some hacking by myself. The numbers are scaled differently.

peter@Viki:~/scripts$ ./fm

----Frequency Response----

Freq     dB     Station
087.6:  -21 	PEFM - L
087.9:  -10 	Metro- P
088.3:  -9 	Ifm
089.2:  -11 	5FM - P
091.0:  -9 	UMW - L
092.3:  -7 	UMW - L
093.8:  -25 	Lotus - L
094.2:  -13 	Algoa - P
095.0:  -15 	Algoa - Z
095.5:  -10 	Algoa - L
097.0:  -17 	KQFM - L
097.5:  -15 	Radio 2000 - P
098.3:  -8 	Lotus - P
098.8:  -9 	Radio 2000 - L
100.5:  -9 	Metro - L
101.0:  -9 	RSG - P
101.8:  -15 	RSG - Z
102.3:  -8 	RSG - L
103.8:  -16 	Kingfisher - L
104.6:  -22 	SAfm - P
105.4:  -10 	SAfm - Z
105.9:  -7 	SAfm - L
107.5:  -20 	Kingfisher - P
107.9:  -5 	Bay FM - P
----Frequency Response----

zuckschwerdt, 2018-11-07:
If you write a scanner and listen at different, set gains you could compute that. But listening with automatic gain you have no reference. The faintest and the strongest signal alike will come in at a fixed point somewhat below 0 dBFS.

merbanan, 2018-11-07:
This works if you have a fixed tuner gain against stationary signals. OOK transmissions with adaptive gains are more tricky.

merbanan, 2018-11-07:
If you ask the tuner for its gain when receiving a signal then you could add that to a meta-data message. But I am not sure the tuner can answer quickly enough for all this to workout in a reliable way.

zuckschwerdt, 2018-11-07:
AFAIK there is no API for that as the automatic gain is on the analog side and probably has no read register at all.

pdo-smith, 2018-11-08:
I have for the time being solved my problem by using rtl_power from the rtl_sdr suite. Its time resolution was limited to 1 second, which was too coarse for capturing short events. So I modified it to have a decisecond resolution. This allowed me to capture short duration events with a resolution of a tenth of a second.

I use a command like this
`./rtl_power -f 433.82M:434.2M:190k -g 20 -i 1 | while read line; do echo $line | cut -f 7- -d , | sed s/,//g | xargs -n 1 | sort | head -1; done`

and this reports the signal power every tenth of a second at a fixed gain of 20 between 433.82M and 434.2M. For example
-22.44
-22.38
-22.39
-22.44
-3.64
0.80
0.54
-0.43
-22.43
-22.49

and another device
-22.46
-22.43
-11.39
-10.00
-9.87
-9.57
-9.43
-9.96
-10.05
-9.80
-9.84
-9.76
-9.66
-9.67
-9.83
-9.79
-9.72
-10.13
-14.60
-22.42
-22.40
-22.48

Now I can see why some devices are troublesome and others work well, especially when there are battery issues.

zuckschwerdt, 2018-11-09:
You can try https://github.com/zuckschwerdt/rtl_433/tree/feat-rssi
There is a crude `fprintf()` output of meta data after each decode:

    FSK meta: 37.1 kHz -28.4 kHz RSSI -0.1 dB (15972) SNR 21.0 dB (15941)

With frequency offset (two for FSK), RSSI and SNR (in dB). I still have to check the dB values, those are likely wrong. Also this needs to be hooked into the output formatters, but for now it's just to try the feature.

zuckschwerdt, 2018-11-09:
Aside: **RSSI** is the "relative signal strength indicator", an attenuation (negative value) up to 0 dB which is full scale strength. **SNR** is the "signal to noise ratio", a "distance" of noise to signal, up to 21 dB for 8-bit input (RTL-SDR).

zuckschwerdt, 2018-11-09:
I've update the patch to use data_make. It's still early WIP but at least the KV looks nice:
```
2018-11-09 11:44:26 :	LaCrosse TX Sensor 	:	98
	Temperature:	 17.2 C
	META:	 ASK
	F:	 -126.1 kHz
	RSSI:	 -16.6 dB
	SNR:	 41.5 dB
2018-11-09 11:44:32 :	Bresser 3CH sensor
	Id:	 32
	Channel:	 1
	Battery:	 OK
	Temperature:	 54.70 F
	Humidity:	 71 %
	Integrity:	 CHECKSUM
	META:	 ASK
	F:	 15.2 kHz
	RSSI:	 -16.8 dB
	SNR:	 41.7 dB
2018-11-09 11:44:57 :	Bresser 3CH sensor
	Id:	 9
	Channel:	 2
	Battery:	 OK
	Temperature:	 63.80 F
	Humidity:	 62 %
	Integrity:	 CHECKSUM
	META:	 ASK
	F:	 4.2 kHz
	RSSI:	 -12.2 dB
	SNR:	 44.1 dB
```

zuckschwerdt, 2018-11-09:
I think if you have a calm noise floor and measure it at a fixed gain and correct it for the gain, then you could calculate the absolute signal strength by just substracting it from the SNR.

pdo-smith, 2018-11-10:
Hi Christian,
This looks very promising indeed. I especially like your inclusion of the centre frequency. It has been quite an eye opener and now I can see why I must use the parameter -s 1024.

There is one problem though. The RSSI figures show little difference for me. For example below are two samples taken close to the antenna and far out in my garden.

<165>1 2018-11-10T04:45:21Z Viki rtl_433 - - - {"time":"2018-11-10 06:45:21","model":"WS Temperature Sensor","id":0,"channel":1,"battery":"LOW","temperature_C":0.000000,"mod":"ASK","freq":433.980072,"rssi":-0.122314,"snr":22.883747}

<165>1 2018-11-10T04:47:20Z Viki rtl_433 - - - {"time":"2018-11-10 06:47:20","model":"WS Temperature Sensor","id":0,"channel":1,"battery":"LOW","temperature_C":0.000000,"mod":"ASK","freq":433.971954,"rssi":-0.297852,"snr":24.212067}

The utility rtl_power shows large differences. I am still working on improvements to rtl_power, which I think is a useful adjunct to what you are doing. If there is any interest I can submit it to rtl_433 as an additional testing tool.

But you are definitely on the right track. This looks very encouraging. One small request though - can you include this data in the output for -a ? It would be very useful during the diagnostic testing phase.

zuckschwerdt, 2018-11-10:
Good idea to also output with `-A`.
The levels are to be expected if the automatic gain works. I mostly see OOK signals going near FS (full scale) with little noise floor. Pick a file from the rtl_433_tests repo and validate that.

zuckschwerdt, 2018-11-10:
I used https://github.com/zuckschwerdt/tx_tools/ to generate sample files at FS, half FS, and quarter FS (1.0, 0.5, 0.25 signal amplitude measured as norm). E.g.
```
code_gen -g 0.5 -r examples/generic.txt halfamp_test_1000k.cu8
```
and the rtl_433 code works out (0, -6, -12 dB):
```
{"time" : "@0.319488s", "model" : "Generic motion sensor", "code" : "800a2",
 "mod" : "ASK", "freq" : 0.105, "rssi" : -0.896, "snr" : 19.665, "up" : 13331, "lo" : 143}

{"time" : "@0.319488s", "model" : "Generic motion sensor", "code" : "800a2",
 "mod" : "ASK", "freq" : 0.105, "rssi" : -6.859, "snr" : 14.832, "up" : 3377, "lo" : 110}

{"time" : "@0.319488s", "model" : "Generic motion sensor", "code" : "800a2",
 "mod" : "ASK", "freq" : 0.104, "rssi" : -12.144, "snr" : 9.355, "up" : 1000, "lo" :
```
("up", "lo" raw values from the estimator added as additional info).

pdo-smith, 2018-11-10:
_"The levels are to be expected if the automatic gain works."_

Yes, that is true. I did some more tests at a fixed gain of 20  and it showed up the differences very clearly, both RSSI and SNR. This is really a very good result. I am so pleased. 

I need to experiment with suitable gain values but it seems that no useful measurements are possible with auto gain. You probably need to disable RSSI and SNR reporting when autogain is enabled.

So my strategy for my enhanced alarm system now becomes this:
1) run it normally in auto gain mode.
2) once a week put it into test mode and exercise all the sensors. In test mode the system will be put into fixed gain mode. By exercising it this way I will know all parts are working as expected and the RSSI reports will let me know if battery voltages of sensors are declining.
3) additionally, when placing or moving sensors I can check that the central control system is receiving a reasonable signal strength. For example, I have two sets of panic buttons and I can already see that the one set only has marginal coverage over the entire property while the other set has excellent coverage.

All in all this is a very useful enhancement to rtl_433. All that is left is to persuade you not to deprecate the -a mode since that seems to be one of the great strengths of rtl_433. I have some more ideas about that but lets deal with one thing at a time.

Thanks for all your work, much appreciated.

zuckschwerdt, 2018-11-11:
Thanks for the feedback. Maybe you can get some info from the SNR. It should rise and fall with the automatic gain adaption, i.e. a high automatic gain means a higher noise floor and lower SNR.
I'll remember not to axe the `-a` lightly ;)

closed 2018-11-16 by zuckschwerdt

roger-, 2018-12-28:
I was trying to use the RSSI value to quickly compare the performance of two antennas, but the reported noise level changes by 8 dB between the two even though I have the gain fixed to 50 dB.

I would think the noise floor should be more consistent? How is it calculated?

zuckschwerdt, 2018-12-28:
It's the estimated amplitude in dBFS, absolute to the received values but relative to the gain setting.

roger-, 2018-12-28:
Does the noise level get estimated from the gaps in the signal, or before/after the signal?

I don't know why it would change with a fixed gain.

zuckschwerdt, 2018-12-29:
Perhaps because the antenna is more sensitive. If you unplug the antenna you should get zero noise.
The noise level is estimated in the gaps.

roger-, 2018-12-29:
The antenna shouldn't affect the noise though. I'll try to look into this, seems like something is off.

zuckschwerdt, 2018-12-29:
Why shouldn't it? It's the source of signal and noise for the receiver. Perhaps you mean something different with "noise"? It's the mean amplitude on the receiver while no data is detected.

roger-, 2018-12-29:
I mean AWGN (noise). 

Assuming the bandwidth is the same for both cases, the LNA/front end noise figure will determine the noise level; see [here](
https://en.m.wikipedia.org/wiki/Minimum_detectable_signal). 

Do you mean "mean square" amplitude? This value should be non-zero even with no antenna.

roger-, 2018-12-29:
Took a quick look at the code and it seems that the noise level is basically the lower threshold in the OOK detector? Seems that it should be the average of the envelope values *under* the threshold. I'm not sure how the threshold is found in the first place though.

zuckschwerdt, 2018-12-29:
See https://github.com/merbanan/rtl_433/blob/master/src/baseband.c#L49 (CU8), https://github.com/merbanan/rtl_433/blob/master/src/baseband.c#L61 (CS16) and https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L322
The value is Q0.14 (i.e. full scale "1.0" is 16384) in both cases. I'm not confident in the level estimates from the pulse_detect, but it should be close enough to what the decoder "sees".

beaune33, 2021-04-11:
After reading this discussion it is not completely clear for me what was implemented. What I would prefer ist an option to provide the RSSI information for each decoding procedure in the same format as the readings itself, e.g. an RSSI json tag. Is it already possible to configure that? Or maybe this is an option to implement? That would really help to have a clear option to see, if the signal strength is still good, maybe also to provide warnings if something has changed.

zuckschwerdt, 2021-04-11:
With `rtl_433 -M` you'll get help on meta data. You are looking for `-M level` adds Modulation, Frequency, RSSI, SNR, and Noise meta data.