merbanan/rtl_433 #860: [Slightly OT] Transmitter debugging with rtl433?

gcormier, 2018-11-11:
I'm trying to spoof the PMV-107J. 

What would be the best way to use rtl_433 to debug my transmitter? Using -A and -D functions is what I've been trying so far. Are there any other functions that dump data? Or any way to view/open the cu8 files it captures in other tools?

With those two flags, I've yet to get a match with a real sensor. I believe in #821 , the half-bit period was 100uS. Assuming that is in reference to the time between the transitions in a Manchester encoding, I'm getting 200uS from -A.

With a 50uS delay for each bit (clock and data), I hit the expected output for pulse width. However the gap width is still off.

Output from a REAL sensor using -A:
```
Total count:   56,  width: 15.71 ms		( 3928 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 1] count:    1,  width:  612 us [612;612]	( 153 S)
 [ 2] count:   38,  width:  100 us [100;104]	(  25 S)
 [ 3] count:   15,  width:  196 us [164;204]	(  49 S)
 [ 4] count:    1,  width:   56 us [56;56]	(  14 S)
Gap width distribution:
 [ 0] count:    2,  width:  684 us [628;740]	( 171 S)
 [ 1] count:   15,  width:  200 us [200;204]	(  50 S)
 [ 2] count:   38,  width:  100 us [100;104]	(  25 S)
Pulse period distribution:
 [ 0] count:    1,  width:  628 us [628;628]	( 157 S)
 [ 1] count:    2,  width:  856 us [812;904]	( 214 S)
 [ 2] count:   32,  width:  200 us [200;204]	(  50 S)
 [ 3] count:    8,  width:  400 us [400;404]	( 100 S)
 [ 4] count:   12,  width:  300 us [300;304]	(  75 S)
Level estimates [high, low]:  15744,     30
Frequency offsets [F1, F2]:    1811, -18183	(+6.9 kHz, -69.4 kHz)
```

Output from spoofer:
```
Detected FSK package	@ 2018-11-11 12:13:45
Analyzing pulses...
Total count:   56,  width: 69.17 ms		(17293 S)
Pulse width distribution:
 [ 0] count:    1,  width: 49880 us [49880;49880]	(12470 S)
 [ 1] count:   20,  width:  200 us [176;240]	(  50 S)
 [ 2] count:   17,  width:   56 us [48;68]	(  14 S)
 [ 3] count:   10,  width:   40 us [40;48]	(  10 S)
 [ 4] count:    8,  width:  112 us [92;132]	(  28 S)
Gap width distribution:
 [ 0] count:   19,  width:  300 us [272;320]	(  75 S)
 [ 1] count:   28,  width:  200 us [168;224]	(  50 S)
 [ 2] count:    6,  width:  148 us [128;160]	(  37 S)
 [ 3] count:    1,  width:  116 us [116;116]	(  29 S)
 [ 4] count:    1,  width:  560 us [560;560]	( 140 S)
Pulse period distribution:
 [ 0] count:    1,  width: 50188 us [50188;50188]	(12547 S)
 [ 1] count:   18,  width:  380 us [336;428]	(  95 S)
 [ 2] count:   26,  width:  252 us [236;276]	(  63 S)
 [ 3] count:    9,  width:  508 us [492;516]	( 127 S)
 [ 4] count:    1,  width:  760 us [760;760]	( 190 S)
Level estimates [high, low]:  15938,    383
Frequency offsets [F1, F2]:    1905,  -5305	(+7.3 kHz, -20.2 kHz)
Guessing modulation: No clue...
```

merbanan, 2018-11-11:
Use rtl_433/tests/sigrok-conv.sh on a file. Then use pulseview to look at the original signal and the cloned one.

gcormier, 2018-11-11:
Ah man, my RF signal looks like crap. But still seems to do a decent job of picking out the modulation. 

[image]

This is supposed to be
> 00000000 11111111 01010101 10101010

zuckschwerdt, 2018-11-12:
You need to zoom I, Q, and AM by clicking the labels and selecting, maybe 0,5V/div. I/Q, along with AM will tell you the most about your signal.
Also for a simple working signal generator use perhaps https://github.com/zuckschwerdt/tx_tools/
Look at https://github.com/zuckschwerdt/tx_tools/blob/master/examples/fsk.txt to easily create a FSK.

gcormier, 2018-11-12:
This is awesome! Thanks so much. I've got the "perfect" output generated with your toolset so I can start comparing what I'm getting with my setup. Firstly, ensuring my DMC algorithm is OK. Second, I think it's probably my sloppy TX - I copied the reference design of the MICRF112 evaluation board. (A v2 might be better to use a seperate antenna rather then try to get into PCB antenna design.) Finally, ensuring the Arduino toolset isn't messing with my setup with it's predefined interrupts. I'm disabling them, but I might have to go native AVR just to rule that out.

A question - if the packet is 66 bits, why do we transmit these 18 bytes?
`{fcb32b32ab4b2d554d52aad4ccb4aaacb52c}`

I've noticed the same behavior in the output of rtl_433. 
```
pulse_demod_pcm(): PMV-107J (Toyota) TPMS 
bitbuffer:: Number of rows: 1 
[00] {144} fc b3 2b 32 ab 4b 2d 55 4d 52 aa d4 cc b4 aa ac b5 2c 

```

zuckschwerdt, 2018-11-12:
I've updated the commit to include a DMC transform. (You need to `git reset --hard HEAD^ ; git pull`).
The available transforms for the payload are ASCII, HEX (the default), DMC, MC (GE Thomas), IMC (inverted or IEEE MC).

The actual payload here is 68 bits which encode to 136 DMC (half-)bits plus a preable of (DMC invalid) 0xFC (i.e. 11111100) -- and that's 144 bits / 18 bytes ;)

gcormier, 2018-11-12:
Ah, so it's encoded into it's own bits that represent DMC - gotcha. 

So 0xFC is not DMC encoded (jives with what I saw in python code on the other repository which did some work on PMV107J).

I guess for my purposes of bit-banging/spoofing (for now), this is just 600uS of high and 200uS of low.

Is there a simple way to combine multiple sources of data into sigrok, so I can overlap each channel?

This is a huge step forward... can probably get this mostly banged out today if the hardware side of things is up to the task :)

gcormier, 2018-11-12:
Ding ding ding :)

Reading samples in async mode...
Tuned to 315.050MHz.
```
2018-11-12 12:40:01 :	PMV-107J 	:	TPMS 	:	079e15a0 	:	24 	:	OK 	:	3 	:	OK 	:	223.200 	:	24.000 	:	CRC
2018-11-12 12:40:03 :	PMV-107J 	:	TPMS 	:	079e15a0 	:	24 	:	OK 	:	3 	:	OK 	:	223.200 	:	24.000 	:	CRC
2018-11-12 12:40:04 :	PMV-107J 	:	TPMS 	:	079e15a0 	:	24 	:	OK 	:	3 	:	OK 	:	223.200 	:	24.000 	:	CRC
```
Need to do some hardware work to get that frequency closer to 315.0MHz I suspect. I'll hardcode the other 3 sensors and see what happens in reality with the car though.

closed 2018-11-13 by gcormier