merbanan/rtl_433 #861: Brennenstuhl RCS 2044 outputs pointer to array instead of system code

jens-nolte, 2018-11-14:
The system code for Brennenstuhl RCS 2044 is not reacting to changes on the dip switches, but instead has a random id that is unique for every rtl_433 run. Upon investigation I found, that the code is generated correctly...
```
int system_code[] =
{
    b[0]>>6 & 1,
    b[0]>>4 & 1,
    b[0]>>2 & 1,
    b[0]    & 1,
    b[1]>>6 & 1
};
```
... but then I think the array address is passed to the output data.
```
 data = data_make(
        "time",          "Time",        DATA_STRING, time_str,
        "model",         "Model",       DATA_STRING, "Brennenstuhl RCS 2044",
        "type",         "Type",         DATA_STRING, "Remote Control Set",
	"id",		"id",		DATA_INT, system_code,
	"key",		"key",		DATA_STRING, control_key[0] ? "A" : (control_key[1] ? "B" : (control_key[2] ? "C" : (control_key[3] ? "D" : "E" ))), 
        "state",	"state",        DATA_STRING, (on ? "ON" : (off ? "OFF" : "BOTH")),  
        NULL);
```

[src/devices/brennenstuhl_rcs_2044.c#L159](https://github.com/merbanan/rtl_433/blob/5e6dde1b9b91fff575dbc638ac8dc94aaaa0c681/src/devices/brennenstuhl_rcs_2044.c#L159)

zuckschwerdt, 2018-11-14:
You are spot on, that's a bug. It's fixed with bc7b5c8. Please close if everything works!

jens-nolte, 2018-11-14:
Bug is fixed, thanks for the fast response time!

closed 2018-11-14 by jens-nolte