merbanan/rtl_433 #873: Command line arguments not parsed

ericstaal, 2018-11-19:
Used the lastest version and build in clean on a rapsberry pi.

WHen execute via rtl_433 is does work and receives values

short output:
`Registered 90 out of 112 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433.920MHz.
2018-11-19 22:04:56 :   HIDEKI TS04 sensor
        Rolling Code:    12
        Channel:         1
        Battery:         OK
        Temperature:     3.3 C
        Humidity:        79 %
`

when adding '-F csv' for example:
`pi@raspberrypi:~ $ rtl_433 -F csv
rtl_433, an ISM band generic data receiver for RTL2832 based DVB-T receivers
version unknown

Usage:  = Tuner options =
`

newtewt, 2018-11-20:
Adding a +1 to this. Trying to use `rtl_433 -F json` . Tried adding quotes around and no luck.

zuckschwerdt, 2018-11-20:
@newtewt thanks for pointing to the right place, fixed with 6306b59.

closed 2018-11-30 by zuckschwerdt