merbanan/rtl_433 #876: output strings include partial messages

bsculley, 2018-11-20:
I am getting partial messages in stdout.  This is true for both regular strings and json formatted data.  In the case of json, it makes every message invalid because of formatting.  It looks like the data for the whole messages is getting sent to data_acquired_handler, but when it is output it's fragmented.  Also, data from multiple messages are sometimes combined in the same string.  It looks like a buffering problem, but I can't see where to fix it.
Here is an example:
```
:59", "model" : "Acurite 5n1 sensor", "sensor_id" : 2838, "channel" : "B", "sequence_num" : 1, "battery" : "OK", "message_type" : 56, "wind_speed_mph" : 8.337, "temperature_F" : 45.900, "humidity" : 94}
{"time" : "2018-11-20 16:11:59", "model" : "Acurite 5n1 sensor", "sensor_id" : 2838, "channel" : "B", "sequence_num" : 2, "battery" : "OK", "message_type" : 56, "wind_speed_mph" : 8.337, "temperature_F" : 45.900, "humidity" : 94}
```

zuckschwerdt, 2018-11-20:
On which version (`-V`) are you? What OS is this running on? Does this also happen with output to a file (`-F json:out.json`)?

bsculley, 2018-11-20:
Hi, thanks for your response.  I am running this on Debian Linux 9 (4.14.70-sunxi on armv7l).  The hardware is Orange Pi PC Plus.
rtl_433 -V reports "version unknown".  No idea why.  I downloaded the master branch yesterday,

This is not a problem when the data is written to a file.

zuckschwerdt, 2018-11-20:
Version is only set if you compile with git around, so that's expected ;)
If you redirect to a file (`rtl_433 -q -F json >file.out`), are the messages chopped of too?

bsculley, 2018-11-20:
The output looks good when redirected.  I also looked at stdout on the console and it's good there too.  
I am intercepting stdout with a Node-Red "daemon" node which I discovered I had misconfigured.  Works great now.  Thanks for the suggestions, and sorry for the false alarm.

closed 2018-11-20 by zuckschwerdt