merbanan/rtl_433 #880: Issue with app_cfg struct and config file parser

winterrace, 2018-11-21:
I played around a bit with the new config file feature (simply using the new example configuration file).

First: It's a nice implementation, I like the recursive nature which allows the existing option handlers to fill the config structure (no matter if they are fed by command line or config file).

But I found another buggy effect:

When calling rtl_433.exe -c rtl_433.example.conf the following stuff will happen in the following order:

- parse_conf_option handles case 'c' and calls parse_conf_file
- parse_conf_file calls readconf to read the config file contents to a new buffer
- parse_conf_file calls parse_conf_text to evaluate the config file

during this parsing loop, parse_conf_option will be invoked again, now on the stuff taken from the config file, e.g.:

    case 'd':
        if (!arg)
            help_device();

        cfg->dev_query = arg;
        break;

Here, cfg->dev_query is set to the device query string specified in the config file (which is the string "0", i.e. "use the first RTL-SDR device")
After all options from the config file have been processed, we find us again in the bottom part of parse_conf_file:
- parse_conf_file frees the config file buffer

Shortly afterwards we find ourselves back in main which tries to:

```
 r = sdr_open(&cfg.dev, &demod->sample_size, cfg.dev_query, !cfg.quiet_mode);
  if (r < 0) {
            exit(1);
        }

```

Here, the program unexpectedly fails and exits.
This is due to cfg.dev_query is just a pointer which was set within the config file buffer. However, this buffer has been freed in meantime and (in my case) contains illegal data now.

As a more robust and failsafe alternative, I'd suggest to avoid most pointers in the app_config struct and use buffers there, like this:

```
struct app_cfg {
    char dev_query[MAX_QUERY_LEN]; // former: char *dev_query;
```

So the corresponding option handler should look like this:

    case 'd':
        if (!arg)
            help_device();

        strcpy(cfg->dev_query, arg);
        break;

I expect the same will be necessary for the other app_cfg members...
    char *gain_str;
    char const *test_data;
    char const *in_filename;

winterrace, 2018-11-21:
By the way, I'm confident that I can upload my librtl_433 very soon (maybe 1 week to go...).
If you like you can have a look on it then...

Despite my first plans, it still remained a pure C library, but in an object-oriented manner without any global variables. I will mainly use it for my GUI-based rtl_fl2k_433 application but it's still possible to compile a command line version of rtl_433 against this lib which should works exactly the same like the current one. And the lib still compiles for both, Linux and Windows...

However, many of the structural changes I experimented with have been integrated to rtl_433 in parallel in a similar fashion ;-) But I will post some comments along with it, so maybe you find some useful features... :)

zuckschwerdt, 2018-11-22:
When I'm done with the current set of additions, I'm going to add a list_t with clean semantics of push() // takes ownership, and push_copy(). That will replace most of the random list-like stuff.

gdt, 2023-09-25:
Is this still a valid bug with the current code?

zuckschwerdt, 2023-10-04:
This was solved with ff6a43f and there is a TODO note in the code to someday further refine this.

closed 2023-10-04 by zuckschwerdt