merbanan/rtl_433 #889: Consider adding RFLink supported protocols

Piruzzolo, 2018-12-01:
[Here](http://www.rflink.nl/blog2/) fot the project.
See [here](https://github.com/Pirionfr/RFLink-Gateway) for the latest source code available. 

It features documentation for multiple protocols

Hope it helps 👍  (Btw, I'm trying to reverse an Avidsen 433 MHz temperature sensor 👷 )

zuckschwerdt, 2018-12-01:
Sorry, isn't RFLink proprietary? We can't take code from there. If you can't figure out a device or want to be sure to get things right use e.g. [Clean room design](https://en.wikipedia.org/wiki/Clean_room_design) – tell us about the device data and we'll implement a device.

closed 2018-12-01 by zuckschwerdt

zuckschwerdt, 2018-12-01:
I'm not sure about the license terms exactly, I guess it would be okay to take hints from the comments to each protocol.
We are working on a comprehensive guide to develop new decoders. The short is to grab signals, look at the `-A` guesses, refine the flex suggestion, put a few code lines into BitBench and figure out a format description there.
The timings as flex parameters and a format description should be all that's needed.
I'm working on a way to use the BitBench format string in flex directly. Adding decoders with a short declarative configuration is the goal.

Piruzzolo, 2018-12-01:
I've found the current development tree, and in the license it says 

> The source code is free for use in any open source project when the original copyright notices, headers and this license file are included. 
> This program is free software: you can redistribute it and/or modify it when the original copyright notices and headers are included. 

So it sould be ok (but see the full license). [Here](https://drive.google.com/file/d/0BwEYW5Q6bg_ZTDhKQXphN0ZxdEU/view)'s the source (how I've found it: [here](https://www.letscontrolit.com/forum/viewtopic.php?p=5078#p5078) -> broken link -> internet archive -> latest copy -> google drive's link not dumped by internet archive -> manual paste link to address bar)

Anyway, I think that just using the header and the comments would be fine for sure.

---------------------

> We are working on a comprehensive guide to develop new decoders. The short is to grab signals, look at the -A guesses, refine the flex suggestion, put a few code lines into BitBench and figure out a [...]

A neat declarative approach is really nice!

zuckschwerdt, 2018-12-01:
Channel settings on sensors very likely just set some bits in the data. Also it might alter the transmission interval (to co-prime values among the channels for least overlap).

Piruzzolo, 2018-12-01:
Hummm the sensor has a selector in the back with channel 1, 2, 3. We'll see when I'll begin reversing it 😄 . Thanks very much, BTW! 👍

zuckschwerdt, 2018-12-01:
Maybe also look through [the notes](https://github.com/merbanan/rtl_433/blob/362e9f6cce815dbc0ce323a5f4b9df0680beca4a/src/devices/digitech_xc0324.README.md) from #857 for ideas.

Piruzzolo, 2018-12-01:
That's great! That's a very good starting point! 
Such an awesome community, thanks! 🎉

Geoff99, 2018-12-02:
Hi @Piruzzolo 
Just a quick heads up / warning.  I'm working on updating the "newbie tutorial" at the moment, to bring it in line with a number of enhancements that have been made in the last week or so to rtl_433.  The enhancements make it even easier to develop new decoders !  

But in the interim, please be aware that are also a few "renames" of functions and parameters that have happened, the nett effect of which is that parts of the tutorial, as it stands **today**, could be slightly but annoyingly wrong (eg `OOK_PULSE_PPM_RAW` is now called `OOK_PULSE_PPM`; and all the bash scripting stuff to produce csv files with filename labels included is obsolete - just use the `-K` option when you run rtl_433.)

Best of luck with your deciphering - I had lots of fun and learnt quite a bit about C from my attempts, (aided a lot by the many helpful tips and suggestions offered here).

Piruzzolo, 2018-12-02:
Thank you for the advice! :smile: and of course, I'll read the step by step
guide thoroughly