merbanan/rtl_433 #892: Portable Android signal recorder

merbanan, 2018-12-03:
Someone ported rtl-sdr to Android and run it via loop back as a rtl_tcp instance (RTL2832U driver). It would be nice if we could add support for rtl_tcp and then write an android app that runs rtl_433 for signal grabbing purposes.

zuckschwerdt, 2018-12-03:
I also just refactored the sample grabber out (no other changes). At a glance the ring buffer logic seemed somewhat broken. I'll polish that up and provide the grabbing to `-A` too.

merbanan, 2018-12-03:
Everything is broken :) its just barely holding together.

zuckschwerdt, 2018-12-03:
Like they say "If it doesn't work, whack it. If it breaks it needed fixing anyway."
Some technical debt crept in with all the additions, but the current refactoring is working well :)

zuckschwerdt, 2018-12-04:
There is some basic rtl_tcp support in #894.

dobrou, 2019-06-14:
FYI, rtl_433 works perfectly on Android.
While there is no standalone rtl_433 Android app, rtl_433 can be successfully cloned, compiled and executed in Termux app.

Install:
- SDR driver: https://play.google.com/store/apps/details?id=marto.rtl_tcp_andro
- Termux: https://play.google.com/store/apps/details?id=com.termux
- AnLinux: https://play.google.com/store/apps/details?id=exa.lnx.a

- load Termux
- with AnLinux, install Ubuntu in Termux
- update Ubuntu installation (apt update)
- in Termux Ubuntu, follow build steps from rtl_433 build guide (git clone ..., apt install ..., cmake..., make install, ...)
- start "SDR driver" app and in Advanced mode start the rtl_tcp server
- in Termux Ubuntu, start rtl_433 with proper rtl_tcp params

Known issues:
- rtl_433 reports frequency and sample rate as 0, but signal detection works. I do not know whether it is bug in tcp driver or in rtl_433 app.

zuckschwerdt, 2019-06-14:
Nice, thanks! Portability, low resources and dependencies is a priority for us. Great to hear this works then without need for any patches!
Perhaps binaries can be cross-compiled for distribution or inclusion with a UI? Would be great if someone can explore this. The CI can then be extended to deliver snaphot builds.

UNOSweeper, 2020-03-10:
Thank you very much for using RTL_433!
I am using it on a Windows 10 (64bit) platform and let it run for a couple of hours, writing the data to a file using -F json:logfile.json. Then I extract the data of my sensor, put it into a worksheet of OpenOffice to make the graph.
In this topic I came across the procedure, that Dobrou described, to run RTL_433 on  Android which is also very interesting for me, because I let collect the data on my Andoid (version 9) tablet and doing  the analysis on my PC.  But as a beginner I need a little bit help to get it managed on my tablet.
On my tablet I have downloaded and installed successfully the SDR driver, Termux and AnLinux using the exact links Dobrou mentioned.
#  I have opened Termux shell, then I am lost how to "install Ubuntu via command from AnLinux"
Then "# install dependencies, compile and install rtl_433" . I hope I can get by with "...or follow detailed instruction from rtl_433 build guide".
I hope that I also may tell you later "FYI, rtl_433 works perfectly on Android."
Thank you for your help in advance.

uriva9w, 2020-10-27:
> FYI, rtl_433 works perfectly on Android.
> While there is no standalone rtl_433 Android app, rtl_433 can be successfully cloned, compiled and executed in Termux app.
> 
> Install:
> 
> * SDR driver: https://play.google.com/store/apps/details?id=marto.rtl_tcp_andro
> * Termux: https://play.google.com/store/apps/details?id=com.termux
> * AnLinux: https://play.google.com/store/apps/details?id=exa.lnx.a
> * Note: all apps also available in F-Droid store
> 
> ```
> # open Termux shell
> # install Ubuntu via command from AnLinux
> # start ubuntu
> ./start-ubuntu.sh
> # install dependencies, compile and install rtl_433 or follow detailed instruction from rtl_433 build guide
> apt update
> apt install git sudo libtool libusb-1.0-0-dev librtlsdr-dev rtl-sdr build-essential autoconf cmake pkg-config
> git clone https://github.com/merbanan/rtl_433
> cd rtl_433/
> mkdir build
> cd build
> cmake ..
> make
> make install
> ```
> 
> Run rtl_433
> 
> ```
> # manually start "SDR driver" app and in Advanced mode start the rtl_tcp server
> # or start the driver automatically from Termux shell via Intent command:
> am start "iqsrc://-a 127.0.0.1 -p 14423 -d 443920000 -s 1024000"
> sleep 4
> # run the rtl_433
> ~/start-ubuntu.sh rtl_433 -C si -G -M newmodel -d rtl_tcp:127.0.0.1:14423
> ```
> 
> BTW steps above should work for any other rtl_tcp compatible tool.
> 
> Tested on:
> 
> * Redmi 4X - Android 7.1.2
> 
> Known issues:
> 
> * rtl_433 reports frequency and sample rate as 0, but signal detection works. I do not know whether it is bug in tcp driver or in rtl_433 app.
> 
> [image]

M Dobrou
I followed the steps to make work rtl 433 in my android phone, install termux, install AnLinux, Ubuntu, SDR driver..with no results
Could you please send me more details? Regards

adewilt8, 2022-08-01:
Anyone recently succeeded installing / using rtl_433 on Android. 
Above installation fails quicky at the first few steps on Pixel 6.

uriva9w, 2022-08-03:
Hi
I followed thé instructions given in thé forum and I had success.
Have any questions I’ll try to help you.
My android bench was a very cheap  phone.
Regards

Msprg, 2022-08-18:
> FYI, rtl_433 works perfectly on Android. While there is no standalone rtl_433 Android app, rtl_433 can be successfully cloned, compiled and executed in Termux app.
>
> Install:
>
>     * SDR driver: [...]

For those having problems with Termux, Install it ONLY from F-Droid! The one from Google Play does not work for a long time now, and likely - never will. Try installing all you need from F-Droid, using Play store only as a fallback.

dobrou, 2022-08-19:
Hi,
it is nice this thread is still alive after so long time.
Since then, Termux added own support for Ubuntu overlay, that makes few steps easier.

Here is updated and simplified manual.
Feel free to add your improvements and suggestions.

**Install apps:**
- Termux:
  - https://f-droid.org/en/packages/com.termux/
  - Use the F-Droid source, do not install from Google Play as their version is broken !
- SDR driver: 
  - https://play.google.com/store/apps/details?id=marto.rtl_tcp_andro
  - OR https://f-droid.org/en/packages/marto.rtl_tcp_andro/

**Build rtl_433:**
```
# open Termux shell
pkg update
pkg install proot-distro
# install and login into Ubuntu overlay in Termux
proot-distro install ubuntu
proot-distro login ubuntu

# inside Ubuntu overlay install dependencies and compile the rtl_433
apt -y update
apt -y install git sudo libtool libusb-1.0-0-dev librtlsdr-dev rtl-sdr build-essential autoconf cmake pkg-config
git clone https://github.com/merbanan/rtl_433
cd rtl_433/
mkdir build
cd build
cmake ..
make
make install
```

**Run rtl_433**
```
# From Termux shell:
# 1) manually start "SDR driver" app and in Advanced mode start the rtl_tcp server
# or start the driver automatically from Termux shell via Intent command:
am start "iqsrc://-a 127.0.0.1 -p 14423 -d 443920000 -s 1024000"
sleep 4
# 2) run the rtl_433
proot-distro login ubuntu -- rtl_433 -d rtl_tcp:127.0.0.1:14423
```
BTW similar steps should work for any other rtl_tcp compatible tool.

**Tested on:**
- Pixel 4a - Android 13
- Galaxy A32 - Android 12
- Redmi 4X - Android 7.1.2

**Known issues:**
- rtl_433 reports frequency and sample rate as 0, but signal detection works. I do not know whether it is bug in tcp driver or in rtl_433 app.

adewilt8, 2022-08-19:
Thanks for the info guys, I.managed to get it working. Apparently the Termux from the PlayStore caused the issue before...

adewilt8, 2022-08-19:
But it would still be lovely when someone build an Android app to access the 433 analyser. :-)

Crsarmv7l, 2022-12-13:
Thought I would provide a small update.

You still need termux from f-droid, as well as the driver app.
You also still need a proot distro, I use debian.

You do not need to compile it from source. Apt install rtl-433 works just fine.

I set up: 
"am start "iqsrc://-a 127.0.0.1 -p 14423 -d 443920000 -s 1024000"
sleep 2"
in my termux .profile along with "proot-distro login debian"

I also set up an alias for "d rtl_tcp:127.0.0.1:14423"

So I plug in the rtl-sdr, launch termux, which starts the rtl-sdr driver then logs into debian. 

At the cli I type: "rtl_433" annd whatever args I want -f 315.00M, etc and my alias.

It is very streamlined and handy

adewilt8, 2022-12-27:
> Thought I would provide a small update.
>
> You still need termux from f-droid, as well as the driver app. You also still need a proot distro, I use debian.
>
> You do not need to compile it from source. [...]

Thanks for the update!

gdt, 2023-09-25:
@Crsarmv7l Is it appropriate for you to create a PR with a README-android.md or some such that explains how to do this, and then we can close this PR?   Where is this "driver app"?  Should we do something else?

Crsarmv7l, 2023-09-26:
The driver is "rtl-sdr driver" in f-Droid.

The whole setup is basically explained in two sets of docs. Termux covers
proot, and accessing a network rtl-sdr is covered in quite a few places,
which is what you are doing with the driver. Everything else is just basic
Linux.

IMO I would just close this

gdt, 2023-09-26:
@dobrou Is there more than the link to rtl_sdr that we should harvest from this issue?  Issues are obviously not an ok place to store docs.  In #2644 I point to the driver, and ideally the driver would have enough docs.  (I don't really understand why there is proot; termux lets you run programs and once rtl_tcp is running rtl_433 is just a program.)

Crsarmv7l, 2023-09-26:
You need the proot to be able to install rtl_433, otherwise you are
building it.. And folks that can't figure these things out would probably
do better with a one line proot install, then installing rtl_433 from the
repo

gdt, 2023-09-26:
Can you figure out where to document this (in rtl_433, or someplace else) and make a PR?  open issues are not an acceptable long-term place for docs to live :-)    But if you are saying that if you just build it, you don't need any of that, and if you want to run ubuntu packages, then you need to do ubuntu stuff to install, then that isn't about rtl_433 at all.    Given that the driver pointer is in a PR, I'm going to call this issue completed.

PR #2644 (2023-09-26): doc: Add rtl_sdr driver pointer
This adds a pointer to an Android package that implements rtl_tcp.

Fixes #892
closed 2023-09-26 by gdt