merbanan/rtl_433 #895: UART data decoding

wiseholder, 2018-12-04:
I have a beacon, that consists from GPS module, that transmit text data in NMEA format through UART interface and 433,9 MHz ASK transmitter, connected directly to TX pin of GPS module.
I want to decode this data, received on RTL-SDR dongle. How can I use rtl_433 for this purpose?
Actual data, that transmits looks like:
```
$GPRMC,161453.00,V,,,,,,,041218,,,N*77
$GPVTG,,,,,,,,,N*30
$GPGGA,161453.00,,,,,0,00,99.99,,,,,,*62
$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30
$GPGSV,3,1,10,05,22,239,,07,34,091,,08,23,044,,11,11,090,*7F
$GPGSV,3,2,10,13,55,294,,15,21,310,22,17,10,161,,27,03,021,24*73
$GPGSV,3,3,10,28,74,174,20,30,68,085,25*76
$GPGLL,,,,,161453.00,V,N*4E
```
Such packets goes one per second. Baud rate is 9600, 8 data bits, 1 stop bit, no parity.

rtl_433 -A gives:
```
Detected OOK package	@ 2018-12-04 20:27:10
Analyzing pulses...
Total count: 1200,  width: 997.30 ms		(249325 S)
Pulse width distribution:
 [ 0] count:  795,  width:   76 us [64;92]	(  19 S)
 [ 1] count:   81,  width:  284 us [276;300]	(  71 S)
 [ 2] count:  323,  width:  180 us [168;200]	(  45 S)
 [ 3] count:    1,  width: 573596 us [573596;573596]	(143399 S)
Gap width distribution:
 [ 0] count:   77,  width:  508 us [436;556]	( 127 S)
 [ 1] count:  429,  width:  128 us [116;140]	(  32 S)
 [ 2] count:  218,  width:  336 us [328;352]	(  84 S)
 [ 3] count:  475,  width:  232 us [224;248]	(  58 S)
Pulse period distribution:
 [ 0] count:  134,  width:  576 us [512;632]	( 144 S)
 [ 1] count:  262,  width:  204 us [196;220]	(  51 S)
 [ 2] count:  394,  width:  416 us [408;428]	( 104 S)
 [ 3] count:  408,  width:  312 us [304;324]	(  78 S)
 [ 4] count:    1,  width: 573928 us [573928;573928]	(143482 S)
Level estimates [high, low]:  15057,     29
RSSI: -0.4 dB SNR: 27.0 dB Noise: -27.5 dB
Frequency offsets [F1, F2]:     439,      0	(+1.7 kHz, +0.0 kHz)
Guessing modulation: No clue...
```
rtl_433 -a gives message "To many pulses detected, probably bad input data or input parameters"
Some files, that I obtained by executing rtl_433 -a -t:
[g001_433.88M_250k.zip](https://github.com/merbanan/rtl_433/files/2645133/g001_433.88M_250k.zip)

zuckschwerdt, 2018-12-04:
Looks like there is a continous carrier. That will likely get you in trouble as this device "blocks" all other 433.92 MHz devices. Look at http://triq.net/iqs -- the white straight white line at 433.88 is the problem.
Try *not* to tune exactly to the signal center frequency (using `-f 433.88M`). Better use the default (of 433.92M).
Also capture with a higher sample rate (`-s 1024k`).

zuckschwerdt, 2018-12-04:
Thinking about what you describe you perhaps got the levels (UART TX to ASK transmitter) wrong. The transmitter always sends and just pauses when there is a `1` bit.

Even if you fix that, utilizing the channel at 40% (sending for 600ms each second) is way to much. Regulatory limits would allow 10% at most and usually devices stay well below 1%.

You should use a much higher bitrate for shorter bursts, something like an CC1101 will prove much better than some cheap ASK module.

Hope this helps you to improve things.

closed 2018-12-04 by zuckschwerdt