merbanan/rtl_433 #897: error building for windows

float23, 2018-12-06:
i am following the instructions but when i run "mingw32-make" if gives errors how can i fix this

`D:\Downloads\librtlsdr-master>mkdir build && cd build

D:\Downloads\librtlsdr-master\build>cmake -G "MinGW Makefiles" ..
-- The C compiler identification is GNU 8.1.0
-- Check for working C compiler: C:/Program Files (x86)/mingw-w64/i686-8.1.0-posix-dwarf-rt_v6-rev0/mingw32/bin/gcc.exe
-- Check for working C compiler: C:/Program Files (x86)/mingw-w64/i686-8.1.0-posix-dwarf-rt_v6-rev0/mingw32/bin/gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Extracting version information from git describe...
fatal: not a git repository (or any of the parent directories): .git
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Looking for libusb_handle_events_timeout_completed
-- Looking for libusb_handle_events_timeout_completed - not found
-- Looking for libusb_error_name
-- Looking for libusb_error_name - not found
-- Found libusb-1.0: C:/Program Files (x86)/mingw-w64/i686-8.1.0-posix-dwarf-rt_v6-rev0/mingw32/include/libusb-1.0, C:/Program Files (x86)/mingw-w64/i686-8.1.0-posix-dwarf-rt_v6-rev0/mingw32/lib/libusb-1.0.dll.a
-- Udev rules not being installed, install them with -DINSTALL_UDEV_RULES=ON
-- Building with kernel driver detaching disabled, use -DDETACH_KERNEL_DRIVER=ON to enable
-- Building for version:  / 0.6git
-- Using install prefix: C:/Program Files (x86)/rtlsdr
-- Configuring done
-- Generating done
-- Build files have been written to: D:/Downloads/librtlsdr-master/build

D:\Downloads\librtlsdr-master\build>mingw32-make
Scanning dependencies of target rtlsdr_shared
[  3%] Building C object src/CMakeFiles/rtlsdr_shared.dir/librtlsdr.c.obj
[  6%] Building C object src/CMakeFiles/rtlsdr_shared.dir/tuner_e4k.c.obj
[  9%] Building C object src/CMakeFiles/rtlsdr_shared.dir/tuner_fc0012.c.obj
[ 12%] Building C object src/CMakeFiles/rtlsdr_shared.dir/tuner_fc0013.c.obj
[ 15%] Building C object src/CMakeFiles/rtlsdr_shared.dir/tuner_fc2580.c.obj
[ 18%] Building C object src/CMakeFiles/rtlsdr_shared.dir/tuner_r82xx.c.obj
[ 21%] Linking C shared library librtlsdr.dll
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x2e5): undefined reference to `libusb_submit_transfer@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x533): undefined reference to `libusb_control_transfer@32'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x57f): undefined reference to `libusb_control_transfer@32'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x67a): undefined reference to `libusb_control_transfer@32'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x6c6): undefined reference to `libusb_control_transfer@32'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x88e): undefined reference to `libusb_control_transfer@32'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x902): more undefined references to `libusb_control_transfer@32' follow
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x1f4a): undefined reference to `libusb_get_device@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x1f5d): undefined reference to `libusb_get_device_descriptor@8'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x1fb3): undefined reference to `libusb_get_string_descriptor_ascii@16'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x2002): undefined reference to `libusb_get_string_descriptor_ascii@16'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x2066): undefined reference to `libusb_get_string_descriptor_ascii@16'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x212b): undefined reference to `libusb_control_transfer@32'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x216d): undefined reference to `libusb_control_transfer@32'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x21bc): undefined reference to `libusb_control_transfer@32'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x2280): undefined reference to `libusb_control_transfer@32'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x22e2): undefined reference to `libusb_control_transfer@32'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x27f7): more undefined references to `libusb_control_transfer@32' follow
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x37e1): undefined reference to `libusb_init@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x3800): undefined reference to `libusb_get_device_list@8'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x3822): undefined reference to `libusb_get_device_descriptor@8'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x3876): undefined reference to `libusb_free_device_list@8'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x3885): undefined reference to `libusb_exit@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x38af): undefined reference to `libusb_init@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x38ce): undefined reference to `libusb_get_device_list@8'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x38ff): undefined reference to `libusb_get_device_descriptor@8'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x3961): undefined reference to `libusb_free_device_list@8'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x3970): undefined reference to `libusb_exit@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x39a4): undefined reference to `libusb_free_device_list@8'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x39b3): undefined reference to `libusb_exit@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x39db): undefined reference to `libusb_free_device_list@8'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x39ea): undefined reference to `libusb_exit@4'
[... 21 lines cut ...]
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x43ea): undefined reference to `libusb_free_device_list@8'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x44f4): undefined reference to `libusb_reset_device@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x46ae): undefined reference to `libusb_release_interface@8'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x46bc): undefined reference to `libusb_close@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x46c9): undefined reference to `libusb_exit@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x474b): undefined reference to `libusb_control_transfer@32'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x479e): undefined reference to `libusb_control_transfer@32'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x485a): undefined reference to `libusb_bulk_transfer@24'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x49ab): undefined reference to `libusb_submit_transfer@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x49ee): undefined reference to `libusb_handle_events_timeout@8'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x4a34): undefined reference to `libusb_free_transfer@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x4b01): undefined reference to `libusb_alloc_transfer@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x4b98): undefined reference to `libusb_cancel_transfer@4'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x4baf): undefined reference to `libusb_handle_events_timeout@8'
[... cut ...]
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x4e51): undefined reference to `libusb_control_transfer@32'
CMakeFiles\rtlsdr_shared.dir/objects.a(librtlsdr.c.obj):librtlsdr.c:(.text+0x4ec7): undefined reference to `libusb_control_transfer@32'
collect2.exe: error: ld returned 1 exit status
mingw32-make[2]: *** [src\CMakeFiles\rtlsdr_shared.dir\build.make:167: src/librtlsdr.dll] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:160: src/CMakeFiles/rtlsdr_shared.dir/all] Error 2
mingw32-make: *** [Makefile:129: all] Error 2

D:\Downloads\librtlsdr-master\build>`

Fra298, 2018-12-06:
Maybe recheck everything, it seems Libusb is not installed properly. Compiling fine for me using MinGW.

winterrace, 2018-12-07:
You could also try a native Windows compilation (i.e. no mingw layer) using the bundled Visual Studio project. See vs15 subdirectory for the files and instructions.

I also have pre-compiled binaries in the release section of my repository:
https://github.com/winterrace/rtl_433_win/releases
The last version is from end of october, I will upload a current release soon.

zuckschwerdt, 2018-12-09:
I take it this is solved (or at least sufficiently addressed).

closed 2018-12-09 by zuckschwerdt