merbanan/rtl_433 #906: messages not making it to MQTT broker

sdrasmussen, 2018-12-10:
Hello.  I am trying to send messages to my MQTT broker. My broker works fine (I get messages from two other sources). My RTL_433 is reading my sensors fine; I get valid data on the CLI if I do not try to pipe to MQTT.  However, when trying to pipe to MQTT, I get no messages.  I am using the following command to attempt this:

./home/pi/wrk/rtl_433/build/src/rtl_433 -F json -U -R 32 -f 433926000 | mosquitto_pub -h 192.168.254.201 -p 1883 -u <myuser> -P <mypassword> -t weather/rtl_433 -l

The command runs successfully, however I simply get no messages from this device in my MQTT broker.  What do I have wrong?  Thanks.

zuckschwerdt, 2018-12-10:
Note that I just now deprecated `-U`, use `-M utc` now. There should be no need to set that center frequency, the default of `-f 433.92M` will work.

I would rather use the  rtl_433_mqtt_relay.py as shown here: http://triq.net/articles/rtl-433-networking

zuckschwerdt, 2018-12-10:
The default in the script listens to rtl_433 messages at port 1433. So your rtl_433 line should be:
```
rtl_433 -F syslog:127.0.0.1:1433
```
You can adapt the `mqttc.publish` lines in the script to the MQTT topics you need.
If you need to debug whats coming into the script add `print(line)` below ` line, addr = sock.recvfrom(1024)`.

sdrasmussen, 2018-12-10:
I am still completely lost...with the above line, where does that tell rtl_433 to send to my MQTT broker?

sdrasmussen, 2018-12-12:
My issue was fixed by the removal of unnecessary flags. The correct format for me ended up being:  

./rtl_433 -F json -f 433926000 | mosquitto_pub -h 192.168.254.201 -p 1883 -u <my-username> -P <my-password> -t weather/rtl_433 -l

Thanks for your help.  Working great!

closed 2018-12-12 by sdrasmussen

zuckschwerdt, 2018-12-13:
You actually don't need the `-f 433926000` also, that's the default ;)

gabrielklein, 2019-02-06:
My version... but I prefer your version :)

{...}/rtl_433 -Fjson | while read line; do mosquitto_pub -h 10.0.0.60 -p 1883 -t 'weather/rtl_433' -m "$line"; done