merbanan/rtl_433 #909: Add option for GPS coordinates in messages

merbanan, 2018-12-11:
There should be an option for static or dynamic GPS coordinates to be added to the messages. If you have a fixed sdr receiver then adding the coordinates statically to a config would be fine. If you are moving reading values from a NMEA compatible GPS over a serial device should be supported. And we need a webservices method for android to be able to update the position on that platform as reading the coordinates from users pace seems to be hard.

zuckschwerdt, 2018-12-11:
I have some early gpsd-based code here https://github.com/zuckschwerdt/rtl_433/tree/feat-gpsd
I just need to verify and clean that. Also the caveat here is that building with gps support needs C11.

merbanan, 2018-12-11:
Code looks good. I do think that we should add the gps parameters statically and copy the values if libgps support is added. That way we can update the values via a webservice as the android location api not only derive the location based on gps only. And static configuration would be possible.

merbanan, 2018-12-11:
It would be nice with a read position from gps command line option and output the text needed for static configuration. That way the setup of a receiver could be very straight forward.

gdt, 2018-12-15:
I second the use of gpsd rather than reinventing it.

A complexity is that it would be nice to get gps timestamps before and after and interpolate, assuming the system clock is synchronized.  This could delay messages up to a second or so, but get better locations.

zuckschwerdt, 2018-12-15:
We have some experimental code for gpsd. I guess the question was more towards a push than a pull data flow for gps. I.e. We could have some gps provider (gpsd, api, webservice) push the info to use, rather than the output pulling it from exactly gpsd. I'll rework the code in that direction.

zuckschwerdt, 2018-12-15:
The idea was not to tie the gps output to one source (gpsd) but leave that open for alternative ways to acquire gps data. (push/pull meaning the internal call chain, not an external interface.)

I don't have gps hardware so I'm speculating, but I guess the libgps/gpsd timestamps related to the instant they were read from the hardware not the time we read and output them?

Any help here to get things right and improve accuracy is welcome!

gdt, 2018-12-15:
It could certainly make sense to allow multiple mechanisms.  But, gpsd aims to interface to all types of positioning hardware and to make the results available in a uniform way, and I expect it to cover 99.9% of actual uses.   Certainly having a procedure inside rtl_433 to accept time, lat, lon, elevation, and then code to get that from gpsd and call it is a perfectly sensible abstraction that leaves the door open to something else (command line fixed position, some other interface).   Also note that elevation is hard, and typically people mean meters above the WGS84 equal-gravity surface (and GPS receivers typically output that).

A GPS receiver produces a solution that contains both time and position, and that timestamp should be carried through.  When it crosses the device to computer boundary, and when it crosses the gpsd to using program boundary will be somewhat later.   My point, which I didn't make clearly, is that if rtl_433 is able to associate timestamps (from the system clock) with samples, and the system clock is correct, then given a position before and after (typically every 1s on the second), it's easy to interpolate the position at the time of sampling.   But taking the currently-stored value is vastly better than what we have now, and doesn't involve delaying output until the next sample arrives.

Maelruan, 2019-07-17:
A NMEA0183 output of wind sensor data would be very useful in marine applications.
See http://www.cruisersforum.com/forums/f134/plugin-dashboard-44087-34.html

zuckschwerdt, 2019-07-24:
@Maelruan do you mean an output format like `-F kv/json/csv`? I think that can be done with a simple converter script (JSON pipe or UDP bridge), see the examples directory. Maybe you can prototype such a script?

mckaymatthew, 2020-01-03:
I submitted a pull request with gpsd support. #1250 
I too would like a more decoupled solution, but gpsd is easy to integrate and provides a solution for many use cases.

PR #1636 (2021-02-01): Add GPSd tags option
Turns out addig GPSd support as simple TCP client is easy ;)

- supersedes  #1635
- supersedes  #1250
- closes #909

Needs a little more work with JSON filtering but is already quite useable.
An option to read other types of tag data from TCP (or UDP, or offer an inbound TCP/UDP tag server) is quick to add, which addresses #909
closed 2021-01-30 by zuckschwerdt