merbanan/rtl_433 #914: Could we merge cfg.verbose and cfg.quiet_mode?

winterrace, 2018-12-14:
Both config members have a partly redundant taste.

Can't we simply delete cfg.quiet mode and handle it like this?:

- rtl_433 --> uses default value 1 for cfg.verbose 
- rtl_433 -q --> sets cfg.verbose to 0
- rtl_433 -D --> uses increased value 2 for cfg.verbose
- rtl_433 -D -D --> uses increased value 3 for cfg.verbose

we had to adjust all uses of cfg.verbose, though
`if(cfg.verbose >1) --> if(cfg.verbose >2)`

zuckschwerdt, 2018-12-14:
It's part of the long term plan. I'll also drop `-q` and `-D` for the more common `-v` (`-vvv` e.g. for debug) then :)

winterrace, 2018-12-14:
OK, fine :) thank you!

zuckschwerdt, 2018-12-14:
Ok, why not do it right away ;) Went in as aa96525.

closed 2018-12-14 by zuckschwerdt