merbanan/rtl_433 #919: LimeSDR mini with SoapySDR, no pulse detected

Xoib, 2018-12-15:
Hello,

I'm trying to do some tests with rtl_433 and the LimeSDR mini but I can't make it work.
Here are the simple cmd line I am using : ```rtl_433 -d "" -G -a -A -f 433871000 -S all```
(targeting a 433.87 Mhz freq)
At the start a pulse is wrongly detected and then nothing.

Any idea what am I missing?

```
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D424BD457D35B'
[DEBUG] Claimed Interface
[DEBUG] Estimated reference clock 40.0017 MHz
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[DEBUG] INT 121, FRAC 0, DIV_LOCH 1, EN_DIV2_DIVPROG 0
[DEBUG] VCO 5000.00 MHz, RefClk 40.00 MHz
[DEBUG] ICT_VCO: 180
[DEBUG] csw=64	cmphl=0
[DEBUG] csw=96	cmphl=0
[DEBUG] csw=112	cmphl=0
[DEBUG] csw=120	cmphl=0
[DEBUG] csw=124	cmphl=0
[DEBUG] csw=126	cmphl=0
[DEBUG] csw=127	cmphl=0
[DEBUG] Failed to lock
[DEBUG] csw=192	cmphl=0
[DEBUG] csw=224	cmphl=2
[DEBUG] csw=240	cmphl=3
[DEBUG] csw=232	cmphl=3
[DEBUG] csw=228	cmphl=2
[DEBUG] csw=230	cmphl=3
[DEBUG] csw=229	cmphl=3
[DEBUG] Failed to lock
[DEBUG] cmphl=2
[DEBUG] VCOL : csw=225 tune ok
[DEBUG] ICT_VCO: 180
[DEBUG] csw=64	cmphl=3
[DEBUG] csw=32	cmphl=3
[DEBUG] csw=16	cmphl=3
[DEBUG] csw=8	cmphl=3
[DEBUG] csw=4	cmphl=3
[DEBUG] csw=2	cmphl=3
[DEBUG] csw=1	cmphl=3
[DEBUG] Failed to lock
[DEBUG] csw=192	cmphl=3
[DEBUG] csw=160	cmphl=3
[DEBUG] csw=144	cmphl=3
[DEBUG] csw=136	cmphl=3
[DEBUG] csw=132	cmphl=3
[DEBUG] csw=130	cmphl=3
[DEBUG] csw=129	cmphl=3
[DEBUG] Failed to lock
[DEBUG] cmphl=3
[DEBUG] TuneVCO(SXT) - failed to lock (cmphl!=2)
[DEBUG] VCOM : csw=0 tune fail
[DEBUG] ICT_VCO: 180
[DEBUG] TuneVCO(SXT) - VCO too high
[DEBUG] VCOH : csw=0 tune fail
[DEBUG] Selected: VCOL
[DEBUG] INT 116, FRAC 0, DIV_LOCH 1, EN_DIV2_DIVPROG 0
[DEBUG] VCO 4800.00 MHz, RefClk 40.00 MHz
[DEBUG] ICT_VCO: 180
[DEBUG] csw=64	cmphl=0
[DEBUG] csw=96	cmphl=0
[DEBUG] csw=112	cmphl=0
[DEBUG] csw=120	cmphl=0
[DEBUG] csw=124	cmphl=0
[DEBUG] csw=126	cmphl=0
[DEBUG] csw=127	cmphl=0
[DEBUG] Failed to lock
[DEBUG] csw=192	cmphl=0
[DEBUG] csw=224	cmphl=3
[DEBUG] csw=208	cmphl=3
[DEBUG] csw=200	cmphl=2
[DEBUG] csw=204	cmphl=3
[DEBUG] csw=202	cmphl=3
[DEBUG] csw=201	cmphl=2
[DEBUG] CSW: lowest=197, highest=201, selected=199
[DEBUG] cmphl=2
[DEBUG] VCOL : csw=199 tune ok
[DEBUG] ICT_VCO: 180
[DEBUG] TuneVCO(SXR) - VCO too high
[DEBUG] VCOM : csw=0 tune fail
[DEBUG] ICT_VCO: 180
[DEBUG] TuneVCO(SXR) - VCO too high
[DEBUG] VCOH : csw=0 tune fail
[DEBUG] Selected: VCOL
[DEBUG] INT 57, FRAC 385875, DIV_OUTCH_CGEN 18
[DEBUG] VCO 2334.72 MHz, RefClk 40.00 MHz
[DEBUG] csw 155; interval [152, 159]
[DEBUG] INT 57, FRAC 385875, DIV_OUTCH_CGEN 18
[DEBUG] VCO 2334.72 MHz, RefClk 40.00 MHz
[DEBUG] csw 155; interval [152, 159]
[DEBUG] M=160, N=4, Fvco=614.400 MHz
[DEBUG] M=160, N=4, Fvco=614.400 MHz
[INFO] LMS7002M calibration values caching Disable
[DEBUG] INT 59, FRAC 0, DIV_OUTCH_CGEN 14
[DEBUG] VCO 2400.00 MHz, RefClk 40.00 MHz
[DEBUG] csw 170; interval [167, 173]
[DEBUG] M=195, N=3, Fvco=1300.000 MHz
[DEBUG] M=195, N=3, Fvco=1300.000 MHz
[DEBUG] SoapyLMS7::setGain(Rx, 0, LNA, 0 dB)
[DEBUG] Actual RxLNA[0] gain 0 dB
[DEBUG] SoapyLMS7::setGain(Tx, 0, PAD, 0 dB)
[DEBUG] Actual TxPAD[0] gain 0 dB
Using device LimeSDR-Mini: boardSerialNumber=0x1d424bd457d35a firmwareVersion=6 gatewareVersion=1.28 hardwareVersion=2 protocolVersion=1 
Found 4 antenna(s): NONE LNAH LNAL_NC LNAW 
Found 3 gain(s): TIA -12 - 61 (step 0) LNA -12 - 61 (step 0) PGA -12 - 61 (step 0) 
Found 2 frequencies: RF BB 
Found 1 frequency range(s): 0 - 3800000000 (step 0) 
Found 1 sample rate range(s): 100000 - 65000000 (step 0) 
Found 1 bandwidth range(s): 1400000 - 130000000 (step 0) 
Found current bandwidth 30000000
Found 3 stream format(s): CF32 CS12 CS16 
Found native stream format: CS16 (full scale: 2048.0)
[DEBUG] INT 59, FRAC 838860, DIV_OUTCH_CGEN 37
[DEBUG] VCO 2432.00 MHz, RefClk 40.00 MHz
[DEBUG] csw 175; interval [173, 178]
[DEBUG] INT 59, FRAC 838859, DIV_OUTCH_CGEN 37
[DEBUG] VCO 2432.00 MHz, RefClk 40.00 MHz
[DEBUG] csw 175; interval [173, 178]
Sample rate set to 249999 S/s.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
[DEBUG] MCU algorithm time: 11 ms
[DEBUG] MCU algorithm time: 0 ms
[DEBUG] MCU algorithm time: 0 ms
[DEBUG] MCU algorithm time: 136 ms
[INFO] Rx calibration finished
Reading samples in async mode...
[DEBUG] INT 82, FRAC 811807, DIV_LOCH 3, EN_DIV2_DIVPROG 1
[DEBUG] VCO 6941.94 MHz, RefClk 40.00 MHz
[DEBUG] ICT_VCO: 180
[DEBUG] TuneVCO(SXR) - VCO too low
[DEBUG] VCOL : csw=0 tune fail
[DEBUG] ICT_VCO: 180
[DEBUG] TuneVCO(SXR) - VCO too low
[DEBUG] VCOM : csw=0 tune fail
[DEBUG] ICT_VCO: 180
[DEBUG] csw=64	cmphl=0
[DEBUG] csw=96	cmphl=2
[DEBUG] csw=112	cmphl=2
[DEBUG] csw=120	cmphl=3
[DEBUG] csw=116	cmphl=3
[DEBUG] csw=114	cmphl=2
[DEBUG] csw=115	cmphl=2
[DEBUG] CSW: lowest=96, highest=115, selected=105
[DEBUG] csw=192	cmphl=3
[DEBUG] csw=160	cmphl=3
[DEBUG] csw=144	cmphl=3
[DEBUG] csw=136	cmphl=3
[DEBUG] csw=132	cmphl=3
[DEBUG] csw=130	cmphl=3
[DEBUG] csw=129	cmphl=3
[DEBUG] Failed to lock
[DEBUG] cmphl=2
[DEBUG] VCOH : csw=105 tune ok
[DEBUG] Selected: VCOH
Tuned to 433.871MHz.
Detected OOK package	2018-12-15 22:43:49
Analyzing pulses...
Total count:    1,  width: 112.09 ms		(28023 S)
Pulse width distribution:
 [ 0] count:    1,  width: 112092 us [112092;112092]	(28023 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:  10916,   1734
RSSI: -3.5 dB SNR: 16.0 dB Noise: -19.5 dB
Frequency offsets [F1, F2]:      13,      0	(+0.0 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

signal_bsize = 393216  -      sg_index = 524288
start_pos    = 3004692  -   buffer_size = 3145728
*** Saving signal to file g005_433.871M_250k.cs16
*** Writing data from 3004692, len 141036
*** Writing data from 0, len 252180
*** signal_start = -8172, signal_end = 39851, signal_len = 48023, pulses_found = 1
Distance coding: Pulse length 28022

Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 28022
bitbuffer:: Number of rows: 0 

^CSignal caught, exiting!
```

zuckschwerdt, 2018-12-15:
I guess you need to set the gains.
Also don't use `-G` (you won't get more data, just more false positives.) or `-a` (old and not very helpful). 
Try `rtl_433 -d "" -g "LNA=20,TIA=8,PGA=2" -f 433.87M -A -S all`

zuckschwerdt, 2018-12-15:
Finding the gain really is the weak point here. Isn't there a way to get automatic gain with newer firmware/drivers? My LimeSDR is pre-production and doesn't support the current firmware with auto gain in FPGA. Also you can run the LimeSDR at much higher sample rates than the puny rtl-sdrs :) `-s 2M` should be no problem, I think `8M` worked well for me.

zuckschwerdt, 2018-12-15:
The SNR looks ok, the Noise looks really well, the RSSI could be higher. Keep a look out for that when you try other gains.

merbanan, 2018-12-15:
You could try with -l [200-3000] also but the signal is not filling the dynamic range properly. The pulse length distribution is very wide. Try with gqrx until you can see the signal in the spectrum.

zuckschwerdt, 2018-12-15:
@merbanan you need to take into account that the 12 bit from the Lime will be processed to a true magnitude and not just amplitude like for 8 bit. I.e. quiet parts are "amplified" and loud parts attenuated. The pulse detect will need a level more in the "center" than the low end.

Xoib, 2018-12-16:
Thanks for your help!
With 2M or more, like 8M samples, the results aren't much better in term of signal recognition.
Also the detection level (-l) make it's harder to capture everything so I don't use it.
But with less gain I get an higher RSI thus that's good.

Regarding gqrx, I can't for now, as on this laptop I'm running Fedora fc29 and it suffer from a libuhd [bug](https://bugzilla.redhat.com/show_bug.cgi?id=1645684); Compiling manually from source also.

Thus I did a capture through urh: [complex file](https://send.tresorit.com/a#WMw7eRxAgEBPu9hwQ1MzhQ)
Raw from rtl_433: [cs16 files](https://send.tresorit.com/a#iY0826bZafU0zckcZZzEwg)
_Password: limesdr_

Any suggestions on how can I narrow this protocol? (btw it's a simple garage door keyfob)

```
Detected OOK package	2018-12-16 09:44:02
Analyzing pulses...
Total count:  102,  width: 188.52 ms		(377036 S)
Pulse width distribution:
 [ 0] count:  102,  width:  497 us [496;500]	( 994 S)
Gap width distribution:
 [ 0] count:   43,  width:  504 us [502;506]	(1008 S)
 [ 1] count:    3,  width: 5219 us [4500;5581]	(10438 S)
 [ 2] count:   39,  width: 1515 us [1500;1582]	(3030 S)
 [ 3] count:   15,  width: 2524 us [2501;2581]	(5047 S)
 [ 4] count:    1,  width: 3501 us [3501;3501]	(7002 S)
Pulse period distribution:
 [ 0] count:   43,  width: 1002 us [998;1004]	(2003 S)
 [ 1] count:    3,  width: 5716 us [4997;6078]	(11432 S)
 [ 2] count:   39,  width: 2012 us [1997;2079]	(4025 S)
 [ 3] count:   15,  width: 3020 us [2998;3079]	(6041 S)
 [ 4] count:    1,  width: 3999 us [3999;3999]	(7998 S)
Level estimates [high, low]:   3719,    130
RSSI: -12.9 dB SNR: 29.1 dB Noise: -42.0 dB
Frequency offsets [F1, F2]:     538,      0	(+16.4 kHz, +0.0 kHz)
Guessing modulation: Pulse Position Modulation with fixed pulse width
Attempting demodulation... short_width: 504, long_width: 1515, reset_limit: 5582, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PPM,s=504,l=1515,g=1582,r=5582'
pulse_demod_ppm(): Analyzer Device 
bitbuffer:: Number of rows: 20 
[00] { 1} 00        : 0
[01] {16} d0 74     : 11010000 01110100 
[02] { 4} c0        : 1100
[03] { 3} 40        : 010
[04] { 4} d0        : 1101
[05] { 3} 20        : 001
[06] { 0}           : 
[07] { 1} 00        : 0
[08] { 3} 40        : 010
[09] { 3} 40        : 010
[10] { 3} 20        : 001
[11] { 3} 20        : 001
[12] {15} 7d d2     : 01111101 1101001
[13] { 1} 00        : 0
[14] { 9} 6d 00     : 01101101 0
[15] { 2} 00        : 00
[16] { 1} 80        : 1
[17] { 6} e8        : 111010
[18] { 4} 40        : 0100
[19] { 0}           : 

signal_bsize = 1703936  -      sg_index = 2097152
start_pos    = 13464  -   buffer_size = 3145728
*** Saving signal to file g051_433.87M_2000k.cs16
*** Writing data from 13464, len 1703936
Detected OOK package	2018-12-16 09:44:02
Analyzing pulses...
Total count:   82,  width: 187.44 ms		(374878 S)
Pulse width distribution:
 [ 0] count:   82,  width:  497 us [496;499]	( 994 S)
Gap width distribution:
[... 733 lines cut ...]
[04] { 2} 40          : 01
[05] { 1} 80          : 1
[06] {14} 37 a4       : 00110111 101001
[07] { 0}             : 
[08] { 3} 20          : 001
[09] { 4} 30          : 0011
[10] { 1} 80          : 1
[11] { 0}             : 
[12] { 0}             : 
[13] { 0}             : 
[14] { 4} 20          : 0010
[15] { 3} 40          : 010
[16] { 2} 40          : 01
[17] { 0}             : 
[18] { 2} 00          : 00
[19] { 1} 80          : 1
[20] { 0}             : 

signal_bsize = 1703936  -      sg_index = 1048576
start_pos    = 2097004  -   buffer_size = 3145728
*** Saving signal to file g065_433.87M_2000k.cs16
*** Writing data from 2097004, len 1048724
*** Writing data from 0, len 655212
^CSignal caught, exiting!
```

merbanan, 2018-12-16:
I think the decoder we have is for 2-DPPM which is a special case of general PPM.

merbanan, 2018-12-16:
Or PDM (Pulse distance coding)

https://www.phidgets.com/docs/IR_Remote_Control_Primer

zuckschwerdt, 2018-12-16:
Sounds like RZ PCM. `-X n=foo,m=OOK_PCM,s=500,l=1000` should do

merbanan, 2018-12-16:
No go with the samples I tried.

zuckschwerdt, 2018-12-18:
I did try with a random sample and it worked.
```
rtl_433 -M level -s 2M -R 0 -X "n=foo,m=OOK_PCM,s=500,l=1000,r=9000" g060_433.87M_2000k.cs16
```
gives
```
{195}c142778f142272859bc5cb513560cd43648cd12778cd23c00
```
(strangely enough running a batch of files results in shorter codes like `{108}a89ab066a1b2466893bc6691e0`, perhaps some buffer is broken. I'll have a look).

zuckschwerdt, 2018-12-18:
Ok, fixed. CS16 buffers need 0 not 128 to clear out ;)
The read codes look about right compared by hand and are:
```
g051_433.87M_2000k.cs16 {197}c15bf56e5736569d1306cd9d3a6aad5ba66b5b3942ad9b9000
g052_433.87M_2000k.cs16 {196}c142478918285c85e3c4eb583060e943072a1cae78fd43600
g053_433.87M_2000k.cs16 {197}c16b95cfd43f44cdaba42856b55cfd6a250d91336ad5139000
g054_433.87M_2000k.cs16 {197}c16ad74c1e3f48cdd3a66916a45cf16b36ad5d336ab5b3f000
g055_433.87M_2000k.cs16 {197}c172766adfb950d99a274a17285ad573644ad9b656cdaad000
g056_433.87M_2000k.cs16 {197}c13ba7e9122e4ec17334e9d43c6cd93bf50d5aa866b9733000
g057_433.87M_2000k.cs16 {196}c1e2d5ac12334ee9d2866b99bc72d9e337291abf74b5e3a00
g058_433.87M_2000k.cs16 {193}c1ca86ca1b217eadfb672e922142a1cab56fd7245a8523000
g059_433.87M_2000k.cs16 {193}c16ba7491e2e48d573a4e9142478f16bf6ad9d286ab93b000
g060_433.87M_2000k.cs16 {195}c142778f142272859bc5cb513560cd43648cd12778cd23c00
g061_433.87M_2000k.cs16 {195}c1d214ed512d5cf1b275aad3bf7ee9d355ce543a50f1e3c00
g062_433.87M_2000k.cs16 {196}c123f7ee52366a9d1356c9dd3c6ae523a50f1ab944ad13600
g063_433.87M_2000k.cs16 {196}c1fba7485e3c48897246a91b244ef1fbf6ab5d2b48b96ae00
g064_433.87M_2000k.cs16 {194}c173966cdfad60f1aa27ea13a87ec172244919ba56d542800
g065_433.87M_2000k.cs16 {196}c13ab7efd5bf66fd5b3429d6ab749d3a850913b36691ca200

```

Xoib, 2018-12-18:
Thank you @merbanan and @zuckschwerdt !
I know get consistent data.
Now that I can assume the right settings, when I try those (width of pulse and bit period plus the max gap size before end of message) in a graphical tool, I get totally different result.
[image]
Any idea?

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2018-12-19 00:04:54
model     : foo          count     : 1             num_rows  : 1             rows      : 
len       : 196          data      : c13bc7ee95ae66c1233489d42b6c9d3ae5ae56a1669d3a200
codes     : {196}c13bc7ee95ae66c1233489d42b6c9d3ae5ae56a1669d3a200
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -22.0 dB     SNR       : 23.8 dB       Noise     : -45.9 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2018-12-19 00:04:54
model     : foo          count     : 1             num_rows  : 1             rows      : 
len       : 197          data      : c16bc74e9b2e42d523a489142178b96ae6cd9ca16a9de2d000
codes     : {197}c16bc74e9b2e42d523a489142178b96ae6cd9ca16a9de2d000
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -21.7 dB     SNR       : 20.7 dB       Noise     : -42.5 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2018-12-19 00:04:55
model     : foo          count     : 1             num_rows  : 1             rows      : 
len       : 196          data      : c1e2b5afd5b366e95a842b99ab729de287ee56b674919ae00
codes     : {196}c1e2b5afd5b366e95a842b99ab729de287ee56b674919ae00
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -21.5 dB     SNR       : 22.1 dB       Noise     : -43.7 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2018-12-19 00:04:55
model     : foo          count     : 1             num_rows  : 1             rows      : 
len       : 193          data      : c17285aa15a15af1fa272b922b7e8972b7e856ae5685e3000
codes     : {193}c17285aa15a15af1fa272b922b7e8972b7e856ae5685e3000
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -21.7 dB     SNR       : 20.6 dB       Noise     : -42.4 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2018-12-19 00:04:55
model     : foo          count     : 1             num_rows  : 1             rows      : 
len       : 197          data      : c1d2d6ad572d48f1d275af13ba7ef1d3348c182750b5e33000
codes     : {197}c1d2d6ad572d48f1d275af13ba7ef1d3348c182750b5e33000
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -21.6 dB     SNR       : 23.0 dB       Noise     : -44.7 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2018-12-19 00:04:56
model     : foo          count     : 1             num_rows  : 1             rows      : 
len       : 197          data      : c1d286aa1d217ef1fa772f122d7ea1d2b48a142e50856a1000
codes     : {197}c1d286aa1d217ef1fa772f122d7ea1d2b48a142e50856a1000
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -21.5 dB     SNR       : 22.3 dB       Noise     : -43.9 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2018-12-19 00:04:56
model     : foo          count     : 1             num_rows  : 1             rows      : 
len       : 196          data      : c1ca4788582b5ce5e366ab5e3066e9cb074c17305afdd2e00
codes     : {196}c1ca4788582b5ce5e366ab5e3066e9cb074c17305afdd2e00
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -21.8 dB     SNR       : 21.4 dB       Noise     : -43.3 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2018-12-19 00:04:56
model     : foo          count     : 1             num_rows  : 1             rows      : 
len       : 195          data      : c143c6ce9b28428523c48e982160b942e5cfdca4789d82400
codes     : {195}c143c6ce9b28428523c48e982160b942e5cfdca4789d82400
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -21.9 dB     SNR       : 24.5 dB       Noise     : -46.6 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2018-12-19 00:04:56
model     : foo          count     : 1             num_rows  : 1             rows      : 
len       : 197          data      : c1e2e5af15a25a910a85cb912b5089e3c42b56ba74896b9000
codes     : {197}c1e2e5af15a25a910a85cb912b5089e3c42b56ba74896b9000
[... cut ...]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2018-12-19 00:04:58
model     : foo          count     : 1             num_rows  : 1             rows      : 
len       : 196          data      : c13bf7ed92276ac1133789dcbc6ce53ba66a19ad66ad6ba00
codes     : {196}c13bf7ed92276ac1133789dcbc6ce53ba66a19ad66ad6ba00
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -22.1 dB     SNR       : 18.3 dB       Noise     : -40.4 dB
``

zuckschwerdt, 2018-12-19:
The encoding here is [Return-to-Zero (RZ)](https://en.wikipedia.org/wiki/Return-to-zero), your other tool assumes NRZ (thus inserts an additional zero everywhere).
E.g. `a002`, and `c1` are the same prefix, just stuffed with `0`s. See the first and last output line in [this BitBench](http://triq.net/bitbench?c=c1&c=a002&f=1bO%20%0A2b%20).

zuckschwerdt, 2018-12-19:
Can you share Manufacturer and Model of the system? It's a recommendable secure system for a garage door opener. It employs a rolling code, I guess no further decryption is possible.
If you just want to spot the signal use that decoder line and look for at least 180 bits starting with "c1".

Xoib, 2018-12-19:
Thanks @zuckschwerdt ! It's a [MEMOBIP](https://www.yesss-fr.com/produits/43082/photos/43082-MEMOBIP_.jpg) (433Mhz first generation) from [URMET](http://www.urmet.fr/catalogue-virtuel/emailing/2015/vithf/doc_vithf.pdf). I didn't find the link for 433Mhz system (they don't produce it nor maintain it any more) but it's the same kind of architecture.

There is a [PIC12F629](https://www.microchip.com/wwwproducts/en/PIC12F629) as programming chip, and for RF there is a T574 3WU (Unfortunately I didn't find any documentation regarding that chip).
Also it contain a 13.56Mhz chip for opening classic door VIGIK system, not related to what we are interested here.

I guess I understand how to narrow the short and long parameter, but the reset parameter, how did you narrow it?

merbanan, 2018-12-19:
Reset should be longer then the longest message bit interval. @zuckschwerdt can you create a flex decoder specification based on the c1 and > 180 bits info?

zuckschwerdt, 2018-12-19:
The longet run of 0 I've seen is 5, thus reset needs to be bigger than 5500 µs (accounts for the preceeding bit space). Something between 6000 and 9000 seems reasonable.

Add `,bits>=180` to require a minimum bit length. Add `,match=c1` to match the byte. It's currently no eas to match a specific positions.

closed 2018-12-31 by zuckschwerdt