merbanan/rtl_433 #931: WH65B incorrectly identified as WH24

cfarquhar, 2018-12-20:
#844 recently added support for the WH65B sensor array.  Unfortunately, my array is being incorrectly identified as WH24.  I'm opening this issue and a [PR](https://github.com/merbanan/rtl_433_tests/pull/250) with some collected samples with the hope that we can improve differentiation between the two models.

SDR is very new to me, so if there's any possibility that having something misconfigured could contribute to the incorrect model identification please let me know.

rudders, 2019-03-30:
I have this issue I believe - can I fix it myself somehow?

merbanan, 2020-01-06:
Should be fixed. But if more heuristics are needed then I think the model number should just be changed to wh24/wh65b.

closed 2020-01-06 by merbanan

rudders, 2020-01-06:
I'll give it a try and report back.