merbanan/rtl_433 #934: Unable to specify gain with RTL-SDR

Mihara, 2018-12-22:
So I have a Gentoo Linux system, on which I'm mucking around with home automation, for which I compile rtl_433 from source... I've had some issues with devices which appear to be very distant interfering with ones I have right next to me, and tried to work around them by setting gain to a fixed value, which improved results a lot when I was using this dongle with gqrx. This appears to lead to weird results:

```
$ rtl_433 -g 40                                                                                                                                         
rtl_433 version 18.12-32-gd233adf branch  at 201812191913                                                                                                                    
...
Reading conf from "/etc/rtl_433/rtl_433.conf".
Registered 61 out of 120 device decoding protocols [ 2-3 8 11-12 16 19 21 23 25-26 29 31-38 40 42-45 47 49-52 56-58 62-63 66-70 73-81 83-85 87 92 94 96-99 102 108 ]        
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to 40.000000 dB.
Frequency correction set to 94 ppm.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
```

The documentation specifies that for an RTL-SDR device, the gain is to be given in tenths of a decibel, but the one I see reported is as if it is interpreted as whole decibels. In fact, seeing how the RSSI output of frequently transmitting nearby devices (`-M level`) does not appear to change as I alter the number, it appears the gain is not getting set at all. Rebuilding with librtlsdr from git doesn't change anything. (The difference between their current git master and 0.5.4 that I've had installed before seems to be negligible anyway.)

rtl_sdr itself reports setting the gain to fixed values from some list, but at least those seem to make more sense:
```
$ rtl_sdr -g 400 -f 433920000 -p 94 /tmp/d
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2048000 S/s.
Tuned to 433920000 Hz.
Tuner gain set to 49.60 dB.
Tuner error set to 94 ppm.
Reading samples in async mode...
Allocating 15 zero-copy buffers
```

(And I get 8.7dB for `-g 10`, 19.7dB for `-g 20` etc...)

What could be wrong here?

merbanan, 2018-12-22:
Tuner gain is not part of the RSSI calculation. @zuckschwerdt maybe we should relay the tuner gain value in the message? Or use it in the RSSI calculation.

Mihara, 2018-12-22:
I only mentioned the RSSI values because I have seen a comment by @zuckschwerdt which [seems to say they are relative to the gain value,](https://github.com/merbanan/rtl_433/issues/745#issuecomment-444385279) -- unless I misunderstood it horribly...

In any case, the docs say "gain is in units equal to 0.1" and what I see printed for `-g 40` is "40dB" instead of "4dB"...

merbanan, 2018-12-22:
Currently as I understood it RSSI and tuner gain are not related. Hope that answers your question.

Mihara, 2018-12-22:
Shouldn't it also `rtlsdr_get_tuner_gains` somewhere, though? Because the values rtl_sdr lists as valid for R820T are not exactly linear:
```
        const int r82xx_gains[] = { 0, 9, 14, 27, 37, 77, 87, 125, 144, 157,
                                     166, 197, 207, 229, 254, 280, 297, 328,
                                     338, 364, 372, 386, 402, 421, 434, 439,
                                     445, 480, 496 };
```
(https://git.osmocom.org/rtl-sdr/tree/src/librtlsdr.c#n970)

zuckschwerdt, 2018-12-22:
Yes, we probably shouldn't set random gains but round to the nearest value from `rtlsdr_get_tuner_gains()`. I hope to add this soon. Thanks!

merbanan, 2020-01-06:
RSSI and friends shouldn't be reported when using auto mode. The values will mostly be self relevant but during transmission collision bogus data will be reported. One most use fixed gain to get proper values.

merbanan, 2020-01-06:
@zuckschwerdt any progress on this?

zuckschwerdt, 2020-01-06:
Still needs doing. I need to add two calls to `int rtlsdr_get_tuner_gains(rtlsdr_dev_t *dev, int *gains)` to clamp the user input to a supported value. Maybe I'll get around to to it in the next few days.

The reported dB values are a more practical representation of hi/low level and ratio in the frame. They are a quality measure in the digital backend only. It would be rather complicated to factor the frontend in. RSSI should perhaps better be named just "signal". RSSI could then only be shown if the current gain is known (as gain+signal).

zuckschwerdt, 2020-01-07:
Exact gain for RTL-SDR went in with 3c9cab8.

closed 2020-01-08 by zuckschwerdt