merbanan/rtl_433 #938: How to display SNR on Output of RTL_433

MaxFun, 2018-12-28:
Hi,

I just download, built and ran rtl_433.
The output now has color, compared to the older build, but it does not display SNR, Noise, ... as shown in the image at https://github.com/merbanan/rtl_433.
What are the options do I need to add to rtl_433 to get a similar displayed output

Thanks

zuckschwerdt, 2018-12-28:
`rtl_433 -h` gives a terse overview of options and `rtl_433 -M` more details (you want `-M level`).
We'll add color to the help output soon to make it easier to follow.

closed 2018-12-28 by zuckschwerdt

MaxFun, 2018-12-28:
That works, Thanks

Another question, when I start rtl_433, I get (see below) that rtl_433 has unknown version.
Is this a problem?

C:\1Mike\Projects\SDR\Rtl_433\Install RTL_433 for Windows\librtlsdr-master\build\src>rtl_433 -M level
rtl_433 version unknown
Trying conf file at "C:\1Mike\Projects\SDR\Rtl_433\Install RTL_433 for Windows\librtlsdr-master\build\src\rtl_433.conf"...
Trying conf file at "C:\Users\BigDell\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered zu out of 96 device decoding protocols [ 1-4 8 11-12 15-21 23 25-26 29-36 38-60 62-64 67-71 73-100 102-103 108-116 119 ]
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2018-12-28 11:08:24
model     : Acurite tower sensor                   id        : 13482
sensor_id : 0x34aa       channel   : A             Temperature: 21.8 C       Humidity  : 16            battery low: 0
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 19.3 dB       Noise     : -19.4 dB

zuckschwerdt, 2018-12-28:
It's just a diagnostic ouput and "unknown" indicates you compiled without git around or with Autotools instead of CMake.