merbanan/rtl_433 #940: Appetite to support Raspberry Pi with simple recivers?

untergasser, 2018-12-31:
rtl_433 is a great project. It also compiles and works well on a Raspberry Pi. But for monitoring just one sensor, it is quite some overhead. As the Raspberry Pi has GPIO pins and simple recivers are cheap, consume low energy and can be read without creating a heavy processor load, I created a simple (141 lines of code) proof of principle working with the Alecto WS-1200:

https://github.com/untergasser/tempRainPi

My implementation suggestion would (maybe a second binary) requireing only the WiringPi. The reciver will only have zero or maximal sinal (as there is binary input) and needs command line flags to tell him about the used pin, the frequency and the modulation of the module (as this is done by the module and unknown to the Raspberry Pi). This information is used to select the sensors able to decode. In short, in this binary you would cut of the rtl stuff and connect it to your sensor database and signal processing.

Of course, only few sensors will be supportet (depending on the recivers OOK modulation here), but I assume it could be simple to implement and maintain, as there is only the Raspberry Pi as hardware dependency and WiringPi as a software dependency. It might be quite helpfull to people interested to build weather stations with the Raspberry Pi (were an SDR might be too much) and they could buy suitable sensors. Furthermore, it would open this project for a big group of new users.

Last, although I still have to work it out, signals could be played back to the displays using a simple sender, allowing to place the Raspberry Pi in the middle.

What do you think?

Best wishes and a happy new year,

Andreas Untergasser

zuckschwerdt, 2018-12-31:
This has been requested a few times and was not easily possible. Our current work towards an API with a pipeline approach (or stackable stages) will make this available.

The idea is that an application can define which transformation steps should be applied to an input.
I'll add an example shortly which takes pulse data (something like the already supported VCD output) and runs that through the decoders.

The refactoring is underway and I estimate this will be done in a few months. Input options and dependencies will be a compile-time choice and thus creating this as a small application with no additional requirements is planned.

untergasser, 2019-01-25:
Great plan, this would be optimal. I am playing around with raspberry pi and maybe I could help with placing it as a input source. Would be great and a low cost device for many applications.

zuckschwerdt, 2019-01-27:
There is some early code in #967. You can perhaps use a script that reads WiringPi and pipe through rtl_433. I'll see if I can test that and then add proper recevier support.

untergasser, 2019-02-03:
Thank you for adding the new code, I will give it a try. I am in a project which has to be ready end of March, so it may take some time till I can provide some results. Is it possible to add Alecto WS-1200 support for rtl_433 in the meantime? I opened an issue with my suggestions.
https://github.com/merbanan/rtl_433/issues/973
I would have loved to send a pull request instead, but I failed to understand the source structure at first sight.
Best, Andreas

zuckschwerdt, 2019-02-03:
If you hold out a few days, then some of the pulse data code should be merged already.

doughecka, 2019-03-24:
This would also be useful to enable the monitoring of multiple frequencies at the same time in real-time without dedicating a full SDR for each band. For example, use the OOK receivers for picking up security sensors (on 443mhz and 345mhz), and let a SDR shift between all the frequencies to gather up the other sensors that don't need to be real-time (weather, temp, etc).

zuckschwerdt, 2019-03-24:
This basically already works. But don't get your hopes up, the results depend very much on cleaning up the OOK data. The cheap OOK receivers only output 1-bit data (on/off). There is no distinction between a signal and random noise.
Use e.g. a python script to read data with WiringPi or something similar. Then filter and frame the data. An idea I tested would be to look for longer on and off sequences with a known "rhythm". I.e. discard very fast changes (noise). Then feed to `rtl_443 -r OOK:-`. If someone really wants to hack on that filtering algorithm then I might post an example script.
Look at the output of `rtl_433 -w OOK:-` to get a feel for the data needed. (pulse-gap lines with `;ook` and `;end` framing).
A much better idea would be e.g. CC1101 ("rf1100se") module or alike, same price but a proper receiver.

merbanan, 2020-01-06:
I think the proper way to implement 1-bit receivers is to create an abstraction layer lib. Then gpio input, serial input hardware and tranceivers can use the same api. Then hook it up to the OOK interface.

Anyway I think this can be closed? @zuckschwerdt

zuckschwerdt, 2020-01-07:
I still need to add the example code. But implementation need to specific to the protocol, so nothing we can deliver out-of-the-box.

closed 2020-01-07 by zuckschwerdt

kevinmehall, 2026-02-09:
I got this working with a CC1101 and wrote https://codeberg.org/kevinmehall/gpio_433 which monitors a Linux gpiodev GPIO pin and streams pulse timings in a format that can be piped into `rtl_433 -r OOK:-` for decoding.

You'll also want https://github.com/merbanan/rtl_433/pull/3464 to pass through timestamps correctly from OOK input.