merbanan/rtl_433 #943: Support for (generic) FS20 sensors

MarcusWernicke, 2019-01-05:
I would like to request for FS20 sensors (motion sensors, door contacts, smoke detectors).

(The ELV device supported (for weather stations) references fs20.c, so it's probably based on a generic decoder. )

Some samples recorded via -s unknown from a smoke detector and a motion detector:
https://drive.google.com/file/d/16xgnUE9xuzWl8uKP5LWzCYcRsYqeR-Ty/view?usp=sharing

Project implementing an FS20 decoder in Python:
https://github.com/eT0M/rtl_sdr_FS20_decoder

zuckschwerdt, 2019-01-07:
Do you know how the ELV EM-1000 relates to the FS20 family? Is there anyone around who has the ELV EM-1000?

MarcusWernicke, 2019-01-10:
> Do you know how the ELV EM-1000 relates to the FS20 family?

I don't think it uses the FS20 protocol (or at least its documentation doesn't mention that). It just seems to be a product made by the same German manufacturer.

closed 2019-04-06 by zuckschwerdt