merbanan/rtl_433 #947: RFC Holman iWeather WH5029

fnarfbargle, 2019-01-07:
G'day all,

I've been working on support for an Australian Holman iWeather device. I have it working, although I haven't been able to sus out the checksum yet. The low nybble is a straight xor but I haven't figured out the high yet. As a result, this file contains extra debugging output and isn't suitable for inclusion yet, but I wanted to get it looked at to see if I'm on the right track for feature and format when I get it finished.

[holman.c.txt](https://github.com/merbanan/rtl_433/files/2731245/holman.c.txt)

The rain units are still a bit handwavy, although it looks pretty close to each count being 1/32 of an inch. I need to push a heap more water through the unit to refine the multiplier, but it's currently good over 50mm.

I have sample recordings and all the other requirements, but I thought I'd get an early version of the code looked at for comment.

merbanan, 2019-01-07:
Look at the use of thebitbuffer_search()  and bitbuffer_find_repeated_row() function. Other then that the code looks fairly good. Regarding the checksum a check for the 3 byte preamble and a 4bit xor is enough to reliably verify the message content. Add a "mic" to the output.

fnarfbargle, 2019-01-07:
Much obliged.

I've used find_repeated_row, but I couldn't see how bitbuffer_search would simplify the code as I haven't yet seen an improperly aligned bitbuffer.

I did get a couple of corrupted messages through the previous version where the corruption was in the id byte and the xor passed, so looking for at least 3 identical rows should nail that down.

```C
    // Test for 3 identical rows (we get 6)
    int r = bitbuffer_find_repeated_row(bitbuffer, 3, 96);
    if (r < 0 || bitbuffer->bits_per_row[r] != 96)
        return 0;

    // Test for preamble
    if (memcmp(bb[r],header,3))
        return 0;

    // Test Checksum.
    if (xor_bytes(bb[r],11) & 0xF)
        return 0;
```
I really would like to nail the whole checksum, if only because it's annoying me, so I'll keep working on it for a while before putting it in git and making a pull request.

merbanan, 2019-01-07:
Then you need to post a few (50?) bit bufffers so we might be able to help out. If the bitbuffer is aligned then there is no need to align it. But only bitbuffer_invert(bitbuffer); after checking the bits per row and that the preamble + header matches.

zuckschwerdt, 2019-01-07:
The wind direction should preferably be in degrees.
Minor stuff: don't use stdint types if you don't need them (uint8_t in for-loop should be int). Don't use memcmp (memcpy, memset, alloc, …).

fnarfbargle, 2019-01-07:
Already there. I refactored this afternoon to move the invert until after all the tests (including the checksum), made the direction in degrees (n * 22.5) and did a proper units test on the rain gauge to get the correct values and make sure I've got samples for every nybble change in the message (except the low bat). That had to wait until I had a chance to get on the roof and put enough water through it to hit the high nybble of the rain counter.

I'll try something else for the preamble to avoid the memcmp. With using bitbuffer_find_repeated_row I've taken the loop out completely.

For the checksum I've gathered a larger set (~3000 samples) of discrete and distinct samples. I've knocked up a (very dirty) python parser to process it to run iterative tests, and they are attached. 

The python code loads the file line by line and builds it into an array of values. I have tested every permuation of poly/init and gen/key on crc8 and lfsr8, various xor combinations, simple addition, per-nybble xor and anything else I could think of. I tried the whole message, sans preamble and sans device type. I've been beating on it for a couple of days, so I'm pretty much out of ideas (and time at the moment). Feel free to embarrass me by taking a look and going "Oh yeah, it's one of *those*". Iv'e embarrassed myself enough just posting the python spaghetti mess.

The python code as configured just loads the gunzipped iWeather.crc.txt file and prints it out nicely. The DoLoop() routine is/was used to try out the different algorithms and permutations.

[iWeather.crc.txt.gz](https://github.com/merbanan/rtl_433/files/2732984/iWeather.crc.txt.gz)
[figureitout.py.txt](https://github.com/merbanan/rtl_433/files/2732985/figureitout.py.txt)

Heres a sample set of lines. Xor all bytes and last nybble is always zero : 
```
aa a5 98 42 00 80 3d 00 00 0d 
aa a5 98 42 00 80 40 00 00 0d 
aa a5 98 42 00 80 41 00 00 01 
aa a5 98 42 00 80 41 00 00 0e 
aa a5 98 42 00 80 42 00 00 03 
aa a5 98 42 00 80 42 00 00 0e 
aa a5 98 42 00 81 3f 00 00 0d 
aa a5 98 42 00 81 40 00 00 0d 
aa a5 98 42 00 81 41 00 00 01 
aa a5 98 42 00 81 41 00 00 0d 
aa a5 98 42 00 81 42 00 00 1d 
aa a5 98 42 00 82 3f 00 00 0d 
aa a5 98 42 00 82 42 00 00 59 
aa a5 98 42 00 83 3e 00 00 0d 
aa a5 98 42 00 83 43 00 00 05 
aa a5 98 42 00 84 3a 00 00 0c 
aa a5 98 42 00 84 3e 00 00 0d 
aa a5 98 42 00 84 43 00 00 07 
aa a5 98 42 00 85 3d 00 00 01 
aa a5 98 42 00 85 3e 00 00 0d 
aa a5 98 42 00 86 3c 00 00 01 
aa a5 98 42 00 86 3d 00 00 01 
aa a5 98 42 00 86 40 00 00 27 
aa a5 98 42 00 86 42 00 00 09 
aa a5 98 42 00 87 3c 00 00 01 
aa a5 98 42 00 87 3f 00 00 26 
aa a5 98 42 00 87 42 00 00 18 
aa a5 98 42 00 88 3b 00 00 04 
aa a5 98 42 00 88 3c 00 00 04 
aa a5 98 42 00 88 41 00 00 2b 
aa a5 98 42 00 89 3b 00 00 0a 
aa a5 98 42 00 89 41 00 00 0b 
aa a5 98 42 00 8a 3b 00 00 0a 
aa a5 98 42 00 8a 40 00 00 4a 
aa a5 98 42 00 8b 3a 00 00 09 
aa a5 98 42 00 8b 3a 00 00 0a 
aa a5 98 42 00 8b 3e 00 00 37 
aa a5 98 42 00 8c 3a 00 00 09 
aa a5 98 42 00 8d 3a 00 00 1a 
aa a5 98 42 00 8d 3a 00 00 09 
aa a5 98 42 00 8d 3d 00 00 18 
aa a5 98 42 00 8e 39 00 00 09 
aa a5 98 42 00 8e 3a 00 00 09 
aa a5 98 42 00 8e 3c 00 00 04 
aa a5 98 42 00 8e 3f 00 00 4a 
aa a5 98 42 00 8f 39 00 00 09 
aa a5 98 42 00 8f 39 00 00 0a 
aa a5 98 42 00 91 39 00 00 09 
aa a5 98 42 00 91 39 00 00 0d 
aa a5 98 42 00 91 3a 00 00 1a 
aa a5 98 42 01 05 1a 03 b0 28 
aa a5 98 42 01 05 1a 03 b0 0d 
aa a5 98 42 01 05 1c 03 b0 00 
aa a5 98 42 01 05 1c 03 b0 2b 
aa a5 98 42 01 05 1d 03 b0 1c 
aa a5 98 42 01 05 1d 03 b0 1d 
aa a5 98 42 01 05 1d 03 b0 1f 
aa a5 98 42 01 05 1d 03 b0 2e 
aa a5 98 42 01 05 1d 03 b0 47 
aa a5 98 42 01 05 1e 03 b0 14 
aa a5 98 42 01 05 1e 03 b0 0d 
aa a5 98 42 01 05 1e 03 b0 0e 
aa a5 98 42 01 51 0f 00 00 29 
aa a5 98 42 01 51 0f 00 00 5c 
aa a5 98 42 01 51 0f 00 00 6c 
aa a5 98 42 01 05 1f 03 b0 1a 
aa a5 98 42 01 05 1f 03 b0 22 
aa a5 98 42 01 51 0f 03 b0 35 
aa a5 98 42 01 51 0f 03 b0 37 
aa a5 98 42 01 51 0f 03 b0 3b 
aa a5 98 42 01 51 0f 03 b0 5c 
aa a5 98 42 01 51 0f 03 b0 69 
aa a5 98 42 01 51 0f 03 b0 88 
aa a5 98 42 01 52 10 26 a0 2e 
aa a5 98 42 01 52 10 26 a0 3a 
aa a5 98 42 01 52 10 03 b0 32 
aa a5 98 42 01 52 10 03 b0 3c 
aa a5 98 42 01 52 10 03 b0 4e 
aa a5 98 42 01 05 29 03 b0 0f 
aa a5 98 42 01 52 0f 00 00 1f 
aa a5 98 42 01 52 0f 00 00 3e 
aa a5 98 42 01 52 0f 00 00 68 
aa a5 98 42 01 52 0f 00 00 6d 
aa a5 98 42 01 52 0f 00 00 7b 
aa a5 98 42 01 52 0f 03 b0 6e 
aa a5 98 42 01 53 10 26 a0 1e 
aa a5 98 42 01 53 10 26 a0 26 
aa a5 98 42 01 53 10 26 a0 38 
aa a5 98 42 01 53 10 26 a0 3e 
aa a5 98 42 01 53 10 26 a0 0d 
aa a5 98 42 01 53 10 03 b0 2c 
aa a5 98 42 01 53 0f 00 00 10 
aa a5 98 42 01 53 0f 00 00 12 
aa a5 98 42 01 53 0f 00 00 1f 
aa a5 98 42 01 53 0f 00 00 83 
aa a5 98 42 01 53 0f 03 b0 24 
aa a5 98 42 01 53 0f 03 b0 38 
aa a5 98 42 01 53 0f 03 b0 46 
aa a5 98 42 01 53 0f 03 b0 77 
aa a5 98 42 01 54 10 26 a0 49 
aa a5 98 42 01 54 10 03 b0 21 
aa a5 98 42 01 54 10 03 b0 23 
aa a5 98 42 01 54 10 03 b0 28 
aa a5 98 42 01 54 10 03 b0 2e 
aa a5 98 42 01 54 10 03 b0 3e 
aa a5 98 42 01 54 10 03 b0 4c 
aa a5 98 42 01 54 10 03 b0 0e 
aa a5 98 42 01 54 0f 00 00 38 
aa a5 98 42 01 54 0f 00 00 5c 
aa a5 98 42 01 54 0f 15 e0 1b 
aa a5 98 42 01 54 0f 03 b0 5a 
aa a5 98 42 01 54 0f 03 b0 79 
aa a5 98 42 01 54 0f 03 b0 99 
aa a5 98 42 01 55 10 03 b0 20 
aa a5 98 42 01 55 0f 00 00 2b 
aa a5 98 42 01 55 0f 00 00 3b 
aa a5 98 42 01 55 0f 00 00 69 
aa a5 98 42 01 55 0f 00 00 79 
aa a5 98 42 01 55 0f 26 a0 2e 
aa a5 98 42 01 55 0f 03 b0 34 
aa a5 98 42 01 56 10 26 a0 1c 
aa a5 98 42 01 56 10 26 a0 2f 
aa a5 98 42 01 56 10 03 b0 1c 
aa a5 98 42 01 56 10 03 b0 2e 
aa a5 98 42 01 56 11 26 a0 5c 
aa a5 98 42 01 56 12 26 a0 39 
aa a5 98 42 01 56 0f 00 00 17 
aa a5 98 42 01 56 0f 00 00 4a 
aa a5 98 42 01 56 0f 00 00 4b 
aa a5 98 42 01 56 0f 26 a0 3c 
aa a5 98 42 01 57 10 26 a0 2c 

```

zuckschwerdt, 2019-04-06:
Did you get around to finish the decoder?

merbanan, 2020-01-06:
@zuckschwerdt should I add this disabled by default? There are lots of payloads that can be used for test.

zuckschwerdt, 2020-01-07:
I'll polish the draft and submit asap.

fnarfbargle, 2020-01-07:
[holman.c.txt](https://github.com/merbanan/rtl_433/files/4030151/holman.c.txt)

I am so sorry guys. I've been using it ever since I put it together and I did have a go at cleaning it up, but real life has meant I've not had time to look at it. It has proven very reliable, which is nice but I never made any progress with the checksum.

I've attached the version I've been using.

fnarfbargle, 2020-01-07:
Oh, sorry I forgot it still has all the code in there to dump the debug info.
I did have a chance over winter to verify the rain gauge calculation. So it is functionally complete.

merbanan, 2020-01-07:
@fnarfbargle can you supply some signal samples also (-S known)? Then we can enable the decoder by default.

fnarfbargle, 2020-01-07:
Can do. The only 2 elements that don't vary in these samples are the low battery indicator and the rain gauge. TBH I forgot to check the low battery before I changed them last time. I'll make a note to check it before I swap them next time.

[Holman-WH5029-samples.zip](https://github.com/merbanan/rtl_433/files/4030361/Holman-WH5029-samples.zip)

I will say I've been running this with decoder 53 (Springfield Temperature and Soil Moisture) disabled as there is something in my vincinity that keeps triggering a false positive. I run this with the Holman sensor and 5 Acurite 606TX units. It also picks up my neighbours Oregon Scientific.

I log the Holman and the 606TX and I've not had any issues with the Holman. No false positives and very few missed packets, so I'm incredibly pleased about how well it works.

merbanan, 2020-01-07:
@fnarfbargle regarding the Springfield Temperature and Soil Moisture decoder, if you record a sample that trigggers false positives we can try to make it more robust. If so open a new issue with the samples.

zuckschwerdt, 2020-01-08:
We already have a "Holman WS5029", so the "WH" here is a typo? (Or H=Hub, S=Sensor?) Could this be the same device with different modulation (PCM vs PWM)? Strange.

fnarfbargle, 2020-01-08:
I've just checked the bottom of the display  and it's WS5029. My error.

I just checked the source and that in-tree one was committed in August 2019. Both refer to the same device sold at the same store, but mine is a few years old now.

I updated to the latest git (without my version) and the in-tree driver fails to detect my WS5029. I wouldn't put it past them to use whatever batch of TX/RX pairs they happen to get cheapest.

So yes, same device, same model number, different encoding I suppose. I note the in-tree driver does nothing with the checksum.

zuckschwerdt, 2020-01-08:
I'll merge your code (esp. checksum) with the other decoder. The actual decoding seems the same, just the modulation differs.

zuckschwerdt, 2020-01-08:
Note: This seems to be an older version with PWM modulation.
See #1119 for the newer version with PCM modulation.

closed 2020-01-08 by zuckschwerdt

fnarfbargle, 2020-01-09:
Just to close this out, updated, compiled & tested. Currently in use. Many thanks!

zuckschwerdt, 2020-01-09:
Thanks for the decoding, development, and feedback!

commit 9a23fbb9 (2020-01-08): Add Holman iWeather WS5029 older PWM (closes #947)