merbanan/rtl_433 #948: Exit code 135, no error message

j2ghz, 2019-01-07:
I'm trying to get rtl_433 working on raspberry pi, in docker. It just stops with error code 135.
```
rtl_433_1  | rtl_433 version 18.12-64-gc8c9a13 branch master at 201901052157
rtl_433_1  | Trying conf file at "rtl_433.conf"...
rtl_433_1  | Trying conf file at "/root/.rtl_433/rtl_433.conf"...
rtl_433_1  | Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
rtl_433_1  | Trying conf file at "/etc/rtl_433/rtl_433.conf"...
rtl_433_1  | Registered 96 out of 120 device decoding protocols [ 1-4 8 11-12 15-21 23 25-26 29-36 38-60 62-64 67-71 73-100 102-103 108-116 119 ]
rtl_433_1  | Detached kernel driver
rtl_433_1  | Found Rafael Micro R820T tuner
rtl_433_1  | Exact sample rate is: 250000.000414 Hz
rtl_433_1  | [R82XX] PLL not locked!
rtl_433_1  | Sample rate set to 250000 S/s.
rtl_433_1  | Tuner gain set to Auto.
rtl_433_1  | Tuned to 433.920MHz.
rtl_433_1  | Allocating 15 zero-copy buffers
rtl_433_rtl_433_1 exited with code 135
```

rct, 2019-01-08:
The exit code had the high bit set, so it died because of a signal.  135 & 0x7f == 7 which is SIGBUS.   So there was a bus error.    

Does it run on that system outside of the container? There is probably something missing from your container.  I'd try enabling core dumps so you can see where the problem is or possibly start a shell in the container and run it under gdb to get some idea of where the problem is.

There is likely a bug that should be filed for whatever is causing the SIGBUS.

j2ghz, 2019-01-09:
I'm running RancherOS (https://github.com/rancher/os#how-this-works). It works in system-docker, but does not work in user docker, so there is probably a problem there.

closed 2019-01-09 by j2ghz