merbanan/rtl_433 #949: LimeSDR mini not recognized

PaulSec, 2019-01-08:
Hello everyone, 

OS: Kali Linux (Debian) x64. 

Unfortunately, my LimeSDR mini does not get recognized and I am wondering what I am missing. 
I tried both on USB-2 and USB-3 ports. 

LimeSDR mini is recognized on my computer as you can see:

```
# LimeQuickTest 
[ TESTING STARTED ]
->Start time: Tue Jan  8 17:48:57 2019

->Device: LimeSDR Mini, media=USB 2.0, module=FT601, addr=24607:1027, serial=1D3AC898072512
Warning: USB3 not available
  Serial Number: 1D3AC898072512

[ Clock Network Test ]
->REF clock test
  Test results: 53771; 1432; 14629 - PASSED
->VCTCXO test
  Results : 6710998 (min); 6711159 (max) - PASSED
->Clock Network Test PASSED

[ FPGA EEPROM Test ]
->Read EEPROM
->Read data: 12 02 17 12 02 17 02
->FPGA EEPROM Test PASSED

[ LMS7002M Test ]
->Perform Registers Test
->External Reset line test
  Reg 0x20: Write value 0xFFFD, Read value 0xFFFD
  Reg 0x20: value after reset 0x0FFFF
->LMS7002M Test PASSED

[ RF Loopback Test ]
->Configure LMS
->Run Tests (TX_2 -> LNA_W):
  CH0 (SXR=1000.0MHz, SXT=1005.0MHz): Result:(-14.1 dBFS, 5.00 MHz) - PASSED
->Run Tests (TX_1 -> LNA_H):
  CH0 (SXR=2100.0MHz, SXT=2105.0MHz): Result:(-12.3 dBFS, 5.00 MHz) - PASSED
->RF Loopback Test PASSED

=> Board tests PASSED <=

Elapsed time: 2.96 seconds
```

When running rtl_* commands, it just states: "No supported devices found." as you can see:

```
# rtl_433
rtl_433 version unknown
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 96 out of 120 device decoding protocols [ 1-4 8 11-12 15-21 23 25-26 29-36 38-60 62-64 67-71 73-100 102-103 108-116 119 ]
No supported devices found.
```

I also tried this: 

```
# rtl_433 -d "" -g "LNA=20,TIA=8,PGA=2" -f 433.87M -A -S all
rtl_433 version unknown
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 96 out of 120 device decoding protocols [ 1-4 8 11-12 15-21 23 25-26 29-36 38-60 62-64 67-71 73-100 102-103 108-116 119 ]
``` 

I flashed the LimeSDR with the latest firmware and works perfectly with URH (https://github.com/jopohl/urh)

Thanks in advance and have a good day,

zuckschwerdt, 2019-01-08:
Can you post the output of
```
rtl_433 -d
```
and please try
```
rtl_433 -d driver=lime
```

PaulSec, 2019-01-08:
Hi there, 

Thanks for your (really fast) answer. Here it is: 

```bash
# rtl_433 -d
rtl_433 version unknown
rtl_433: option requires an argument -- 'd'
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
rtl_433: option requires an argument -- 'd'
	RTL-SDR device driver is available.
[-d <RTL-SDR USB device index>] (default: 0)
[-d :<RTL-SDR USB device serial (can be set with rtl_eeprom -s)>]
	To set gain for RTL-SDR use -g <gain> to set an overall gain in dB.
	SoapySDR device driver is not available.
[-d "" Open default SoapySDR device
[-d driver=rtlsdr Open e.g. specific SoapySDR device
	To set gain for SoapySDR use -g ELEM=val,ELEM=val,... e.g. -g LNA=20,TIA=8,PGA=2 (for LimeSDR).
[-d rtl_tcp[:host[:port]] (default: localhost:1234)
	Specify host/port to connect to with e.g. -d rtl_tcp:127.0.0.1:1234
```

```bash
# rtl_433 -d driver=lime
rtl_433 version unknown
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 96 out of 120 device decoding protocols [ 1-4 8 11-12 15-21 23 25-26 29-36 38-60 62-64 67-71 73-100 102-103 108-116 119 ]
```

Thanks in advance,

zuckschwerdt, 2019-01-08:
>  rtl_433 -d
> …
>	SoapySDR device driver is not available.

You need to recompile with SoapySDR lib and headers installed.
In the first few lines cmake will warn if SoapySDR is not found.

zuckschwerdt, 2019-01-08:
Out "version" line should probably include the information if RTLSDR or SoapySDR are available or not.

closed 2019-01-08 by zuckschwerdt

PaulSec, 2019-01-08:
Thank you so much, this is working!