merbanan/rtl_433 #950: RTL433 decoding FineOffset WH1080 pushing information to PLC Siemens S7-300 using Snap7

loengrin2000, 2019-01-09:
Dear Merbanan and Community, 

RTL433 receiving splendidly information from WH1080. I have found all the information (temp,humidity etc) in fineoffset_wh1080.c, however I have stacked while I cannot find in rtl_433.c how fineoffset_wh1080.c is called. I need to insert the Snap7 function call I assume in rtl_433.c main(). How can I have access in main() for the variables handled in fineoffset_wh1080.c? (msg_type, device_id, temperature, humidity, wind, gust, rain, battery status, winddir_str, hour, minute seconds, year, month and day). Thaks for the help!

zuckschwerdt, 2019-01-09:
You mention "Snap7", I guess you want to feed data to a PLC?
Use something like https://github.com/zuckschwerdt/rtl_433_relay/blob/master/rtl_433_mqtt_relay.py.
Maybe with a Snap7 wrapper like https://github.com/SCADACS/snap7

loengrin2000, 2019-01-09:
Hello Christian,
Yes, it is correct I am trying to write the float, int and char data obtained from fineoffset_WH1080 via rtl433 to a Siemens S7-300 PLC DB.  Since rtl433 written in c, I thought the easiest is to use the Snap7-iot-arm-version -> examples->plain-c-> client.c tailored and inserted to rtl433 to suit my needs. Its directly writes the values supposedly coming from fineoffset_WH1080.c to a PLC DB. I am not familiar with C programing, I am coming from PLC side. I have already made other interfacing with Snap7 plain-c client. I am pretty confident I am able to make Snap7 client code working in rtl_433.c the only problem is to how to get access to the int, float and char type of data generated in fineoffset_WH1080.c in rtl_433.c (if this question has any sense?)  (In  fineoffset_WH1080.c there is no - main () -  call but in rtl433.c there is so I assume there I can insert the Snap7 code snippet in the main() call but not without the data to be sent to the PLC available right in rtl433.c. The Snap 7 communication functions:
Client=Cli_Create();
Cli_SetAsCallback(Client, CliCompletion,NULL);
if (Cli_Connect())
{
MultiWrite(); //actual writing job
CliDisconnect();
};
Cli_Destroy(&Client);

I attach both my fineoffset_WH1080.c where the whole Snap7 data handling is located at the moment and the rtl433.c where the above function calls for snap7 comms job is located. fineoffset_WH1080.c is compiling fine but rtl433.c is not compiling without the declarations located in fineoffset_WH1080.c. Any suggestion is greatly appreciated!
[fineoffset_with_snap7.zip](https://github.com/merbanan/rtl_433/files/2741199/fineoffset_with_snap7.zip)

zuckschwerdt, 2019-01-09:
I unpacked the code here: https://github.com/zuckschwerdt/rtl_433/commit/67a311ab637ce0c8a61adeaf59b52685c26840d6

If you want this quick and dirty, then you are almost done.
Replace your changes in main() with
```
void init_all_the_things_snap7(void);
init_all_the_things_snap7();
```
and add at the end of main():
```
void kill_all_the_things_snap7(void);
kill_all_the_things_snap7();
```
Then in fineoffset_WH1080.c add
```
void init_all_the_things_snap7()
{
Client=Cli_Create();
Cli_SetAsCallback(Client, CliCompletion,NULL);
Cli_Connect();
}

void kill_all_the_things_snap7()
{
CliDisconnect();
Cli_Destroy(&Client):
}
```
The ` MultiWrite(); ` should go at the end of `fineoffset_wh1080_callback()` likely.

zuckschwerdt, 2019-01-09:
It would be great if you post the working code. It's interesting to see and we might add an output or something for Snap7 someday.

loengrin2000, 2019-01-10:
Dear Christian,
I guess on my current C programming capability I will go the quick and dirty.Compiler already complaining about questionable programming practices in my modified snap7 seasoned fineoffset_wh1080.c I appreciate your kind help! I will going to post you all the code, programming environment and testing rig configuration including Siemens Simatic Manager PLC program project file once I made this working. I will proceed with the proposed change and update you on the result! Thanks again!

zuckschwerdt, 2019-01-10:
The definitions from snap7.h are recognized but the functions are not available. You need to link the snap7 lib or something like that.
You should use CMake, not the old configure & make.
```
mkdir build
cd build
cmake ..
make
```

edit src/CMakeLists.txt and add
```
target_link_libraries(rtl_433 snap7)
```
or something like that.

loengrin2000, 2019-01-11:
Hello Christian, 
Your valuable instruction again allow me to take one step further however I sucked yet again. Compilation goes OK, but during linking I got the error message as shown in the picture (undefined reference to 'MultiWrite':
[image]
Also the call structure in fineoffset_wh1080.c correctly is like:
//------------------------------------------------------------------------------
// Unit communication
//------------------------------------------------------------------------------
void init_all_the_things_snap7()
{
	Client=Cli_Create();
	Cli_SetAsCallback(Client, CliCompletion,NULL);
	if (CliConnect())
	{
	MultiWrite();
	};
}

void kill_all_the_things_snap7()
{
	CliDisconnect();
	Cli_Destroy(&Client);
}

MultiWrite(); is defined in fineoffset_wh1080_callback also during compilation there is a warning that MultiWrite(); is a nested function and that is not ISO C (pedantic C99?) compatible. 
[all_file_modified_or_critical.zip](https://github.com/merbanan/rtl_433/files/2750330/all_file_modified_or_critical.zip)
As always your help is highly appreciated!

loengrin2000, 2019-01-14:
Hello Christian,
Thank to you kind help compilation was finally went through without error and initial result are shown below: (I will send later some description on the test environment and all the code).
1: Siemens S7-300 compatible PLC -  DB92 (data block) displaying the correct values coming from Fineoffset WH1080 station on 868.3MHz captured by rtl433 and relayed by snap7. Data formatting can be improved but also the existing data can be processed easily by the PLC itself:
[image]
executing rtl433 shows PLC connection has been established on 192.168.0.10 address (hardcoded in fineoffset_wh1080.c) and the multiwrite function executes without error writing the displayed variables to PLC:
[image]
Cyclic updates coming from Fineoffset WH1080 and processed by rtl433:
[image]

zuckschwerdt, 2019-01-14:
Very nice! Perhaps you can describe where/why a PLC useful for your project (only very general motivation).
It is really interesting to see all the places where rtl_433 can help :)

loengrin2000, 2019-01-14:
I will include a short description among the other notes. I will try to make that ready this week. To be still done: 1. test the time data coming from the weather station. (not happening very often so it might take a while.
2. data format to be improved
3. IP, rack and slot parameters as rtl_433 calling arguments to be passed over to embedded snap7.
4. general revision of the questionable C programming practices in the modified code. (quick and dirty solution)

loengrin2000, 2019-01-17:
Hello Christian,
Here is the final version of this proof of concept. A pdf description with photo illustration included. Source codes are also included and the zipped Siemens Simatic Manager project archive is also there, all in one zip file below.
[RTL433_Snap7_integration_description.zip](https://github.com/merbanan/rtl_433/files/2768004/RTL433_Snap7_integration_description.zip)

merbanan, 2020-01-06:
I think the purpose of this issue is fulfilled. One of the project goals was to make it possible for this kind of use so it nice to see that someone actually was able to do it.

closed 2020-01-06 by merbanan

zuckschwerdt, 2020-01-07:
I need to get around to package this or write some instructions. It's a neat use case.