merbanan/rtl_433 #951: Support for LaCrosse WS-3600

pong-at, 2019-01-10:
Hello, first I would like to thank you all for your great work.

The WS-3600/TX13 has an similar protocol to the WS-2310 (lacrossews.c), with a few minor differences - new preamble, different temperature offset and slightly different short/long pulses. The CRC8 check still works. In my file, I set short_width=340 / long_width=1440, but the original 2310 values have no problem detecting the 3600. Probably best to not change them? I have no idea how these r_device settings should be set up for 2 stations with very similar protocols. 

here is my file: 

https://github.com/pong-at/rtl433test/blob/master/lacrossews.c

Let me know what you think, sorry I'm not a coder...

merbanan, 2019-01-10:
Hi, we need signal samples can you provide some?

pong-at, 2019-01-10:
hope this works, here is the first file - I will record some more.

https://github.com/pong-at/rtl433test/blob/master/lacrossws3600_433.8M_250k.cu8

recorded with rtl_433 -f 433800000 -S all

rtl_433 -r lacrossws3600_433.8M_250k.cu8
rtl_433 version 18.12-64-gc8c9a13 branch master at 201901052157
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/zero/.rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 96 out of 120 device decoding protocols [ 1-4 8 11-12 15-21 23 25-26 29-36 38-60 62-64 67-71 73-100 102-103 108-116 119 ]
Test mode active. Reading samples from file: lacrossws3600_433.8M_250k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.294440s   brand     : LaCrosse
model     : WS-3600      ws_id     : 6             id        : 196
Temperature: 0.3 C
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.563612s   brand     : LaCrosse
model     : WS-3600      ws_id     : 6             id        : 196
Humidity  : 87
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.829244s   brand     : LaCrosse
model     : WS-3600      ws_id     : 6             id        : 196
Wind speed: 0.0 m/s      Direction : 180.000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @1.898608s   brand     : LaCrosse
model     : WS-3600      ws_id     : 6             id        : 196
Wind speed: 0.0 m/s      Direction : 180.000

pong-at, 2019-01-11:
I recorded some new data, as suggested at 433.75 and 433.7 MHz. The default settings are out of range for the Realtek Sdr. I originally set the frequency to 433.8 so that I can access all my sensors, for test records this is obviously not a good idea, thanks for pointing that out. I found some records from the WS-2310 in the lacrosse/03 directory, they are very similar and get decoded correctly.

https://github.com/merbanan/rtl_433_tests/pull/259

zuckschwerdt, 2019-04-06:
Support added with 7ef7b8a. Reopen if you notice problems.

closed 2019-04-06 by zuckschwerdt