merbanan/rtl_433 #952: error: require no space between '-L' and '-lrtlsdr'

akkana, 2019-01-10:
I checked out the latest on Debian testing, and followed the autotools instructions in BUILDING.md:

sudo apt-get install libtool libusb-1.0.0-dev librtlsdr-dev rtl-sdr build-essential autoconf cmake pkg-config
autoreconf --install
./configure
make

The cmake instructions worked, except that it required an additional sudo apt-get install doxygen.
The compilation failed with:

  CCLD     rtl_433
libtool:   error: require no space between '-L' and '-lrtlsdr'
make[2]: *** [Makefile:878: rtl_433] Error 1
make[2]: Leaving directory '/home/akkana/outsrc/rtl_433/src'
make[1]: *** [Makefile:497: all-recursive] Error 1
make[1]: Leaving directory '/home/akkana/outsrc/rtl_433'
make: *** [Makefile:383: all] Error 2

akkana, 2019-01-29:
Just got the doxygen warning on a new build on raspbian, and you're right, it went ahead and built, so it was just a warning that I misinterpreted as an error.

zuckschwerdt, 2019-01-31:
What does `pkg-config librtlsdr --libs` show on your system?

zuckschwerdt, 2019-02-13:
Reopen if this appears again and post the output of `pkg-config librtlsdr --libs`.

closed 2019-02-13 by zuckschwerdt

alexmyczko, 2019-03-22:
```
$ pkg-config librtlsdr --libs
-L -lrtlsdr
```

zuckschwerdt, 2019-03-22:
Ah, nice. That's looks like the culprit. Just `-L` isn't valid. Can you find and post `librtlsdr.pc`? Should be something like `/usr/local/lib/pkgconfig/librtlsdr.pc`or `/usr/lib/pkgconfig/librtlsdr.pc`.

alexmyczko, 2019-03-25:
Sure, I only have 
```
# locate librtlsdr.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/librtlsdr.pc
```

and it says:
```
prefix=
exec_prefix=
libdir=
includedir=

Name: RTL-SDR Library
Description: C Utility Library
Version: 0.6.0
Cflags: -I${includedir}/
Libs: -L${libdir} -lrtlsdr
Libs.private:  -lusb-1.0 
```

zuckschwerdt, 2019-03-25:
Yes, that's a broken file. The first lines should read something like
```
prefix=/usr
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=${prefix}/include
```
Whatever installed those (is this the librtlsdr-dev deb?) did a bad job.