merbanan/rtl_433 #961: No binary data and only Hex

ogonbat, 2019-01-18:
Hi All

first of all it's my first experience with rtl_433 and maybe my question is stupid
I want todo a copy of my remote garage door.
i have started rtl_433 and obtained this output:

> signal_bsize = 262144  -      sg_index = 0
> start_pos    = 2579274  -   buffer_size = 3145728
> *** Saving signal to file g002_433.92M_250k.cu8
> *** Writing data from 2579274, len 262144
> *** signal_start = 2874429, signal_end = 2984689, signal_len = 110260, pulses_found = 312
> Iteration 1. t: 81    min: 57 (173)    max: 106 (139)    delta 17
> Iteration 2. t: 81    min: 57 (173)    max: 106 (139)    delta 0
> Distance coding: Pulse length 81
> 
> Short distance: 39, long distance: 100, packet distance: 15155
> 
> p_limit: 81
> bitbuffer:: Number of rows: 4 
> [00] {77} 00 13 d9 73 3a 2d 15 33 14 20 
> [01] {77} 00 13 d9 73 3a 2d 15 33 14 28 
> [02] {77} 00 13 d9 73 3a 2d 15 33 14 20 
> [03] {77} 00 13 d9 73 3a 2d 15 33 14 28 

I have attached a file as file as well
[g002_433.92M_250k.log](https://github.com/merbanan/rtl_433/files/2774386/g002_433.92M_250k.log)

Thanks in advance for your help

zuckschwerdt, 2019-01-18:
There is a new wiki entry which should help. https://github.com/merbanan/rtl_433/wiki/View-a-saved-cu8-file

Geoff99, 2019-01-19:
Hi @ogonbat 

Welcome to a lot of fun :-)

Not sure why your attached file ends in .log, not .cu8, but I managed to have a quick look at it using sigrok / PulseView anyway (see the wiki entry mentioned above for further details)

Then I used the flex decoder option (see 

* https://github.com/merbanan/rtl_433/wiki/digitech_xc0324.README#test-the-demodulation-parameters  or 
* run `rtl_433 -X -h`)

 to have a look at your signals.

`rtl_433 -X "n=garage_door,m=OOK_PPM,s=225,l=424,g=3000,r=80000" -Mbits -r <path to your file>`
 worked for me (though the demodulation parameters might need a bit more finessing).

If you want the output as a csv file, include additional options `-R0 -Fcsv:<filename>`

The `-R0` disables all the other decoders (which stops them putting out all their header lines for the csv file),
and the `-F...` tells `rtl_433` that you want output as csv

PS @zuckschwerdt  On my testing a moment ago, it looks like the flex decoder output doesn't generate the bit level field output even though it puts out the `bits` column header ,only the hex `codes` value appears - I haven't checked, but maybe the wrong decoder  util function is being invoked?

Geoff99, 2019-01-19:
PPS @ogonbat 

Your signal appears to have a wake up sync of 12 bits (`OxFFF`), followed by a small gap then a 66 bit message (all of which, except for the last byte, is constant in the signal data you have posted).

Geoff99, 2019-01-19:
PPPS @ogonbat 

Just spotted a typo in my first reply :-( 

.  I had `OOK_PPM` instead of `OOK_PWM` like it should be.  Fixed now on Github, but posting this in case you try o use the version sent by email.

zuckschwerdt, 2019-01-19:
@Geoff99 Yes the flex decoder output needs to be changed to the proper output functions.

ogonbat, 2019-01-19:
Hi @Geoff99 and @zuckschwerdt any idea about wich kind of decoder output i have to configure?
thanks in advance

ogonbat, 2019-01-19:
I have uploaded two cu8 files
the 003 is the open and 004 is the close command
maybe this can help you :-)

[Archivo comprimido.zip](https://github.com/merbanan/rtl_433/files/2775884/Archivo.comprimido.zip)

Geoff99, 2019-01-20:
Hi @ogonbat 

I've done the best I can with your files.  I'm not all that familiar with door opener signals (I'm quite new at this myself, only deciphered a single (very simple) temperature sensor so far), but anyway :

* I ran `tests/sigrok-open.sh` on your files.  They : 

  + start with 12 constant width pulses (~220 us (microseconds) on, 160 us off), then 
  + a gap of 1930 us, then 
  + a sequence of variable width pulses (the main message); followed by 
  + 60,000 us pause and another repeat of the same sequence
  + in total there are 4 repeats per file

* I ran the analyser over them `rtl_433 -A -r <path to your files>` and it suggested a set of arguments
  for the flexible decoder (`-X` option).

* I experimented with the `gap=` and `reset=` parameters to get what seemed to me to be a better result
  (the 1930us and 6000us gaps "mess" things up a tiny bit, or (more likely) I just haven't figured out how to specify things the best way yet), and I've put the resulting version into a `rtl_433.conf` file, so I don't have to keep typing them in on the command line.  The conf file is in the attached temp.zip file.  Running `rtl_433 -X help` will tell you the options for storing it if you want rtl_433 to use it.

* I also ran `rtl_433 -F csv:temp.csv -r <path to your files/g*cu8>` with that conf file on my machine,  and have included the temp.csv file in the temp.zip file as well.  It produces a hex representation of your signals (4 repeated rows per file).  All your messages begin with `Oxfff` (that's the first 12 bits which I think are a wake up / sync signal).  Apart from that, I can't figure out what the remaining bytes mean (part will be an id for **your** garage door, part will be the message; there may well be some security mechanism in there; rolling id codes ?  Anyway, nothing I know much about or can help with unfortunately

Hope that's enough to help you along a bit.

[temp.zip](https://github.com/merbanan/rtl_433/files/2776183/temp.zip)

zuckschwerdt, 2019-09-05:
I hope this is resolved. Post more if you need further help.

closed 2019-09-05 by zuckschwerdt