merbanan/rtl_433 #965: Add Klimalogg/30.3180.IT Support

devZer0, 2019-01-27:
is there somebody willing on supporting adding support for these sensors?
i have them running with tfrec but would like to see support in rtl_433 because it is a more capable tool. i can provide any input... i have access to 5 sensors

# tfrec -D
Registering demod for TFA_1 KlimaLoggPro
Registering demod for TFA_2 sensors, 17240 bit/s
type 0x1: Samples per bit: 22.3
Registering demod for TFA_3 sensors, 9600 bit/s
type 0x2: Samples per bit: 40.0
Found Rafael Micro R820T tuner
AUTO GAIN
Frequency 868.2500MHz
Samplerate 1536000
START READ THREAD
#000 1548582298  2d d4 3f 14 85 23 34 60 20 56 c3           ID 3f14 +12.3 52%  seq 2 lowbat 0 RSSI 81
#001 1548582298  2d d4 43 56 85 75 2e 60 10 56 78           ID 4356 +17.5 46%  seq 1 lowbat 0 RSSI 81
#002 1548582301  2d d4 63 41 86 27 22 60 a0 56 f2           ID 6341 +22.7 34%  seq a lowbat 0 RSSI 74
#003 1548582302  2d d4 6f 44 86 44 1c 60 30 56 be           ID 6f44 +24.4 28%  seq 3 lowbat 0 RSSI 77
#004 1548582302  2d d4 39 96 85 84 2c 60 10 56 9d           ID 3996 +18.4 44%  seq 1 lowbat 0 RSSI 81

# rtl_433 -f 868.25M -s 1536000 -G
rtl_433 version 18.12-71-gb8578dc branch master at 201901231237 inputs file rtl_tcp RTL-SDR
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 114 out of 120 device decoding protocols [ 1-4 6-8 10-26 29-64 67-120 ]
rtl_433: warning: 104 "Wireless M-Bus, Mode C&T, 100kbps (-f 868950000 -s 1200000)" does not support CSV output
rtl_433: warning: 105 "Wireless M-Bus, Mode S, 32.768kbps (-f 868300000 -s 1000000)" does not support CSV output
rtl_433: warning: 106 "Wireless M-Bus, Mode R, 4.8kbps (-f 868330000)" does not support CSV output
rtl_433: warning: 107 "Wireless M-Bus, Mode F, 2.4kbps" does not support CSV output
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sample rate set to 1536000 S/s.
Tuner gain set to Auto.
Tuned to 868.250MHz.

zuckschwerdt, 2019-01-27:
Sure, I'll take a look.
I contacted the [tfrec](https://github.com/baycom/tfrec) guys on joining efforts some time ago but the project seems somewhat dormant.

devZer0, 2019-01-27:
i would rename the issue title of https://github.com/baycom/tfrec/issues/7 to better reflect the intention

zuckschwerdt, 2019-01-27:
I've added NRZ-S decoding (cce293c) so support should be simple. Can you grab some (3-5) signals and add them along with a readme describing the sensor and the approx. grabbed values. S.a. https://github.com/merbanan/rtl_433_tests#contributing

devZer0, 2019-01-27:
great. how to submit? should i fork into my own repo and then add that files and create a pull request - or better send via email to you ? (unsure, because seeing unmerged pull requests at https://github.com/merbanan/rtl_433_tests/pulls )

zuckschwerdt, 2019-01-27:
PR to rtl_433_tests is prefered. Some PRs there are pending on various updates and changes, but generally we always merge directly.

anse1, 2019-01-28:
devZer0 writes:

> is there somebody willing on supporting adding support for these sensors?

I do have a working gnuradio receiver for this sensor:
https://github.com/anse1/klimalogg/

I tried getting it into rtl_433 but got frustrated when the ad-hoc
manual decoder was extremely sensitive to manual level adjustment for it
to keep synchronization.  I might give it another try now that there's
more than one person interested…

regards,
Andreas

$ for f in *cu8; do ./rtl_433 -r $f -l 773 -X name=klimalogg,modulation=OOK_PCM,short=48,long=49,reset=65,match=0x55df755dd; done
Test mode active. Reading samples from file: g001_868.3M_1000k.cu8
time      : @0.267816s
model     : klimalogg    count     : 1             num_rows  : 1             rows      : 
len       : 553          data      : fdd555555555555555555555555555555555555555555555555555555555555555555555555555555555df755ddff755fddf7d5775f555f5557d55557ddd5ddf55555555550
codes     : {553}fdd555555555555555555555555555555555555555555555555555555555555555555555555555555555df755ddff755fddf7d5775f555f5557d55557ddd5ddf55555555550
Test mode active. Reading samples from file: g002_868.3M_1000k.cu8
time      : @0.075392s
model     : klimalogg    count     : 1             num_rows  : 1             rows      : 
len       : 553          data      : fdd555555555555555555555555555555555555555555555555555555555555555555555555555555555df755ddfd755fddf7d5775f555f5557d55d57ddd77f755555555550
codes     : {553}fdd555555555555555555555555555555555555555555555555555555555555555555555555555555555df755ddfd755fddf7d5775f555f5557d55d57ddd77f755555555550
Test mode active. Reading samples from file: g003_868.3M_1000k.cu8
time      : @0.325256s
model     : klimalogg    count     : 1             num_rows  : 1             rows      : 
len       : 553          data      : fdd555555555555555555555555555555555555555555555555555555555555555555555555555555555df755ddff755fddf7d57fdf555f5557d55757dddf7df55555555550
codes     : {553}fdd555555555555555555555555555555555555555555555555555555555555555555555555555555555df755ddff755fddf7d57fdf555f5557d55757dddf7df55555555550

devZer0, 2019-02-03:
i collected testdata for klimalogg and have done a pull-request.

isn't that a little bit too large binary repo? checking out >1gig for adding some megabytes is rather inefficient and afair, git is not made for storing large amounts of binary data

zuckschwerdt, 2019-02-04:
Nearly 5000 files, each around 200KB. Git is still well suited for that (large files wouldn't be too good, many files are ok).
But you are right, adding samples should be easier and better structured (guided, example Readme).

Ideas anyone?

merbanan, 2019-02-04:
We could rebuild the git. That should help getting rid of some history. But other then that the only way is to thin out the sample archive. One could try to check out a subset. But I'm not sure how that would work.

zuckschwerdt, 2019-02-04:
I just tested recreating the repo and it isn't any smaller. We are mostly only ever adding stuff so that sounds about right.
We could perhaps shave of 5% by removing files. But there are no obvious offenders:
```
find tests -type d | while read d ; do echo $(find "$d" -type f -maxdepth 1 |wc -l) $(du -sh "$d") ; done | sort -g
```
Even the over stuffed sample collections (the last lines there) don't weigh too heavy.

Perhaps we could set up a mostly empty branch to pull (shallow clone) and commit to? Not sure how much hassle it would be to move the new commits.

merbanan, 2019-02-04:
As far as I understood the code logic the demodulated signal is low pass filtered. Then 2 estimators of the low and high level needs to converge (the symbol threshold is then just the mid point between these levels). I think that is the hardest demand of this generic demodulator. These level estimators are just set by some initial heuristics that might not be optimal for this type of signal. How to make it better is not really clear to me but in theory because we have the complete signal we can analyse the complete signal in one pass to recover optimal parameters that we use in a second bit recovery pass.

Maybe we need to move the problem to the the frequency domain. There was someone who did an OOK demod via fftw.

zuckschwerdt, 2019-02-04:
I guess FFT would be really expensive. I made a note to look into wavelets for cheaper FSK, but so far I havn't delved into that. Also someday I want to try a FIR lowpass filter instead of the PD_MIN_PULSE_SAMPLES counter.

For the estimators we can perhaps use a proper mean+variance band and better see where transitions are. That would apply to (non-gaussian) OOK only. For GFSK we probably need another detector -- measuring the direction/slope of the frequency shift looks more reliable than looking at the level. I'll try to finish the modularization soon, so people can mess around with (many) different detectors easily.

MichaelRenner, 2019-04-07:
Is there any news here? I would like to use the KlimaLOGG too. Apparently rtl_433 does not know the sensor yet.

merbanan, 2019-04-07:
I am slowly working on a new FSK capable demodulator. If/when that is in place we will be able to better support more types of FSK signals.

MichaelRenner, 2019-04-07:
Thanks for this info, good to know. And thank you for writing such an useful application!

merbanan, 2019-04-07:
Try with changing  PD_MIN_PULSE_SAMPLES to 5. If that works we can start figuring out the payload structure. And try to record a clear set of sample signals.

MichaelRenner, 2019-04-10:
I gave up. Regardless of the gain settings, a lot of crap was received, sometimes with only one bit - even with a dummyload instead of an antenna. And even worse: Even with tfrec, which actually supports this device, no decode was possible.
Now I read the signals of the neighbors :-)

merbanan, 2019-04-15:
Well adding a few signals (from each sensor) should be a fair contribution.

MichaelRenner, 2019-04-16:
Well, that motivated me to keep looking for it. I found the sensor 100kHz higher than expected. I collect the data soon.

merbanan, 2019-10-13:
The samples in the samples repository are distorted. We now have a wip fsk demod that might work with this sensor.

https://github.com/merbanan/rtl_433/tree/feat-fsk_demod_new

tophee, 2019-11-11:
How can I test this? (Sorry, if this is a dumb question. I am just getting started with rtl_433...)

merbanan, 2020-01-07:
This signal is ASK modulated, not a FSK based one. Looks like a demodulator rework is needed to be able to decode this signal.

merbanan, 2020-01-07:
@tophee nothing to test yet.

merbanan, 2020-01-07:
This is highly suspicious. The tfrec project uses a FSK demod for this. The samples are all distorted, I need better samples before I do any further work.

tophee, 2020-01-09:
How can I provide you with those samples?

merbanan, 2020-01-09:
Upload them to the issue.

tophee, 2020-01-09:
Yes, but how do I obtain samples from my Klimalogg devices and ensure that they are usable (i.e. not distorted)?

merbanan, 2020-01-09:
rtl_433 -S unknown -g 30, then open them with examples/sigrok-open.sh and look at the waveform. The bits should be decoded properly and the waveform should not have any clipping. You will probably need to adapt the gain parameter (-g).

alxlo, 2020-01-31:
Here are some samples recorded with rtl_433 -S unknown -g 42 -f 868250000
I hope I did this right and it helps.
[tfa_30.3180_samples.zip](https://github.com/merbanan/rtl_433/files/4137204/tfa_30.3180_samples.zip)

If it would be any help, I'm very willing to donate one of these sensors to you.

merbanan, 2020-01-31:
The signals look ok, but they are ASK modulated. And the current code misses the whole preamble and a few bits in the payload. Anyway rtl_433_tests/tests/TFA-KlimaLogg-30.3180.IT/02/g006_868.25M_1536k.cu8 demodulated correctly.

src/rtl_433 -A -r ../rtl_433_tests/tests/TFA-KlimaLogg-30.3180.IT/02/g006_868.25M_1536k.cu8 -X name=klimalogg,modulation=OOK_PCM,short=80,long=80,reset=6500,preamble=0x55df755dd

mueckinger, 2020-01-31:
I love this sensors. They are cheap and extremely energy effiicent. Battery life is more than 3-5 years. I operate 6 of them and look forward to read them with rtl_433.

merbanan, 2020-02-15:
src/rtl_433 -l 15000 -s 1536k -A -r ../rtl_433_tests/tests/TFA-KlimaLogg-30.3180.IT/01/g001_868.25M_1536k.cu8

Adding -l 15000 gives a proper pulse recovery and this sample can be decoded by tfrec.

```
./tfrec -L ../../rtl_433_tests/tests/TFA-KlimaLogg-30.3180.IT/01/g001_868.25M_1536k.cu8
Registering demod for TFA_1 KlimaLoggPro
Registering demod for TFA_2 sensors, 17240 bit/s
type 0x1: Samples per bit: 22.3
Registering demod for TFA_3 sensors, 9600 bit/s
type 0x2: Samples per bit: 40.0
Dumpmode -1 (LOAD), dumpfile ../../rtl_433_tests/tests/TFA-KlimaLogg-30.3180.IT/01/g001_868.25M_1536k.cu8
TFA1 ID 3f14 +18.9 43% seq 9 lowbat 0 RSSI 82
done reading dump
```

alxlo, 2020-02-16:
Seriously, I would be more than happy to let Amazon ship one or two of these sensors to you or help you in any other way.

merbanan, 2020-02-16:
There is now a PR for this sensor https://github.com/merbanan/rtl_433/pull/1319. Try it out you might need to play with the -l parameter.

merbanan, 2020-02-16:
@alxlo can you ship me some development time? :) Anyway these sensors does not fit into the current generic dsp model. The current code might work with a samples rate of -s 1536k and -l in the range of 5000-15000.

alxlo, 2020-02-16:
With e. g.  -g 30 -f 868250000 -l 15000 -s 1536k -R 150  #1319 provides quite consistent results with a sensor in the same room as long as -g is not to high. The outdoor sensor (15m and 3 brick walls away) gives no results, even with max gain and any amount of fiddling with -l

@merbanan  If you find a means for shipping dev time I gladly would :-)  The effort and time you invest in this project is greatly appreciated!

mueckinger, 2020-02-17:
With -l 15000 I receive 2 out of 6 sensors (wooden wall).
With -l 24000 I receive 2 others (brick wall and wooden ceiling) but not the ones above.
(Documentation says -I 16384 is maximum !?!)
Any idea why I cannot receive 4 sensors with same settings?
Actually my Klimalogg base station is able to receive 5/6 sensors.
So I am missing one sensor.
-g was set to 20 (19.2dB seems to be the max gain of my tuner).

zuckschwerdt, 2020-02-17:
@mueckinger using RTL-SDR (8 bit) the maximum is 32768 with full scale at 16384 (amplitude, i.e. linear), for SoapySDR the maximum is 22144, full scale 16384 (magnitude, i.e. log).
Everything above full scale is a distorted signal. (think of full scale as sine wave at full amplitude).

merbanan, 2020-02-18:
As I said the current dsp code is not suited at all for this sensor. I am quite sure this is not the proper way to demodulate this signal. Ther must be a better way.

And I have no idea on how to get rid of the -l option.

zuckschwerdt, 2020-02-18:
The Spectrogram looks like FSK with those two dominat frequencies, which is strange. Looking at Sigrok Pulseview the samples look like OOK (or ASK). A simple moving average threshold seems to cut the levels well. Look at .e.g. 02/g013 -- that's FSK, two frequencies and two strong harmonics plain visible. But the amplitude carries a clean preamble where the frequency looks random. The sample is clipped badly (too much gain) but the FSK looks more like a clock than data? Not sure I understand the signal.

All samples hit the DC, which is not great. Can we get a sample at lower gain and with a frequency offset? E.g. `-f 868M`. 02/g006 has the right gain -- in https://triq.org/iqs/ white pixels need to be rare.

zuckschwerdt, 2020-02-18:
@merbanan I've added an old demod evaluation example, update for the native sigrok writer with f5e5a6a. I find it useful to quickly try some algorithms there and visualize the output.

merbanan, 2020-02-20:
@zuckschwerdt I mainly prototype things in Octave. Anyway the tfrec code has this:

```
//-------------------------------------------------------------------------
// Vastly reduced mixed demodulator for FM-NRZS:
// It doesn't need to be linear, nor do we care about frequency shift direction
//-------------------------------------------------------------------------
int fm_dev_nrzs(int ar, int aj, int br, int bj)
{
	int cr=ar*br+aj*bj;

	// This limits also the max RSSI
	if (cr>1e9)
		cr=1e9;
	if (cr<-1e9)
		cr=-1e9;
	return cr;
}
```

ar = I[n]
aj = Q[n]
br = I[n-1]
bj = Q[n-1]

I'm not sure if it makes sense to do any further work. At least without the real sensor. And currently I'd rather work on enabling 2 fsk demods and some other parts that I need so I'm not sure sending a sensor to me is the best use of resources.

The signal could be dual tone modulated. But this puzzles me much. We might have to look at the transceiver chip or ask an expert.

zuckschwerdt, 2020-02-20:
That's a very fast phase difference estimator. But it's amplitude sensitive.
Usually you want [arg](https://en.wikipedia.org/wiki/Argument_(complex_analysis))(x[n] × [conj](https://en.wikipedia.org/wiki/Complex_conjugate)(x[n-1])), e.g. here [atan2](https://en.wikipedia.org/wiki/Atan2)(aj×br-ar×bj, ar×br+aj×bj)

So they are looking at the FSK. I can't seem to get anything from that. At least with the DC hitting samples. I wouldn't expect a OOK to be visible? Maybe an artifact of DC correction or such. We do need samples at lower gain and with a frequency offset.

zuckschwerdt, 2020-02-20:
Ah, wait, with that crude estimation they (unintentionally?) *do* look at the OOK…

merbanan, 2020-02-21:
The calling code calculates how many 1-bit duration's fit between the "OOK" pulses. I do the same in my code but this is unfiltered whole rtl_433 uses and IIR filter.

cr=ar*br+aj*bj; -> i[n]*i[n-1] + q[n]*q[n-1] -> ~ I^2 + Q^2

That is the same as rtl_433 uses to demodulate OOK.

For some reason that works, but looking at a frequency spectrum one can see 2 frequencies present. But looking at the waveform both frequencies are present at the same time at least when one of the bits are transmitted.

zuckschwerdt, 2020-02-21:
If you use x[n] and x[n-1] then you are looking at (FSK) phase differences. If you want a (OOK) amplitude that's x[n]^2, and for the magnitude it's sqrt(x[n]^2).

merbanan, 2020-02-21:
I know I just thought that at 1.5Ms that  n~=n-1. But the difference might be 4. Anyway I figured out the signal. Octave code follows:

```
m=0:0.0003:3*2*pi;
l=0:0.0001:2*pi;
sig=sin(l)+sin(m);
plot(l,sig)
```
[Klimalogg_signal.pdf](https://github.com/merbanan/rtl_433/files/4238314/Klimalogg_signal.pdf)

This matches what is seen when looking at the signal visually. And it makes sense for this type of device. There is a carrier always transmitted at Y Hz when a one symbol is transmitted and 3*YHz+YHz (actually modulated 2 times) when a zero symbol is transmitted (with a slowly drifting phase). What can this be called, maybe OOK added with a carrier. From the Audacity frequency analyzer I get 555Hz on the lower frequency and 1670Hz on the high frequency. This confirms the frequency difference of 3 times.

So what to do from here?

merbanan, 2020-02-21:
The Hz values are from defauls of 44100Hz with working with Audacity. The times 3 factor should still be valid. And a PSK demodulator might work. Or high-pass filter and then a envelope detector.

zuckschwerdt, 2020-02-22:
The plot is time vs. amplitude in the I (or Q) signal? I don't grasp how changing the carrier frequency affects the amplitude. Are the 0-amplitude parts the edges of the data or are they one of the bits?

merbanan, 2020-02-22:
Well I only figured out how the signal was generated on the transmitter side. There is no amplitude changes in reality just this addition of a sine wave with a times 3 frequency. When you do an envelope estimation it doesnt match how we expect a transmission would look. A sin(x) is the modell we assume. The estimation recovers the A.  And we have sin(x)+sin(3x).

merbanan, 2020-02-22:
Ok, the above description might be wrong but not so far away, by looking at the signal there are parts that look like a pure sine wave and others that looks like a sine wave + sine wave at an offset.

merbanan, 2020-02-22:
If you apply a band-pass filter to the offset signal (ie get rid of the carrier) it becomes very clear that the offset signal turns on and off (OOK). But how to do generic pulse recovery is highly unclear to me.

Anyway 

```
env(i) = in_i(i)*in_i(i-4) + in_q(i)*in_q(i-4);
env(i) = in_i(i)*in_i(i) + in_q(i)*in_q(i);
```
gives almost the same pulse output.

merbanan, 2020-02-22:
[Klimalogg_demod.pdf](https://github.com/merbanan/rtl_433/files/4239399/Klimalogg_demod.pdf)

alxlo, 2020-02-25:
In the hope that this will be helpful, here are some more samples taken with varying gain settings via
`rtl_433 -S unknown -g <40,35,30,25,20,15,10> -f 868250000`

[2020-02-23_samples.zip](https://github.com/merbanan/rtl_433/files/4249440/2020-02-23_samples.zip)

Setup and conditions where as follows. This was a place where no or minimal interference with other sensors should occur. There might be interference from other strong radio sources due to an airport nearby.

| ID |  type | °C / %rH<br>before  | °C / %rH<br>after  | location  |
|----|---|---|---|---|
|  0cde  | 30.3180  | 11.1 / 81 | 11.5 / 80  | 4m away, through wooden wall/window  |
|  0706  |  30.3180 |  8.4 / 79 | 8.5 / 81  | 14m away, outside tool shed, through 2 wooden walls  |
|  081a  |  30.3180 |  9.4 / 73 | 9.4 / 74  | 5m away, adjacent room through 2 wooden walls  |
|  0642  |  30.3180 | ? / ?  | ? / ?  |  7m, enclosed outside supply duct |
|  04c9  |  30.3181 |  11.7 / --- |  12.1 / --- | 2m, same room  |

04c9 has no humidity sensor, only an external temp sensor.

The Dostmann base station has reliable reception of all sensors except 0642, that is located in a very unsuitable place and is planned to be replaced with 04c9.

Tests with 'tfrec' had serious trouble picking up the sensors which came as a surprise, as in another setup it produced quite good results . As 04c9 was so close, reception was only possible with -g 20. 081a could only be received in auto gain mode. I still wonder why there was no reception of 0cde at all. 0706 should technically be not that hard to receive, but there were no signs of it whatsoever.

Terminal output for measurements was
```
> ./tfrec -D -g 25
Registering demod for TFA_1 KlimaLoggPro
Registering demod for TFA_2 sensors, 17240 bit/s
type 0x1: Samples per bit: 22.3
Registering demod for TFA_3 sensors, 9600 bit/s
type 0x2: Samples per bit: 40.0
Found Rafael Micro R820T tuner
GAIN 25.4
Frequency 868.2500MHz
Samplerate 1536000
START READ THREAD
#000 1582457458  2d d4 04 c9 85 18 6a 60 00 56 8d           TFA1 ID 04c9 +11.8 0% seq 0 lowbat 0 RSSI 77
#001 1582457468  2d d4 04 c9 85 18 6a 60 00 56 8d           TFA1 ID 04c9 +11.8 0% seq 0 lowbat 0 RSSI 77
#002 1582457479  2d d4 04 c9 85 18 6a 60 00 56 8d           TFA1 ID 04c9 +11.8 0% seq 0 lowbat 0 RSSI 77
#003 1582457489  2d d4 04 c9 85 18 6a 60 00 56 8d           TFA1 ID 04c9 +11.8 0% seq 0 lowbat 0 RSSI 77
#004 1582457499  2d d4 04 c9 85 18 6a 60 00 56 8d           TFA1 ID 04c9 +11.8 0% seq 0 lowbat 0 RSSI 77
#005 1582457509  2d d4 04 c9 85 18 6a 60 00 56 00           TFA1 BAD 1 RSSI 77 (CRC 00 8d)

##############################################################

> ./tfrec -D 
Registering demod for TFA_1 KlimaLoggPro
Registering demod for TFA_2 sensors, 17240 bit/s
type 0x1: Samples per bit: 22.3
Registering demod for TFA_3 sensors, 9600 bit/s
type 0x2: Samples per bit: 40.0
Found Rafael Micro R820T tuner
AUTO GAIN
Frequency 868.2500MHz
Samplerate 1536000
START READ THREAD
#000 1582457612  2d d4 04 c9 05 33 a8 81 01 b0 f2           TFA1 BAD 1 RSSI 79 (CRC f2 7e)
#001 1582457615  2d d4 08 1a 04 14 49 60 00 56 fc           TFA1 BAD 2 RSSI 69 (CRC fc a3)
#002 1582457625  2d d4 08 1a 84 94 49 60 00 56 fc           TFA1 ID 081a +9.4 73% seq 0 lowbat 0 RSSI 68
#003 1582457636  2d d4 08 1a 84 94 49 40 00 56 fc           TFA1 BAD 3 RSSI 67 (CRC fc 85)
#004 1582457642  2d d4 08 22 8e c0 0c 6a 60 00 56           TFA1 BAD 4 RSSI 80 (CRC 56 73)
#005 1582457646  2d d4 08 1a 84 94 01 60 00 56 fc           TFA1 BAD 5 RSSI 67 (CRC fc 00)
#006 1582457657  2d d4 08 1a 84 94 49 60 00 56 fc           TFA1 ID 081a +9.4 73% seq 0 lowbat 0 RSSI 67

##############################################################

> rtl_433 -S unknown -g 40 -f 868250000
rtl_433 version 19.08-235-ge49f53c branch master at 202002161316 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/bigalex/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
[... 172 lines cut ...]
##############################################################

> rtl_433 -S unknown -g 10 -f 868250000
rtl_433 version 19.08-235-ge49f53c branch master at 202002161316 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/bigalex/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 122 out of 149 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to 12.500000 dB.
Tuned to 868.250MHz.
*** Saving signal to file g061_868.25M_1000k.cu8 (49265 samples, 131072 bytes)
*** Saving signal to file g062_868.25M_1000k.cu8 (49266 samples, 131072 bytes)
*** Saving signal to file g063_868.25M_1000k.cu8 (43597 samples, 131072 bytes)
*** Saving signal to file g064_868.25M_1000k.cu8 (49264 samples, 131072 bytes)
*** Saving signal to file g065_868.25M_1000k.cu8 (49266 samples, 131072 bytes)
*** Saving signal to file g066_868.25M_1000k.cu8 (49317 samples, 131072 bytes)
*** Saving signal to file g067_868.25M_1000k.cu8 (49319 samples, 131072 bytes)
^CSignal caught, exiting!
```

merbanan, 2020-02-25:
Can you redo it with -s 2560k? and also some samples with -s 1536k?

alxlo, 2020-02-25:
I will not be in this place for some time. What I can do is take as many samples as needed here at my flat with only 2 sensors and a much noisier environment with some other sensors around. Will this help? What are useful gain values (3dB steps might possibly make more sense)?

merbanan, 2020-02-25:
If you look at the samples we just don't want them to clip. To look at the samples use sigrok or audacity. And 2-3 samples will be enough but it would be nice with a high samplerate.

pjenglund, 2020-02-26:
Is this capture something useful? Not quite sure how to verify it Audacity and sigrok but the latter showed some binary signal...

[g002_868.25M_2560k.cu8.zip](https://github.com/merbanan/rtl_433/files/4255368/g002_868.25M_2560k.cu8.zip)

> rtl_433 -S unknown -g 30 -f 868250000 -s 2560k
rtl_433 version 20.02-1-gf5e5a6a branch master at 202002181022 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 122 out of 149 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sample rate set to 2560000 S/s.
Tuner gain set to 32.800000 dB.
Tuned to 868.250MHz.
*** Saving signal to file g001_868.25M_2560k.cu8 (115872 samples, 262144 bytes)
*** Saving signal to file g002_868.25M_2560k.cu8 (76475 samples, 262144 bytes)
*** Saving signal to file g003_868.25M_2560k.cu8 (76468 samples, 262144 bytes)
*** Saving signal to file g004_868.25M_2560k.cu8 (78354 samples, 262144 bytes)
*** Saving signal to file g005_868.25M_2560k.cu8 (78355 samples, 262144 bytes)
*** Saving signal to file g006_868.25M_2560k.cu8 (76472 samples, 262144 bytes)
*** Saving signal to file g007_868.25M_2560k.cu8 (76475 samples, 262144 bytes)
*** Saving signal to file g008_868.25M_2560k.cu8 (76467 samples, 262144 bytes)
*** Saving signal to file g009_868.25M_2560k.cu8 (95421 samples, 262144 bytes)
*** Saving signal to file g010_868.25M_2560k.cu8 (72485 samples, 262144 bytes)
*** Saving signal to file g011_868.25M_2560k.cu8 (76473 samples, 262144 bytes)

zuckschwerdt, 2020-02-26:
Great! The gain is good and the (time)resolution is enough to make out the actual sine of the carrier. It's really looks like the "second" frequency is a harmonic only.

merbanan, 2020-02-26:
@zuckschwerdt what every it is, when I run the code on this sample it actually works because of the sample rate. So I'm guessing this will work with the current code if you use 2Ms/s as sample rate. As such I am inclined to add the decoder disabled with the note that you need a high sample rate for it to work.

Later on when some one reworks the pulse recovery flow so it will be possible for several FSK algorithms we could also add other algorithms and the Klimalogg_demod.pdf is basically the output of unfiltered I^2+Q^2. Then it would be possible to run at 1Ms/s, it almost works right now.

Is that an acceptable solution?

[12 more replies]

closed 2020-02-29 by merbanan