merbanan/rtl_433 #970: Davis Instruments Support

AyrTech, 2019-02-01:
Just wondering if anyone has had luck getting a Davis Vantage Vue working with rtl_433?

zuckschwerdt, 2019-02-18:
Do you have one? Can you supply samples?

zuckschwerdt, 2019-03-20:
Reopen if you try the weather station out and can supply sample data.

closed 2019-03-20 by zuckschwerdt

mc-knife, 2022-01-09:
i am trying out a davis vantage vue iss. i am able to see the data pings of all fife channels in gqrx (see image below). 
how do i go on from there to supply sample data?

reverse ingeneering of the signals is already done for this [rfm69 project](https://github.com/dekay/DavisRFM69).
there is even rtldavis, a [davis rtl-sdr receiver](https://github.com/lheijst/rtldavis) and [discussion](https://groups.google.com/g/weewx-development/c/jhGleLxu-p0) around its development.

however, i dont manage to receive the signals of my station with rtldavis. that is why i am trying rtl_433. further i am sorry that i have no clue how to transfer the findings of these existing projects into a driver for rtl_433.

[image]
?

zuckschwerdt, 2022-01-10:
Doesn't look too complicated. The hopping is new, but not a concern as rtl_433 does not track channels.
Can you supply a recording of (either `-w myfile_868.3M_1000k.cu8 -T 60` or files from `-S unknown`) of such a stream of transmissions?

mc-knife, 2022-01-10:
hey, thanks for your quick answer!
here are the recordings (i hope i did it right with -f flag):
roughly 256 files/10min of `rtl_433 -f 868300000 -S unknown` packed in [vue-S-unknown.tar.gz](https://filehorst.de/d/edanImdu)
and [vue_868.3M_1000k.cu8](https://filehorst.de/d/edJGyCtd) (118mb) as result of:
```
*:~$ rtl_433 -w vue_868.3M_1000k.cu8 -T 60 -f 868300000
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/mois/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
New defaults active, use "-Y classic -s 250k" for the old defaults!
Registered 122 out of 149 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
Detached kernel driver
Found Fitipower FC0013 tuner
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
Time expired, exiting!
Reattached kernel driver
*:~$
```

zuckschwerdt, 2022-01-10:
The files are great and it somewhat works with `rtl_433 -f 868.3M -X 'n=name,m=FSK_PCM,s=50,l=50,r=1024' FILE.cu8`
But the FSK deviation is tight and a good low pass filter for the resulting PCM would be needed. Might be a good time to implement that...

[image]

mc-knife, 2022-01-10:
hmm. i am realizing that i know nothing about radio signal data transfer. had to work my way through your reply with the search engine. and that is just the basic concepts you mentioning. i'd like to help though so if there is something I can contribute with my set of linux, rtl-sdr-dongle and vue sensor station, let me know.