merbanan/rtl_433 #971: pulse_FSK_detect(): Maximum number of pulses reached!

tgtcd, 2019-02-01:
I am trying to map  the WA5WH32B with rtl_433 to add new device support.  When run:

rtl_433 -G -f 914980000

I see good data from my WH65B and WH31E without issue, but I do not see the WH32B.  When I attempt to restart the WS32B by removing the battery and replacing, I receive:

pulse_FSK_detect(): Maximum number of pulses reached!

I have done this several times and the error is consistently reproducible. I also tried removing and replacing the battery on my WH31E and received the same error.  

I am not sure if the devices just send some junk on the first packet, or if the code is misreading what it is getting.

I also do not see the WH32B in the trace, but I see all other data.

Help on the error trapping would be great.

{"time" : "2019-02-01 16:37:05", "model" : "Fine Offset WH65B", "id" : 41, "temperature_C" : 9.400, "humidity" : 44, "wind_dir_deg" : 119, "wind_speed_ms" : 1.084, "gust_speed_ms" : 1.530, "rainfall_mm" : 134.112, "uv" : 231, "uvi" : 0, "light_lux" : 16585.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2019-02-01 16:37:21", "model" : "Fine Offset WH65B", "id" : 41, "temperature_C" : 9.400, "humidity" : 46, "wind_dir_deg" : 128, "wind_speed_ms" : 1.084, "gust_speed_ms" : 1.530, "rainfall_mm" : 134.112, "uv" : 209, "uvi" : 0, "light_lux" : 15980.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2019-02-01 16:37:31", "model" : "AmbientWeather-WH31E", "id" : 226, "channel" : 2, "battery" : "OK", "temperature_C" : 19.000, "humidity" : 29, "data" : "0300000000", "mic" : "CRC"}
{"time" : "2019-02-01 16:37:31", "model" : "AmbientWeather-WH31E", "id" : 226, "channel" : 2, "battery" : "OK", "temperature_C" : 19.000, "humidity" : 29, "data" : "0300000000", "mic" : "CRC"}
{"time" : "2019-02-01 16:37:37", "model" : "Fine Offset WH65B", "id" : 41, "temperature_C" : 9.400, "humidity" : 45, "wind_dir_deg" : 125, "wind_speed_ms" : 1.402, "gust_speed_ms" : 1.530, "rainfall_mm" : 134.112, "uv" : 192, "uvi" : 0, "light_lux" : 15500.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2019-02-01 16:37:50", "model" : "AmbientWeather-WH31E", "id" : 219, "channel" : 1, "battery" : "OK", "temperature_C" : 7.900, "humidity" : 51, "data" : "1300000000", "mic" : "CRC"}
{"time" : "2019-02-01 16:37:50", "model" : "AmbientWeather-WH31E", "id" : 219, "channel" : 1, "battery" : "OK", "temperature_C" : 7.900, "humidity" : 51, "data" : "1300000000", "mic" : "CRC"}
pulse_FSK_detect(): Maximum number of pulses reached!

{"time" : "2019-02-01 16:38:09", "model" : "Fine Offset WH65B", "id" : 41, "temperature_C" : 9.300, "humidity" : 45, "wind_dir_deg" : 119, "wind_speed_ms" : 0.255, "gust_speed_ms" : 0.510, "rainfall_mm" : 134.112, "uv" : 178, "uvi" : 0, "light_lux" : 14942.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2019-02-01 16:38:25", "model" : "Fine Offset WH65B", "id" : 41, "temperature_C" : 9.300, "humidity" : 46, "wind_dir_deg" : 119, "wind_speed_ms" : 0.701, "gust_speed_ms" : 1.020, "rainfall_mm" : 134.112, "uv" : 174, "uvi" : 0, "light_lux" : 14880.000, "battery" : "OK", "mic" : "CRC"}

tgtcd, 2019-02-01:
For reference - using a NooElec NESDR Mini USB RTL-SDR & ADS-B Receiver Set, RTL2832U & R820T Tuner, MCX Input on Raspberry Pi running current version of CentOS.

zuckschwerdt, 2019-02-03:
The overlong message on reset could be a sync pattern. Try to determine the frequency of the new sensor first. Then record some data.
Use CubicSDR or Gqrx and note the frequencies. Or try rtl_433 with a wider sample rate and meta data output (`-s 1024k -M level -A`) and note the frequencies.
Try a center frequency with a little offset, `-f 915M` should work well if the sensors is on 914.98M.

phinnay, 2019-02-16:
I am also attempting to get the WH32B to work. I see good data from both my WH65B and the WH31E and have saved a few cu8 files that could possibly be from the WH32B? How do you know what recording is useful and what is just noise? Attached is a zip with all the recordings that were not the 65 or 31 with a text file showing the console output
[signal.zip](https://github.com/merbanan/rtl_433/files/2871367/signal.zip)

I've also recorded whatever is coming out of the WH32B when you remove / replace the batteries via CubicSDR and have attached it here - it sounded clearest with FM modulation
[WH32B_wav.zip](https://github.com/merbanan/rtl_433/files/2871354/WH32B_wav.zip)

I'd love to get this device supported! if there is anything else I can do to assist let me know.

zuckschwerdt, 2019-02-16:
> How do you know what recording is useful and what is just noise?

Drop the sample file in triq.net/iqs and verify that it looks "proper". Broadly speaking there should be one "streak" of data.

Best advice is to remove the antenna from the receiver and place the device under test about 10 cm from the receiver. That way you only get that single device.

zuckschwerdt, 2019-02-16:
From your signal zip: 4,7,12,15 look like proper FSK. 13, 21 are too weak.
The others are strong (and harshly switched) FSK signals that use an exceedingly wide band by aliasing heavily. They show no data, just a regular pulsing for 3 seconds. If that pattern appears too frequently I would locate that device and remove it. It acts like a jammer.

Data from your good samples can be read with
```
rtl_433 -R 0 -X 'n=new,m=FSK_PCM,s=58,l=58,r=290' FILENAME.cu8
```
Looks like there is a sync word of `2dd4`, try:
```
rtl_433 -R 0 -X 'n=new,m=FSK_PCM,s=58,l=58,r=290,preamble=2dd4' FILENAME.cu8
```
Now write down the codes you see (e.g. `{60}e9f27b2a273be20`) along with the expected sensors readings. A table of 10-20 entries should suffice to start locating the bits.

phinnay, 2019-02-16:
interesting, so that 3 second screech at battery insertion is just noise? What would be the reason for that? 

I disconnected the antenna from my RTLSDR and placed the WH32B in the enclosure with it:
[image]
[image]

The unit is in a basement and cannot hear any other devices without the antenna from there, I recorded new cu8 files with the following command:
```rtl_433 -f 915000000 -S all```

then decoded them as per your instruction:
```rtl_433 -R 0 -X 'n=new,m=FSK_PCM,s=58,l=58,r=290,preamble=2dd4' FILENAME.cu8```
attached below is a ZIP with the recording and a text file with the output. 
[signal2.zip](https://github.com/merbanan/rtl_433/files/2872188/signal2.zip)

resulting code output:
```
codes     : {61}e392552727799100
codes     : {61}e392552727799100
codes     : {60}e39259262777920
codes     : {60}e39259262777920
codes     : {59}e3925c252777940
codes     : {59}e3925c252777940
codes     : {61}e3925f2527799900
codes     : {61}e3925f2527799900
codes     : {58}e39261242777980
codes     : {58}e39261242777980
codes     : {59}e392642427789c0
codes     : {59}e392642427789c0
codes     : {61}e392662327769b00
codes     : {61}e392662327769b00
codes     : {61}e392672327779d00
codes     : {61}e392672327779d00
codes     : {56}e3926a232777a0
codes     : {56}e3926a232777a0
codes     : {61}e3926f222776a300
codes     : {61}e3926f222776a300
codes     : {25}e392700
codes     : {25}e392700
codes     : {25}e392700
codes     : {25}e392700
codes     : {59}e39271212776a40
codes     : {59}e39271212776a40

```
The sensor remained in the box in the basement for the whole time, why the variation in code length?  I'd think that would remain the same length of bits. You can see the values on the LCD screen in the pics.

zuckschwerdt, 2019-02-16:
Great! I'll look into decoding this. The trailing bits (mostly 0) are just an artefact of the FSK demodulation.

zuckschwerdt, 2019-02-16:
It could reasonably be something like [`ID?8d FLAGS?4h TEMP?12d HUM:8d ?16h CHK:8h`](http://triq.net/bitbench?c=e3925527277991&c=e3925926277792&c=e3925c25277794&c=e3925f25277999&c=e3926124277798&c=e392642427789c&c=e392662327769b&c=e392672327779d&c=e3926a232777a0&c=e3926f222776a3&c=e39271212776a4%20%5B22.5C%20%3F%2033%25%20%3F%5D&f=ID%3F8d%20FLAGS%3F4h%20TEMP%3F12d%20HUM%3A8d%20%3F16h%20CHK%3A8h)
(where temperature is °C offset 40 scaled 10 and the checksum is plain addition with carry).
Use that bitbench to decode data with exactly known value, best case very high and low values.

phinnay, 2019-02-16:
this is fun :D

I captured six packets at known temperatures, it looks like you got humidity figured out, but temp and pressure don't line up. 
```
69.1 F / 32% RH / 29.80 inmg
e3925e20

69.3 F / 33% RH / 29.80 inmg
e3925f21276c880

69.1 F / 33% RH / 29.81 inmg
e3925e21276e8900

e3925e21276d880
69.1 F / 33% RH / 29.80 inmg

e3925f21276d8900
69.3 F / 33% RH / 29.80 inmg

e3925f21276c880
69.3 F / 33% RH / 29.80 inmg
```

zuckschwerdt, 2019-02-16:
Very good! If you convert the F to C, scale and offset you get the "raw" values:
- 69.1°F = 20.6°C (+40 x10 = 606)
- 69.3°F = 20.7°C (+40 x10 = 607)

Same for the Pressure from inHg to hPa:
- 29.80 * 33.864 = 1009.14
- 29.81 * 33.864 = 1009.48

See in this [BitBench](http://triq.net/bitbench?c=e3925f21276c88%20%5B69.3%20F%20%2F%2033%25%20RH%20%2F%2029.80%20inmg%5D&c=e3925e21276e89%20%5B69.1%20F%20%2F%2033%25%20RH%20%2F%2029.81%20inmg%5D&c=e3925e21276d88%20%5B69.1%20F%20%2F%2033%25%20RH%20%2F%2029.80%20inmg%5D&c=e3925f21276d89%20%5B69.3%20F%20%2F%2033%25%20RH%20%2F%2029.80%20inmg%5D&c=e3925f21276c88%20%5B69.3%20F%20%2F%2033%25%20RH%20%2F%2029.80%20inmg%5D&f=ID%3F8h%20FLAGS%3F4h%20TEMP_C%3A12d%20HUM%3A8d%20HPA%3A16d%20CHK%3A8h).

If you now take the battery out a few times and note where the bits change we can identify the ID position and then we are done. (Bonus: use a known flat battery or variable power supply to find the battery_low flag.)

phinnay, 2019-02-16:
Ahh, i forgot to offset that's why my numbers didn't work. 

I hooked the WH32 up to a power supply and dropped the voltage until the LCD screen was dim. It did not like that. The LCD showed dashes for temp and humidity however I still picked up packets that coincided with the transmit indicator on the screen.

low battery  (2v)
```
-- F / -- RH / 29.96 inmg
e0ffffff27a0a40

-- F / -- RH / 29.96 inmg
e0

-- F / -- RH / 29.96 inmg
e0

-- F / -- RH / 29.96 inmg
e0ffffff27a2a60

I then put good batteries back in and captured 5 more packets, didn't record the climate info since that's already figured out

```
efa2482727759c0
efa2472727769c0
efa2462727759a0
efa2462727759a0
efa245272775990
```

zuckschwerdt, 2019-02-17:
Updated [BitBench](http://triq.net/bitbench?c=e3925f21276c88%20%5B69.3%20F%20%2F%2033%25%20RH%20%2F%2029.80%20inmg%5D&c=e3925e21276e89%20%5B69.1%20F%20%2F%2033%25%20RH%20%2F%2029.81%20inmg%5D&c=e3925e21276d88%20%5B69.1%20F%20%2F%2033%25%20RH%20%2F%2029.80%20inmg%5D&c=e3925f21276d89%20%5B69.3%20F%20%2F%2033%25%20RH%20%2F%2029.80%20inmg%5D&c=e3925f21276c88%20%5B69.3%20F%20%2F%2033%25%20RH%20%2F%2029.80%20inmg%5D&c=e0ffffff27a0a4%20%5B--%20F%20%2F%20--%20RH%20%2F%2029.96%20inmg%5D&c=e0ffffff27a2a6%20%5B--%20F%20%2F%20--%20RH%20%2F%2029.96%20inmg%5D&c=efa2482727759c&c=efa2472727769c&c=efa2462727759a&c=efa2462727759a&c=efa24527277599&f=TYPE%3A4h%20ID%3A8d%20FLAGS%3F%20TEMP_C%3A12d%20HUM%3A8d%20HPA%3A16d%20CHK%3A8h).
The first 4 bit seem fixed to `0xe`. The next 8 bits could be ID.
The battery_low flag then could be the top bit of temperature, but hard to say as the temperature reading failed (the voltage was too low).

What is the claimed range for temperature? The values can represent
- -40.0 to
- 62.3 (at 10 bit)
- 164.7 (at 11bit)
- 369.5 (at 12 bit).

I'd say 10 bit is too close. But 11 bit will work well. So the top bit could be battery_low.

If you want to make this bullet proof:
- Cycle the battery 10 or more times and record data each time (no need for the actual reading). This will tell if the first 4 bit are really fixed and if the next 8 bits all change.
- Drop the voltage low but just high enough that there the reading still works. This will have a proper temp value and show the battery_low bit.

phinnay, 2019-02-17:
I'm away on business this week, if someone else can capture the packets needed that'd be awesome! If not we will have to pick this up next week.

zuckschwerdt, 2019-02-19:
@phinnay since your build already looks great here some quick hints to make it really shine:

Never plug the receiver directly into a board. At least the Raspi (and I guess all unshielded SoCs) have nasty emissions. Even a short USB extension cable will make a difference (looks like you got room to space the receiver with a 10cm cable.

The cheap RTLs have no heat sinking. Fine for 250k but at 2048k heat will be a problem (90°C at the chip surface). At 3200k I have killed some of my unfortunate receivers.
Just clip the case open and use a self-sticking heat pad to put a tiny heat sink on the R820T(2) -- the chip near the antenna port.

phinnay, 2019-02-24:
Thanks for the tips on the RTL receiver, I'll make the adjustments!

Here is the device, power cycled 10 times and recorded:
[device_id.zip](https://github.com/merbanan/rtl_433/files/2898298/device_id.zip)
```
ed72742726ee0
e712622926e6900
e1d272280
ef12672a26e49c0
ee80
e280
efa2762526f4460
e7c26f2826e64c0
e6f2700
ec72722726e60
```

Here is 10 readings with low battery (2.5v)
[low_battery.zip](https://github.com/merbanan/rtl_433/files/2898306/low_battert.zip)

```
1	72.1F / 39% / 29.50InHg
eaca6e272700

2	71.6F / 39% / 29.50InHg
eaca6c27270a780

3	71.4F / 39% / 29.51InHg
eaca6b272709760

4	71.4F / 39% / 29.51InHg
eaca6b2727087500

5	71.2F / 39% / 29.52InHg
eaca6a27270b7700

6	71.1F / 40% / 29.51InHg
eaca69282708740

7	70.7F / 40% / 29.50InHg
eaca67280

8	70.9F / 49% / 29.50InHg
eaca680

9	70.5F / 40% / 29.50InHg
eaca66282700

10	70.5F / 40% / 29.50InHg
eaca66280
```

phinnay, 2019-02-26:
Teriffic! Looking forward to pulling the new version.

zuckschwerdt, 2019-02-26:
Will take a few days, sorry.

phinnay, 2019-02-26:
no problem, thank you so much for helping with this! rtl_433 is a great project.

steepleian, 2019-03-23:
I was just wondering if there is any further progress?

zuckschwerdt, 2019-03-23:
This will take more time to implement, the transmission is more intricate, sorry. Ping back in a week or two.

steepleian, 2019-03-23:
Ok many thanks for the update.

zuckschwerdt, 2019-04-07:
Support added with #1040. Reopen if you notice problems.

closed 2019-04-07 by zuckschwerdt

slvrscoobie, 2019-08-02:
I have a WH32B as well as 8 WH31. The problem im running into is the newer version issues on April 7 now sees my WH31 devices as WH32 devices but has bad data, and missing info. IE: the humidity is 171% on my one device, and it lost the 'channel' information which helped me understand which WH31 was which. 
Can you help decode? Im including output I grabbed including my WH65B and WH25 mixed in, please ignore :) Including some JSON and non JSON data. 
Registered 1 out of 133 device decoding protocols [ 78 ]
usb_claim_interface error -6
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 914.980MHz.
{"time" : "2019-08-02 16:35:52", "model" : "Fineoffset-WH32B", "id" : 190, "temperature_C" : 25.800, "humidity" : 44, "pressure_hPa" : 1010.700, "battery" : "OK", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:35:52", "model" : "Fineoffset-WH32B", "id" : 5, "temperature_C" : 35.200, "humidity" : 229, "pressure_hPa" : 1713.300, "battery" : "LOW", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:35:54", "model" : "Fineoffset-WH32B", "id" : 10, "temperature_C" : -17.400, "humidity" : 118, "pressure_hPa" : 1567.900, "battery" : "OK", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:36:00", "model" : "Fine Offset WH65B", "id" : 10, "temperature_C" : 29.800, "humidity" : 47, "wind_dir_deg" : 267, "wind_speed_ms" : 1.466, "gust_speed_ms" : 2.040, "rainfall_mm" : 1070.864, "uv" : 1493, "uvi" : 3, "light_lux" : 54901.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2019-08-02 16:36:01", "model" : "Fineoffset-WH32B", "id" : 2, "temperature_C" : 56.200, "humidity" : 154, "pressure_hPa" : 1363.400, "battery" : "LOW", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:36:02", "model" : "Fineoffset-WH32B", "id" : 13, "temperature_C" : 49.800, "humidity" : 180, "pressure_hPa" : 1364.200, "battery" : "OK", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:36:03", "model" : "Fineoffset-WH32B", "id" : 14, "temperature_C" : -23.800, "humidity" : 172, "pressure_hPa" : 1740.400, "battery" : "LOW", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:36:16", "model" : "Fine Offset WH65B", "id" : 10, "temperature_C" : 29.800, "humidity" : 50, "wind_dir_deg" : 299, "wind_speed_ms" : 1.530, "gust_speed_ms" : 2.040, "rainfall_mm" : 1070.864, "uv" : 1511, "uvi" : 3, "light_lux" : 55226.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2019-08-02 16:36:32", "model" : "Fine Offset WH65B", "id" : 10, "temperature_C" : 29.800, "humidity" : 49, "wind_dir_deg" : 307, "wind_speed_ms" : 0.382, "gust_speed_ms" : 1.020, "rainfall_mm" : 1070.864, "uv" : 1316, "uvi" : 3, "light_lux" : 49135.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2019-08-02 16:36:36", "model" : "Fineoffset-WH32B", "id" : 7, "temperature_C" : -25.400, "humidity" : 254, "pressure_hPa" : 944.800, "battery" : "OK", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:36:37", "model" : "Fineoffset-WH32B", "id" : 12, "temperature_C" : -19.200, "humidity" : 231, "pressure_hPa" : 1881.100, "battery" : "OK", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:36:39", "model" : "Fine Offset Electronics, WH25", "id" : 224, "temperature_C" : 23.500, "humidity" : 49, "pressure_hPa" : 1007.600, "battery" : "OK", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:36:48", "model" : "Fine Offset WH65B", "id" : 10, "temperature_C" : 29.800, "humidity" : 50, "wind_dir_deg" : 307, "wind_speed_ms" : 0.255, "gust_speed_ms" : 0.510, "rainfall_mm" : 1070.864, "uv" : 733, "uvi" : 1, "light_lux" : 31325.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2019-08-02 16:36:52", "model" : "Fineoffset-WH32B", "id" : 190, "temperature_C" : 25.800, "humidity" : 44, "pressure_hPa" : 1010.500, "battery" : "OK", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:36:53", "model" : "Fineoffset-WH32B", "id" : 2, "temperature_C" : -22.300, "humidity" : 148, "pressure_hPa" : 1032.700, "battery" : "LOW", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:37:00", "model" : "Fineoffset-WH32B", "id" : 5, "temperature_C" : 35.200, "humidity" : 229, "pressure_hPa" : 1713.300, "battery" : "LOW", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:37:01", "model" : "Fineoffset-WH32B", "id" : 10, "temperature_C" : -17.400, "humidity" : 117, "pressure_hPa" : 1655.000, "battery" : "OK", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:37:03", "model" : "Fineoffset-WH32B", "id" : 13, "temperature_C" : 49.800, "humidity" : 180, "pressure_hPa" : 1364.200, "battery" : "OK", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:37:04", "model" : "Fine Offset WH65B", "id" : 10, "temperature_C" : 29.800, "humidity" : 50, "wind_dir_deg" : 309, "wind_speed_ms" : 0.191, "gust_speed_ms" : 0.510, "rainfall_mm" : 1070.864, "uv" : 522, "uvi" : 1, "light_lux" : 25280.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2019-08-02 16:37:06", "model" : "Fineoffset-WH32B", "id" : 2, "temperature_C" : 56.200, "humidity" : 154, "pressure_hPa" : 1363.400, "battery" : "LOW", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:37:06", "model" : "Fineoffset-WH32B", "id" : 14, "temperature_C" : -23.800, "humidity" : 172, "pressure_hPa" : 1751.500, "battery" : "LOW", "mic" : "CHECKSUM"}
{"time" : "2019-08-02 16:37:20", "model" : "Fine Offset WH65B", "id" : 10, "temperature_C" : 29.800, "humidity" : 51, "wind_dir_deg" : 314, "wind_speed_ms" : 0.064, "gust_speed_ms" : 0.510, "rainfall_mm" : 1070.864, "uv" : 496, "uvi" : 1, "light_lux" : 24505.000, "battery" : "OK", "mic" : "CRC"}
ç^CSignal caught, exiting!
[... cut ...]
Channel   : 3            Battery   : OK            Temperature: 28.4 C       Humidity  : 68 %          Extra Data: 1500000000    Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-02 16:39:12
model     : AmbientWeather-WH31E                   id        : 232
Channel   : 3            Battery   : OK            Temperature: 28.4 C       Humidity  : 68 %          Extra Data: 1500000000    Integrity : CRC

zuckschwerdt, 2019-08-06:
The protocols are somewhat similar. The AmbientWeather-WH31E has a stronger checksum which will reject almost all bad data. The Fineoffset-WH32B might get false positives, but it should only (wrongly) decode the AmbientWeather-WH31E packets occasionally.
I'll have a look at the sample data you supplied with https://github.com/merbanan/rtl_433_tests/issues/296

zuckschwerdt, 2019-08-06:
Should be fixed with 78b7449