merbanan/rtl_433 #972: Acurite 899 Rain Gauge Fails

K2AAE, 2019-02-01:
Just tried testing a new Acurite 0899 Rain Gauage, fails with: Acurite 5n1 sensor 0X0426 Ch C, Status 0, Unknown message type 0x30
This message repeats three times.

I can add -R 10 and only run 0896 which still detects nothing.
I ran every Acurite device and only 40 responds, my 5n1 and a 'tower' device with the 0899 rain gauge above message mixed in.

rct, 2019-02-01:
This sounds like the 899 rain gauge is a new/different device than the older 896. 

The error message is essentially saying it's sending the same type of message as the 5-n-1, 592TXR Tower, 6045 lightning, etc.   but the message type 0x30 is one we haven't seen yet.  So it's a new device that support is needed for.

The normal thing to do is submit some sample messages to the rtl_433_tests repo.

Do you have some sort of display for that rain gauge?    It would help to be able to figure out the encoding.

K2AAE, 2019-02-03:
I purchased the device while I was waiting on my 5n1 to arrive, after I played with it I realized the rain gauge wouldn't work with the 5n1, nor would any extra temperature probes hence the reason I'm here. I cloned the git but when I try to make it:
 sudo make test
python bin/run_test.py -I time --first-line
Traceback (most recent call last):
  File "bin/run_test.py", line 13, in <module>
    from deepdiff import DeepDiff
ImportError: No module named deepdiff
makefile:5: recipe for target 'test' failed
make: *** [test] Error 1
I tried to run the rtl_433_test.sh I get:
/home/pi/Projects/rtl_433_tests’: No such file or directory
I cloned rtl_433 at /home/pi which created /home/pi/rtl_433 and now /home/pi/rtl_433/rtl_433_tests

zuckschwerdt, 2019-02-03:
For reference this was also mentioned in #799

zuckschwerdt, 2019-02-03:
Run rtl_433 with `-S all`. Check the recorded samples with e.g. `rtl_433 g001.cu8, then post them here or better PR to rtl_433_tests, along with a short text what value you expect to see (temperature, humidity, …). Thanks!

tonious, 2019-05-26:
So I do not have a rain gauge of any kind, but `rtl_433` compatibility is one of my requirements, so I dug into this a little and found a [post from 2014](https://rayshobby.net/wordpress/reverse-engineer-wireless-temperature-humidity-rain-sensors-part-3/) that describes message type 0x30.

It looks like bytes 3-7 are a straight count of tipping bucket... tips, I guess.  7 significant bits per byte, with the high bit out of each byte acting as a parity bit.  Most significant byte first.

Each tip should correspond to 1mm of rainfall.

walthowd, 2019-06-12:
Not sure if this is the right track, but it looks like someone figured it out here:

https://groups.google.com/forum/#!topic/rtl_433/7ImutJT-dKY

I was going to buy a 899, but switched to a different model.

walthowd, 2019-06-18:
Ehh, bought one anyway -- Trying to figure out how to add support. 

Here is the signal with a few bucket tips back and forth:
https://www.dropbox.com/s/0e7d9u664afg8zf/g001_433.92M_250k.cu8?dl=0

And then here is the signal with the next reading of a single bucket tip:
https://www.dropbox.com/s/b35i176argdvaiy/g002_433.92M_250k.cu8?dl=0

And finally the last signal save with two more bucket tips:
https://www.dropbox.com/s/ep5xbma8nh4yhu0/g003_433.92M_250k.cu8?dl=0

zuckschwerdt, 2019-06-18:
I'll have a look. Will take a few days.

walthowd, 2019-06-28:
I took a stab at adding support here, but my C is pretty rusty -- so far I don' think I'm matching the message type so this new code is not being evaluated:

https://github.com/walthowd/rtl_433/commit/cae3ccfcf0001453e8808e03c4f78db21113ee25

zuckschwerdt, 2019-08-06:
@walthowd did you get any results with that modification? The samples you provided have different timing (PWM 500/1000) than the Acurite TXR (PWM 200/400) this code hooks into.

zuckschwerdt, 2019-08-06:
There is a test branch in `feat-acurite899` now. Please collect some bitbuffer outputs for different e.g. channels and id (replace battery to reset).

walthowd, 2019-08-06:
@zuckschwerdt No, I didn't get any results from that modification and after I started digging into the signalling, I saw my differences were pretty large. Perhaps Acurite changed the hardware at some point in the last two years?

Thanks for looking into this! I'll pull the the test branch and collect some data.

walthowd, 2019-08-07:
@zuckschwerdt So I'm an idiot. I bought what was listed as a 899 on eBay but they actually sent me a Acurite 875 (00875TX). 

I didn't realize it until I wasn't able to get the channel and IDs to change and was looking at the 899 manual and realized I didn't have the jumper/switch to change channels. Then I looked right there on the bottom of the device and it has 00875TX printed on it.

So sorry for muddying the waters. I tested the branch tonight and here is the output:
https://pastebin.com/TQCjqAvb

I wasn't able to get the channels or ID to change with resetting the batteries.

zuckschwerdt, 2019-08-07:
:) No worries.
From your data:
```
{48} 06 9a 18 00 6a 0e : 00000110 10011010 00011000 00000000 01101010 00001110
{48} 06 9a 18 00 6b 0f : 00000110 10011010 00011000 00000000 01101011 00001111
--
{48} 06 95 b8 00 00 db : 00000110 10010101 10111000 00000000 00000000 11011011
{48} 06 95 b8 00 02 d9 : 00000110 10010101 10111000 00000000 00000010 11011001
{48} 06 95 b8 00 03 d8 : 00000110 10010101 10111000 00000000 00000011 11011000
--
{48} 06 9f c8 00 01 40 : 00000110 10011111 11001000 00000000 00000001 01000000
{48} 06 9f c8 00 09 48 : 00000110 10011111 11001000 00000000 00001001 01001000
{48} 06 9f c8 00 0e 4f : 00000110 10011111 11001000 00000000 00001110 01001111
{48} 06 9f c8 00 0f 4e : 00000110 10011111 11001000 00000000 00001111 01001110
```
Should help to figure out the checksum. Some devices indeed buffer the ID even without power. It's very rare though. The channel setting should be something like a DIP switch?
What other readings does this sensor have? Maybe the `a1`, `5b`, `fc` above are the ID? Or `9a`, `95`, `9f` could be something too, not an ID though.

walthowd, 2019-08-07:
No dip switches present that I can see.

Here is some more output:

1) Right after pulling batteries
2) Three bucket tips, batteries still inserted
3) Pulled batteries again
4) Two bucket tips, batteries still inserted
5) Pulled batteries again
6) One bucket tip, batteries still inserted
```

acurite_rain_899_decode: (9c 156) bitbuffer:: Number of rows: 7 
[00] {48} 06 97 b8 00 00 b9 : 00000110 10010111 10111000 00000000 00000000 10111001 
[01] {48} 06 97 b8 00 00 b9 : 00000110 10010111 10111000 00000000 00000000 10111001 
[02] {48} 06 97 b8 00 00 b9 : 00000110 10010111 10111000 00000000 00000000 10111001 
[03] {48} 06 97 b8 00 00 b9 : 00000110 10010111 10111000 00000000 00000000 10111001 
[04] {48} 06 97 b8 00 00 b9 : 00000110 10010111 10111000 00000000 00000000 10111001 
[05] {48} 06 97 b8 00 00 b9 : 00000110 10010111 10111000 00000000 00000000 10111001 
[06] { 0}                   : 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-07 13:04:55
model     : Acurite-Rain899                        id        : 6
channel   : 0            Battery   : 1             Rain count: 0

acurite_rain_899_decode: (9e 158) bitbuffer:: Number of rows: 7 
[00] {48} 06 97 b8 00 03 ba : 00000110 10010111 10111000 00000000 00000011 10111010 
[01] {48} 06 97 b8 00 03 ba : 00000110 10010111 10111000 00000000 00000011 10111010 
[02] {48} 06 97 b8 00 03 ba : 00000110 10010111 10111000 00000000 00000011 10111010 
[03] {48} 06 97 b8 00 03 ba : 00000110 10010111 10111000 00000000 00000011 10111010 
[04] {48} 06 97 b8 00 03 ba : 00000110 10010111 10111000 00000000 00000011 10111010 
[05] {48} 06 97 b8 00 03 ba : 00000110 10010111 10111000 00000000 00000011 10111010 
[06] { 0}                   : 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-07 13:05:55
model     : Acurite-Rain899                        id        : 6
channel   : 0            Battery   : 1             Rain count: 3

acurite_rain_899_decode: (130 304) bitbuffer:: Number of rows: 7 
[00] {48} 06 91 e8 00 00 4f : 00000110 10010001 11101000 00000000 00000000 01001111 
[01] {48} 06 91 e8 00 00 4f : 00000110 10010001 11101000 00000000 00000000 01001111 
[02] {48} 06 91 e8 00 00 4f : 00000110 10010001 11101000 00000000 00000000 01001111 
[03] {48} 06 91 e8 00 00 4f : 00000110 10010001 11101000 00000000 00000000 01001111 
[04] {48} 06 91 e8 00 00 4f : 00000110 10010001 11101000 00000000 00000000 01001111 
[05] {48} 06 91 e8 00 00 4f : 00000110 10010001 11101000 00000000 00000000 01001111 
[06] { 0}                   : 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-07 13:06:21
model     : Acurite-Rain899                        id        : 6
channel   : 0            Battery   : 1             Rain count: 0
acurite_rain_899_decode: (134 308) bitbuffer:: Number of rows: 7 
[00] {48} 06 91 e8 00 02 4d : 00000110 10010001 11101000 00000000 00000010 01001101 
[01] {48} 06 91 e8 00 02 4d : 00000110 10010001 11101000 00000000 00000010 01001101 
[02] {48} 06 91 e8 00 02 4d : 00000110 10010001 11101000 00000000 00000010 01001101 
[03] {48} 06 91 e8 00 02 4d : 00000110 10010001 11101000 00000000 00000010 01001101 
[04] {48} 06 91 e8 00 02 4d : 00000110 10010001 11101000 00000000 00000010 01001101 
[05] {48} 06 91 e8 00 02 4d : 00000110 10010001 11101000 00000000 00000010 01001101 
[06] { 0}                   : 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-07 13:07:21
model     : Acurite-Rain899                        id        : 6
channel   : 0            Battery   : 1             Rain count: 2

acurite_rain_899_decode: (f8 248) bitbuffer:: Number of rows: 7 
[00] {48} 06 95 f8 00 00 9b : 00000110 10010101 11111000 00000000 00000000 10011011 
[01] {48} 06 95 f8 00 00 9b : 00000110 10010101 11111000 00000000 00000000 10011011 
[02] {48} 06 95 f8 00 00 9b : 00000110 10010101 11111000 00000000 00000000 10011011 
[03] {48} 06 95 f8 00 00 9b : 00000110 10010101 11111000 00000000 00000000 10011011 
[04] {48} 06 95 f8 00 00 9b : 00000110 10010101 11111000 00000000 00000000 10011011 
[05] {48} 06 95 f8 00 00 9b : 00000110 10010101 11111000 00000000 00000000 10011011 
[06] { 0}                   : 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-07 13:07:45
model     : Acurite-Rain899                        id        : 6
channel   : 0            Battery   : 1             Rain count: 0
acurite_rain_899_decode: (fa 250) bitbuffer:: Number of rows: 7 
[00] {48} 06 95 f8 00 01 9a : 00000110 10010101 11111000 00000000 00000001 10011010 
[01] {48} 06 95 f8 00 01 9a : 00000110 10010101 11111000 00000000 00000001 10011010 
[02] {48} 06 95 f8 00 01 9a : 00000110 10010101 11111000 00000000 00000001 10011010 
[03] {48} 06 95 f8 00 01 9a : 00000110 10010101 11111000 00000000 00000001 10011010 
[04] {48} 06 95 f8 00 01 9a : 00000110 10010101 11111000 00000000 00000001 10011010 
[05] {48} 06 95 f8 00 01 9a : 00000110 10010101 11111000 00000000 00000001 10011010 
[06] { 0}                   : 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-07 13:08:45
model     : Acurite-Rain899                        id        : 6
channel   : 0            Battery   : 1             Rain count: 1
```

zuckschwerdt, 2019-08-07:
(One good code per transmission is enough, shortened your table for brevity.)
Looks like the `069` is constant, the next two nibbles are the ID, then maybe a nibble of flags?
The checksum is something like XOR, but it doesn't fit exactly. I'll update the code to reflect the current understanding of the data.

walthowd, 2019-08-07:
@zuckschwerdt Thank you! I also disassembled the unit to check for hidden dip switches but didn't find anything:

https://imgur.com/a/779lM5d

zuckschwerdt, 2019-08-07:
Ultra-cheap chip-on-board blob ;) The channel could reasonably be hardcoded. Notice the link at R3 and then R8, R9 unpopulated. Those could be settings. Looks like the trace goes form VCC through the link then the cap to GND though? J3 (jumper) is obviously some setting. And the contacts near "Reset" could be test points to trigger a reset. But don't bridge anything without checking the traces before.

zuckschwerdt, 2019-08-07:
If you want to try the (force push) updated branch: `git reset --hard HEAD^ ; git pull`

walthowd, 2019-08-08:
Got it up and running, seems to be working perfectly so far! Sorry again on the confusion about the 875 vs the 899.

zuckschwerdt, 2019-08-09:
We can leave the channel and battery flag aside, but we need to figure out the checksum before enabling this code -- there are false positives with other decoders otherwise. Maybe you can collect a long list of codes? Just redirect to a log, then grep for the code lines and condense with `uniq -c` so we know the codes are good.

walthowd, 2019-08-09:
Sure, I can do that -- when you are referring to the codes you mean the bitbuffer output?

Like this:
`{48} 06 9a 18 00 6b 0f : 00000110 10011010 00011000 00000000 01101011 00001111`

zuckschwerdt, 2019-08-09:
Yes!

walthowd, 2019-08-09:
Here you go. The line with 1440 entries was the unit sitting idle overnight. Then about three hours ago I started a slow drip into the unit to start incrementing values. 

Full raw file is here:
https://pastebin.com/eK3qqFhp

```
docker logs rtl_433 2>&1 | grep {  | sed -e 's/^.....//' | sort | uniq -c
    379 { 0}                   : 
      7 {11} 06 80             : 00000110 100
      1 { 3} 00                : 000
      1 {37} 2f c0 28 db a8    : 00101111 11000000 00101000 11011011 10101
      1 {37} 2f c0 29 aa d8    : 00101111 11000000 00101001 10101010 11011
      1 {37} 2f c0 2f ec 98    : 00101111 11000000 00101111 11101100 10011
      1 {37} 2f c0 31 40 a8    : 00101111 11000000 00110001 01000000 10101
      1 {37} 2f c0 32 13 f8    : 00101111 11000000 00110010 00010011 11111
      1 {37} 2f c0 32 93 78    : 00101111 11000000 00110010 10010011 01111
      1 {37} 2f c0 39 31 50    : 00101111 11000000 00111001 00110001 01010
      1 {45} 34 af c0 30 39 d0 : 00110100 10101111 11000000 00110000 00111001 11010
   1440 {48} 06 95 f8 04 05 5a : 00000110 10010101 11111000 00000100 00000101 01011010 
      6 {48} 06 95 f8 04 0d 52 : 00000110 10010101 11111000 00000100 00001101 01010010 
      6 {48} 06 95 f8 04 15 4a : 00000110 10010101 11111000 00000100 00010101 01001010 
      6 {48} 06 95 f8 04 1e 41 : 00000110 10010101 11111000 00000100 00011110 01000001 
      6 {48} 06 95 f8 04 26 79 : 00000110 10010101 11111000 00000100 00100110 01111001 
      6 {48} 06 95 f8 04 2e 71 : 00000110 10010101 11111000 00000100 00101110 01110001 
      6 {48} 06 95 f8 04 36 69 : 00000110 10010101 11111000 00000100 00110110 01101001 
      6 {48} 06 95 f8 04 3e 61 : 00000110 10010101 11111000 00000100 00111110 01100001 
      6 {48} 06 95 f8 04 45 1a : 00000110 10010101 11111000 00000100 01000101 00011010 
      6 {48} 06 95 f8 04 4d 12 : 00000110 10010101 11111000 00000100 01001101 00010010 
      6 {48} 06 95 f8 04 55 0a : 00000110 10010101 11111000 00000100 01010101 00001010 
      6 {48} 06 95 f8 04 5d 02 : 00000110 10010101 11111000 00000100 01011101 00000010 
      6 {48} 06 95 f8 04 64 3b : 00000110 10010101 11111000 00000100 01100100 00111011 
      6 {48} 06 95 f8 04 6b 34 : 00000110 10010101 11111000 00000100 01101011 00110100 
      6 {48} 06 95 f8 04 73 2c : 00000110 10010101 11111000 00000100 01110011 00101100 
      6 {48} 06 95 f8 04 7a 25 : 00000110 10010101 11111000 00000100 01111010 00100101 
      6 {48} 06 95 f8 04 82 dd : 00000110 10010101 11111000 00000100 10000010 11011101 
      6 {48} 06 95 f8 04 8a d5 : 00000110 10010101 11111000 00000100 10001010 11010101 
      6 {48} 06 95 f8 04 91 ce : 00000110 10010101 11111000 00000100 10010001 11001110 
      6 {48} 06 95 f8 04 98 c7 : 00000110 10010101 11111000 00000100 10011000 11000111 
      6 {48} 06 95 f8 04 9f c0 : 00000110 10010101 11111000 00000100 10011111 11000000 
      6 {48} 06 95 f8 04 a7 f8 : 00000110 10010101 11111000 00000100 10100111 11111000 
      6 {48} 06 95 f8 04 ae f1 : 00000110 10010101 11111000 00000100 10101110 11110001 
      6 {48} 06 95 f8 04 b5 ea : 00000110 10010101 11111000 00000100 10110101 11101010 
      6 {48} 06 95 f8 04 bc e3 : 00000110 10010101 11111000 00000100 10111100 11100011 
      6 {48} 06 95 f8 04 c3 9c : 00000110 10010101 11111000 00000100 11000011 10011100 
      6 {48} 06 95 f8 04 ca 95 : 00000110 10010101 11111000 00000100 11001010 10010101 
      6 {48} 06 95 f8 04 d1 8e : 00000110 10010101 11111000 00000100 11010001 10001110 
      6 {48} 06 95 f8 04 d8 87 : 00000110 10010101 11111000 00000100 11011000 10000111 
      6 {48} 06 95 f8 04 df 80 : 00000110 10010101 11111000 00000100 11011111 10000000 
      6 {48} 06 95 f8 04 e5 ba : 00000110 10010101 11111000 00000100 11100101 10111010 
      6 {48} 06 95 f8 04 ec b3 : 00000110 10010101 11111000 00000100 11101100 10110011 
      6 {48} 06 95 f8 04 f2 ad : 00000110 10010101 11111000 00000100 11110010 10101101 
      6 {48} 06 95 f8 04 f9 a6 : 00000110 10010101 11111000 00000100 11111001 10100110 
      6 {48} 06 95 f8 05 00 6e : 00000110 10010101 11111000 00000101 00000000 01101110 
      6 {48} 06 95 f8 05 07 69 : 00000110 10010101 11111000 00000101 00000111 01101001 
      6 {48} 06 95 f8 05 0d 63 : 00000110 10010101 11111000 00000101 00001101 01100011 
      6 {48} 06 95 f8 05 14 7a : 00000110 10010101 11111000 00000101 00010100 01111010 
      5 {48} 06 95 f8 05 1b 75 : 00000110 10010101 11111000 00000101 00011011 01110101 
      6 {48} 06 95 f8 05 22 4c : 00000110 10010101 11111000 00000101 00100010 01001100 
      6 {48} 06 95 f8 05 28 46 : 00000110 10010101 11111000 00000101 00101000 01000110 
      6 {48} 06 95 f8 05 2e 40 : 00000110 10010101 11111000 00000101 00101110 01000000 
      5 {48} 06 95 f8 05 35 5b : 00000110 10010101 11111000 00000101 00110101 01011011 
      6 {48} 06 95 f8 05 3b 55 : 00000110 10010101 11111000 00000101 00111011 01010101 
      6 {48} 06 95 f8 05 41 2f : 00000110 10010101 11111000 00000101 01000001 00101111 
      6 {48} 06 95 f8 05 47 29 : 00000110 10010101 11111000 00000101 01000111 00101001 
      6 {48} 06 95 f8 05 4e 20 : 00000110 10010101 11111000 00000101 01001110 00100000 
      6 {48} 06 95 f8 05 54 3a : 00000110 10010101 11111000 00000101 01010100 00111010 
      6 {48} 06 95 f8 05 5a 34 : 00000110 10010101 11111000 00000101 01011010 00110100 
      6 {48} 06 95 f8 05 60 0e : 00000110 10010101 11111000 00000101 01100000 00001110 
[... cut ...]
      6 {48} 06 95 f8 07 19 15 : 00000110 10010101 11111000 00000111 00011001 00010101 
      6 {48} 06 95 f8 07 1d 11 : 00000110 10010101 11111000 00000111 00011101 00010001 
      6 {48} 06 95 f8 07 21 2d : 00000110 10010101 11111000 00000111 00100001 00101101 
      5 {48} 06 95 f8 07 26 2a : 00000110 10010101 11111000 00000111 00100110 00101010 
      6 {48} 06 95 f8 07 2a 26 : 00000110 10010101 11111000 00000111 00101010 00100110 
      6 {48} 06 95 f8 07 2e 22 : 00000110 10010101 11111000 00000111 00101110 00100010
```

merbanan, 2019-08-09:
```
06 95 f8 07 2e 22
  6^ 5^ 8^ 7^ e =2
```

Seems to match 4 of the bits.

zuckschwerdt, 2019-08-09:
But the upper nibbles are trouble.

zuckschwerdt, 2019-08-10:
@walthowd do you known what a bucket tip amounts to? It's usually some fraction of a mm.

walthowd, 2019-08-10:
@zuckschwerdt its 0.2mm

iandees, 2019-09-13:
Looks like a change was committed in to support the 899 rain gauge: https://github.com/merbanan/rtl_433/commit/4c4182e99f0d9fba513a16958e988b0211e6bf5c

Should that be considered complete? From the comments here it sounds like you're still working on it.

zuckschwerdt, 2019-09-14:
@iandees yes, I merged some code to get us started, but the values still need to be verified. The E.g. `rain_mm` is a raw counter output and probably *not* a mm value. @walthowd noted that 5 counts should be one mm, can you check and confirm that?

iandees, 2019-09-14:
I'm having a hard time getting my SDR to pick up the rain gauge's signal, but it seems that one tip of the bucket (the magnet passing over the sensor once) shows up as 0.01 mm on the display.

zuckschwerdt, 2019-09-14:
So 1/100 of a mm – 100 counts to a mm? That doesn't sound right.

walthowd, 2019-09-16:
I bought the 899 on eBay but was sent the 875 which is what the current branch added support for. I can definitely verify each bucket tip is 0.2mm which is also reflected on the display. 

This is documented as well in the manual:
https://www.dropbox.com/s/9hnkv96rka4l9gi/Screenshot%202019-09-16%2008.32.44.png?dl=0

https://www.acurite.com/media/manuals/00875-instructions.pdf (Page 11)

zuckschwerdt, 2019-09-16:
Thanks for the verification. The 899 then should be broadly in the same range, right? Maybe 0.1 but rather not 0.01?

walthowd, 2019-09-16:
The 899 manual says it has the same increments (0.2mm):

https://www.dropbox.com/s/d8pdkto1c6iqr6m/Screenshot%202019-09-16%2008.42.06.png?dl=0

https://www.acurite.com/media/manuals/00899-02445-instructions.pdf (Page 12)

iandees, 2019-09-16:
Looking closer at the screen I see that the 0.01 I mentioned above was displayed in inches, not millimeters. I must have pulled the batteries out and reset my desired units.

Mine is definitely an 899, but I'm still debugging connection problems. When I put the batteries in, rtl433 gets a couple data points and then stops seeing anything. The base station/screen doesn't see anything and blinks the antenna icon indicating no signal. I have fresh batteries in, checked the channel (both are A) and have them sitting right next to one another. I'll probably return it and try again...

iandees, 2019-09-17:
After putting in fresh batteries and being patient (it seems to send messages every 1-5 minutes), I'm also able to confirm that one tip of the bucket shows up on the screen as 0.2mm.

rtl_433 is able to see the device, but it only outputs timestamps. I can see a timestamp-only message from rtl_433 that coincides with when the rain counter screen updates.

Anything I can do to help with parsing?

zuckschwerdt, 2019-09-17:
That's really strange. There should never be only a timestamp. Is there more useful info with `-v` or `-vv` or `-vvv`?
If you can grab a sample (`-S all`) that only shows a timestamp when read back in, put it here (as zip).

iandees, 2019-09-17:
Here's a sample that contains the timestamp-only message: [g001_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/3622941/g001_433.92M_250k.cu8.zip)

Here's what the console saw during this:

```
/usr/local # rtl_433 -S all
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 106 out of 136 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-103 108-116 119 121 124-128 131-136 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-17 18:38:22
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-17 18:38:22
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-09-17 18:38:22
*** Saving signal to file g001_433.92M_250k.cu8 (68190 samples, 262144 bytes)
```

Note that for some reason I get three copies of the message output for everything, not just this.

iandees, 2019-09-17:
Also: during this sample, the screen read "0.5 mm".

zuckschwerdt, 2019-09-17:
Does it show timestamps only even when you read it back (`rtl_433 g001_433.92M_250k.cu8`)? Works fine here:
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.317064s
model     : Acurite-Rain899                        id        : 2935
channel   : 0            Battery   : 0             Rainfall Accumulation: 717252128 mm
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.317064s
model     : Acurite-Rain899                        id        : 2935
channel   : 0            Battery   : 0             Rainfall Accumulation: 717251904 mm
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.317064s
model     : Acurite-Rain899                        id        : 2935
channel   : 0            Battery   : 0             Rainfall Accumulation: 717252016 mm
```

iandees, 2019-09-17:
Yep:

```
Test mode active. Reading samples from file: /tmp/parenttmp/g001_433.92M_250k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.317064s
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.317064s
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.317064s
```

This is built from https://github.com/merbanan/rtl_433/commit/3d36258a92052e655b419a750894494c4706e967.

I'll try again with a newer rev.

iandees, 2019-09-17:
Same thing with abb2a4dd and 6d75c7db. 🤔

zuckschwerdt, 2019-09-18:
Strange. Something is broken on your setup. Try other outputs (`-F json`).

iandees, 2019-09-18:
Other outputs also only show timestamp.

I'm running this on my Synology NAS via Docker (the SDR is connected to it via USB). I'll try directly from my laptop later.

iandees, 2019-09-18:
Apologies for making this a bit off-topic. If there's an IRC channel or you'd like me to file a separate ticket, let me know.

To confirm, I'm using [BUILDING.md](https://github.com/merbanan/rtl_433/blob/master/docs/BUILDING.md#cmake) and doing the following:

```sh
curl -Ls https://github.com/merbanan/rtl_433/archive/19.08.tar.gz | tar -zxf -
cd rtl_433-19.08/
mkdir build
cd build
cmake ..
make
```

and getting

```sh
# src/rtl_433 -V
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
```

Should the build process be inserting the version number in the `-V` output?

zuckschwerdt, 2019-09-18:
You only get a build number when using git. (`git clone https://github.com/merbanan/rtl_433.git ; cd rtl_433 ; mkdir build ; cd build ; cmake .. ; make`)

iandees, 2019-09-19:
Built it on my Mac and got the same timestamp-only results when reading the capture:

```
$ src/rtl_433 ~/Downloads/g001_433.92M_250k.cu8
rtl_433 version 19.08-13-gabb2a4d branch master at 201909172102 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/iandees/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 108 out of 138 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-103 108-116 119 121 124-128 131-138 ]
Test mode active. Reading samples from file: /Users/iandees/Downloads/g001_433.92M_250k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.317064s
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.317064s
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.317064s
```

iandees, 2019-09-20:
@zuckschwerdt which version of librtlsdr are you compiling with? Might that cause a problem?

zuckschwerdt, 2019-09-20:
I don't suspect that to be an issue. If RTL-SDR works it's supported.
The problem here is that the output data is either truncated in the decoder or on print. Very strange...
Does reading other tes files from the rtl_433_tests repo work? Is it limited to this particular protocol / decoder?

iandees, 2019-09-20:
I get data from my other sensors. It's all repeated three times, but it does come through correctly.

```
# rtl_433 -F json
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 106 out of 136 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-103 108-116 119 121 124-128 131-136 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
Failed to allocate zero-copy buffer for transfer 0
Falling back to buffers in userspace
{"time" : "2019-09-20 21:02:19", "model" : "Acurite tower sensor", "id" : 13142, "sensor_id" : 13142, "channel" : "B", "temperature_C" : 24.800, "humidity" : 46, "battery_low" : 1}
{"time" : "2019-09-20 21:02:19", "model" : "Acurite tower sensor", "id" : 13142, "sensor_id" : 13142, "channel" : "B", "temperature_C" : 24.800, "humidity" : 46, "battery_low" : 1}
{"time" : "2019-09-20 21:02:19", "model" : "Acurite tower sensor", "id" : 13142, "sensor_id" : 13142, "channel" : "B", "temperature_C" : 24.800, "humidity" : 46, "battery_low" : 1}
{"time" : "2019-09-20 21:02:25", "model" : "Acurite tower sensor", "id" : 2935, "sensor_id" : 2935, "channel" : "A", "temperature_C" : 20.600, "humidity" : 63, "battery_low" : 0}
{"time" : "2019-09-20 21:02:25", "model" : "Acurite tower sensor", "id" : 2935, "sensor_id" : 2935, "channel" : "A", "temperature_C" : 20.600, "humidity" : 63, "battery_low" : 0}
{"time" : "2019-09-20 21:02:25", "model" : "Acurite tower sensor", "id" : 2935, "sensor_id" : 2935, "channel" : "A", "temperature_C" : 20.600, "humidity" : 63, "battery_low" : 0}
{"time" : "2019-09-20 21:02:26", "model" : "Acurite tower sensor", "id" : 12824, "sensor_id" : 12824, "channel" : "A", "temperature_C" : 31.300, "humidity" : 58, "battery_low" : 0}
{"time" : "2019-09-20 21:02:26", "model" : "Acurite tower sensor", "id" : 12824, "sensor_id" : 12824, "channel" : "A", "temperature_C" : 31.300, "humidity" : 58, "battery_low" : 0}
{"time" : "2019-09-20 21:02:26", "model" : "Acurite tower sensor", "id" : 12824, "sensor_id" : 12824, "channel" : "A", "temperature_C" : 31.300, "humidity" : 58, "battery_low" : 0}
```

When running it with data from tests it seems to work (although maybe duplicating output?)

```
rtl_433 acurite/Acurite_986/Acurite_TX_51.cu8
rtl_433 version 19.08 branch  at 201908291429 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 106 out of 136 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-103 108-116 119 121 124-128 131-136 ]
Test mode active. Reading samples from file: acurite/Acurite_986/Acurite_TX_51.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.123836s
model     : Acurite 986 Sensor                     id        : 17021
channel   : 1R           temperature: 51.000000 F  battery   : OK            status    : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.123836s
model     : Acurite 986 Sensor                     id        : 17021
channel   : 1R           temperature: 51.000000 F  battery   : OK            status    : 0
```

zuckschwerdt, 2019-09-20:
Can you confirm it's really that decoder returning the bad data? `rtl_433 -R 40 filename.cu8`

iandees, 2019-09-21:
Yep, looks like it is:

```
# rtl_433 -R 40 /tmp/parenttmp/g001_433.92M_250k.cu8
rtl_433 version 19.08 branch  at 201908291429 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 1 out of 136 device decoding protocols [ 40 ]
Test mode active. Reading samples from file: /tmp/parenttmp/g001_433.92M_250k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.317064s
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.317064s
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.317064s
```

zuckschwerdt, 2019-09-21:
Perhaps sprinkle the code with printf's to see where the `data` goes missing.
Put e.g. `printf("%s : %d = %p\n", __func__, __LINE__, data);` in acurite.c after `// Rain Fall Gauge 899`, before `data_make` and `decoder_output_data`.

iandees, 2019-09-21:
Found a fix to get data out of `data_make()` and filed a PR for it.

zuckschwerdt, 2019-09-21:
Oh no, I'm so sorry. I have an uncommited change …`raincounter * 0.2` there. That's why it was working for me. Thanks for seeing this through and fixing it!

iandees, 2019-09-21:
No problem! Thanks for the debug pointers. Seems to be working.

merbanan, 2020-01-07:
@zuckschwerdt can this be closed?

closed 2020-01-07 by zuckschwerdt

zuckschwerdt, 2022-06-10:
"899 Rain" is now added to the list with 5c17e4e.